mavlink/mavros #1644: Unable to Offboard

iamrajee, 2021-11-08:
----

### Issue details
Terminal1: `roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"`
Terminal2: `roslaunch px4 posix_sitl.launch`
Terminal3: `rosrun mavros mavsafety arm`
Terminal4: `rosrun mavros mavsys mode -c OFFBOARD`

Error:
Terminal1: `CMD: Unexpected command 176, result 1`
Terminal2: 
```
INFO  [commander] Armed by external command	
INFO  [tone_alarm] notify negative
INFO  [commander] Disarmed by auto preflight disarming	
INFO  [logger] closed logfile, bytes written: 1873827
```
Terminal3: `Timed out!`

### MAVROS version and platform

<! -- Mavros: ?0.18.4? -->
ROS: ?Noetic?
Ubuntu: ?20.04?

### Autopilot type and version

[  ] PX4

Version: ?3.7.1?

### Node logs
<details> 
 <summary>Node logs</summary>
```
... logging to /home/rajendra/.ros/log/18daa76e-4087-11ec-b188-d915890be05a/roslaunch-u20-86915.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://u20:40867/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
[... 778 lines cut ...]
        key: "Estimated time offset (s)"
        value: "1636370958.136772394"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 6150 messages, from 1 addresses
sys:comp   list of messages
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/timer.py", line 237, in run
    self._callback(TimerEvent(last_expected, last_real, current_expected, current_real, last_duration))
  File "/home/rajendra/drone_ws/src/mavros/mavros/scripts/checkid", line 82, in timer_cb
    for address, messages in self.message_sources.iteritems():
AttributeError: 'dict' object has no attribute 'iteritems'

```

Jaeyoung-Lim, 2021-11-08:
@iamrajee What are you trying to achieve? If you don't send setpoints, you will not be able to change your mode into offboard mode since there is no setpoint to track

gssammy, 2021-11-09:
i am seeing this also. I'm publishing setpoints at a rate of 10Hz, and px4 sitl is not able to transition to offboard mode.

PX4 : 1.12.3
mavros: 1.9.0
Ubuntu: 20.04

Trying to run: `roslaunch px4 mavros_posix_sitl.launch vehicle:=iris_rplidar` and command to the drone to go up to 1m. 

position setpoint: 
```
rostopic hz /mavros/setpoint_position/local 
subscribed to [/mavros/setpoint_position/local]
WARNING: may be using simulated time
average rate: 10.045
	min: 0.096s max: 0.100s std dev: 0.00126s window: 10
average rate: 10.042
	min: 0.096s max: 0.100s std dev: 0.00123s window: 20
average rate: 10.028
	min: 0.096s max: 0.104s std dev: 0.00146s window: 30
average rate: 10.031
	min: 0.096s max: 0.104s std dev: 0.00167s window: 40
average rate: 10.025
	min: 0.096s max: 0.104s std dev: 0.00149s window: 50
average rate: 10.027
	min: 0.096s max: 0.104s std dev: 0.00145s window: 60
average rate: 10.023
	min: 0.096s max: 0.104s std dev: 0.00151s window: 70
```

iamrajee, 2021-11-17:
> @iamrajee What are you trying to achieve? If you don't send setpoints, you will not be able to change your mode into offboard mode since there is no setpoint to track

I tried to the public the set point as below but it didn't helped.
```
rostopic pub /mavros/setpoint_position/local geometry_msgs/PoseStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: 'map'
pose:
  position:
    x: 0.0
    y: 0.0
    z: 2.0
  orientation:
    w: 0.0" 
```

But surprising when I tried below script it take off:
```
/**
 * @file offb_node.cpp
 * @brief offboard example node, written with mavros version 0.14.2, px4 flight
 * stack and tested in Gazebo SITL
 */

#include <ros/ros.h>
#include <geometry_msgs/PoseStamped.h>
#include <mavros_msgs/CommandBool.h>
#include <mavros_msgs/SetMode.h>
#include <mavros_msgs/State.h>

mavros_msgs::State current_state;
void state_cb(const mavros_msgs::State::ConstPtr& msg){
    current_state = *msg;
}

int main(int argc, char **argv)
{
    ros::init(argc, argv, "offb_node");
    ros::NodeHandle nh;

    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>
            ("mavros/state", 10, state_cb);
    ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>
            ("mavros/setpoint_position/local", 10);
    ros::ServiceClient arming_client = nh.serviceClient<mavros_msgs::CommandBool>
            ("mavros/cmd/arming");
    ros::ServiceClient set_mode_client = nh.serviceClient<mavros_msgs::SetMode>
            ("mavros/set_mode");

    //the setpoint publishing rate MUST be faster than 2Hz
    ros::Rate rate(20.0);

    // wait for FCU connection
    while(ros::ok() && current_state.connected){
        ros::spinOnce();
        rate.sleep();
    }

    geometry_msgs::PoseStamped pose;
    pose.pose.position.x = 0;
    pose.pose.position.y = 0;
    pose.pose.position.z = 2;

    //send a few setpoints before starting
    for(int i = 100; ros::ok() && i > 0; --i){
        local_pos_pub.publish(pose);
        ros::spinOnce();
        rate.sleep();
    }

    mavros_msgs::SetMode offb_set_mode;
    offb_set_mode.request.custom_mode = "OFFBOARD";

    mavros_msgs::CommandBool arm_cmd;
    arm_cmd.request.value = true;

    ros::Time last_request = ros::Time::now();

    while(ros::ok()){
        if( current_state.mode != "OFFBOARD" &&
            (ros::Time::now() - last_request > ros::Duration(5.0))){
            if( set_mode_client.call(offb_set_mode) &&
                offb_set_mode.response.mode_sent){
                ROS_INFO("Offboard enabled");
            }
            last_request = ros::Time::now();
        } else {
            if( !current_state.armed &&
                (ros::Time::now() - last_request > ros::Duration(5.0))){
                if( arming_client.call(arm_cmd) &&
                    arm_cmd.response.success){
                    ROS_INFO("Vehicle armed");
                }
                last_request = ros::Time::now();
            }
        }

        local_pos_pub.publish(pose);

        ros::spinOnce();
        rate.sleep();
    }

    return 0;
}
```
I'm not sure I might be doing wrong when manual publishing set point. Also can you please tell me the correct order of "Arming", "Setting Offboard mode" and "publishing set point"? Thank you so much!

Jaeyoung-Lim, 2021-11-17:
@iamrajee You need to be publishing your setpoint for at least 2Hz in order to switch to offboard mode. Your code is publishing a few setpoints then trying to switch to offboard mode

iamrajee, 2021-11-17:
Thank you so much @Jaeyoung-Lim, I tried your suggestions and it worked. I'm new to mavros and didn't knew that. 

I'm describing my steps herewith in case anyone faces a similar issue: (works!)

**Terminal1:** `roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"`
**Terminal2:** `roslaunch px4 posix_sitl.launch`
**Terminal3:**
```
rostopic pub -r 10 /mavros/setpoint_position/local geometry_msgs/PoseStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: 'map'
pose:
  position:
    x: 0.0
    y: 0.0
    z: 2.0
  orientation:
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0" 
```
**Terminal4:** `rosrun mavros mavsys mode -c OFFBOARD`
**Terminal5:** `rosrun mavros mavsafety arm`

Jaeyoung-Lim, 2021-11-17:
@iamrajee Glad that it worked, and thanks for sharing the solution!

iamrajee, 2021-11-21:
I'm facing the same issue in melodic as well. However, I couldn't resolve it by the above method. I tried both cpp script and manually publishing setpoint but it dint work. Surprisingly same thing works in noetic. Also. I've tried both binary and source versions of mavros with no luck. Here is how I'm running mavros:

**Terminal 1: roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"**
```
logging to /home/rajendra/.ros/log/1c22cf90-4ad3-11ec-9285-7cd30a81f90b/roslaunch-u18-29152.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://u18:40415/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@127....
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
[... 546 lines cut ...]
  frame_id: ''
connected: True
armed: False
guided: True
manual_input: False
mode: "AUTO.RTL"
system_status: 3
```

# rostopic echo /mavros/state (After running arm and offboard)
```
header: 
  seq: 194
  stamp: 
    secs: 1637502385
    nsecs: 932277774
  frame_id: ''
connected: True
armed: True
guided: True
manual_input: False
mode: "AUTO.RTL"
system_status: 4
```
</details>

PaulBauer99, 2022-01-19:
I am facing the exact same issue in ROS Melodic. Can anybody confirm/refute that this is an issue with the version? Did you get it working in Melodic @iamrajee?

Kind regards

peterminh227, 2022-01-21:
It seems to be a problem with the current firmware version.  
* it works fine with v1.10.2
* I used the following steps:
```
cd PX4-Autopilot
sudo git checkout v1.10.2 
git submodule update --init --recursive
```
```
cd ..
chmod -R 777 PX4-Autopilot/
```
Install tools & dependencies
```
bash ./PX4-Autopilot/Tools/setup/ubuntu.sh
sudo apt install -y ninja-build exiftool python-argparse python-empy python-toml python-numpy python-yaml python-dev python-pip ninja-build protobuf-compiler libeigen3-dev genromfs xmlstarlet libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev
sudo pip3 install --user pandas jinja2 pyserial cerberus pyulog numpy toml pyquaternion  
```
Edit an error in v1.10.2 in
```
gedit ./PX4-Autopilot/Tools/sitl_gazebo/include/gazebo_opticalflow_plugin.h
```
please change as following
```
#define HAS_GYRO true (make TRUE lowercase).
```
```
cd PX4-Autopilot
make px4_sitl_default gazebo
```
A usefull script to add paths mavros & SITL: launch_common.sh
```
#!/bin/bash

px4_dir=$(pwd)/../PX4-Autopilot

source $px4_dir/Tools/setup_gazebo.bash $px4_dir $px4_dir/build/px4_sitl_default

export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$px4_dir
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$px4_dir/Tools/sitl_gazebo

```
and running mavros & SITL
```
#!/bin/bash

source ./launch-common.sh
# mavros
gnome-terminal -- /bin/sh -c 'roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557";exec bash'
# px4-sitl
gnome-terminal -- /bin/sh -c 'source /home/samurai/shell_script/launch-common.sh;roslaunch px4 posix_sitl.launch;exec bash'

```

iamrajee, 2022-01-21:
@peterminh227 Thank you sharing your solution :)

peterminh227, 2022-01-21:
> @peterminh227 Thank you sharing your solution :)

My pleasure :D

closed 2026-10-30 by github-actions[bot]