mavlink/mavros #1653: Distance sensor quaternion field should be initialized to zero, not identity

mortenfyhn, 2021-11-24:
There may be a mistake in https://github.com/mavlink/mavros/commit/137a296c3b90ac674b924945524d499a9668f21c: For distance sensors specifically, I believe the quaternion field should be zero (not identity) when it's unused. It's unused when the orientation field (which is a MAV_SENSOR_ORIENTATION enum) is set to anything else than MAV_SENSOR_ROTATION_CUSTOM.

So in the distance sensor plugin code, I think it should be initialized to zero.

For reference:

https://mavlink.io/en/messages/common.html#DISTANCE_SENSOR says

> Quaternion of the sensor orientation in vehicle body frame (w, x, y, z order, zero-rotation is 1, 0, 0, 0). Zero-rotation is along the vehicle body x-axis. This field is required if the orientation is set to MAV_SENSOR_ROTATION_CUSTOM. Set it to 0 if invalid."

vooon, 2021-11-24:
Well, with sending identity rotation, we're on safe side if someone didn't check for (0 0 0 0). And since CUSTOM is required, we're safe to send it.
Only what we loose - is a bit longer message on the wire, since zero-tail algo can't fully drop that field.
But yes, maybe it's worth it on high rate devices.

mortenfyhn, 2021-11-24:
Yeah, probably either way is fine. I'm only basing this argument on the description in the mavlink message ("Set it to 0 if invalid."). If you disagree, feel free to close this.

vooon, 2021-11-24:
I think that's mainly said because of that field is an extension, which can be zeroed, and because that allow to shrink message a bit.
I'll revert back to zero.

commit 2dc98490 (2021-11-24): extras: distance_sensor: revert back to zero quaternion

Fix #1653
closed 2021-11-24 by vooon