mavlink/mavros #1656: HDOP and VDOP in mavros?

itzmejawad, 2021-11-25:
Hi.. is there any way to request HDOP and VDOP data

Subscribing to /mavros/global_position/raw/satellites , it gives only the number of satellites..

[image]

and also if i subscribe to  /mavros/global_position/raw/fix, it gives 

[image]

 but how can i get the position accuracy value, hdop and vdop? is there any topic for that?
 
 Otherwise what is this position co variance... what is the safe threshold for that?
 

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?18.04?

[ x ] ArduPilot
[ x ] PX4

vooon, 2021-11-25:
`position_covariance`.

Harishferose, 2021-11-25:
@vooon 

I am new to mavros, I need  HDOP and VDOP Values

Cant able to understand the position_covariance matrix

Can you share how to fetch/calculate the HDOP and VDOP Values from position_covariance

Thanks in advance

vooon, 2021-11-25:
https://github.com/mavlink/mavros/blob/c121f6bde0b47bc031bf92a9a746bac4970944c7/mavros/src/plugins/global_position.cpp#L180-L193

selvanathans, 2021-11-25:
@vooon  Could you share [this](https://github.com/mavlink/mavros/issues/1656#issuecomment-979168289)  snippet as python?

vooon, 2021-11-25:
That's just a code that computes covariance in the plugin. You can extract eph/epv or h/v accuracy doing opposite.
I.e.:
1. if type==1: `eph, epv = sqrt(cov[0,0]), sqrt(cov[2,2])` (NOTE: eph/v are unitless)
2. if type==2: `h_acc, v_acc  = sqrt(cov[0,0]), sqrt(cov[2,2])` (NOTE: h/v acc is in meters)

itzmejawad, 2021-11-25:
@vooon Thank You :)

closed 2026-10-29 by github-actions[bot]