mavlink/mavros #1657: Values cannot be echoed or subscribed to, set_stream_rate service call inconsistent behavior

weicong96, 2021-11-26:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I have read other issues filed that set_stream_rate is required to resolve the issue with no satellites found. However, calling set_stream_rate has inconsistent behavior for me.

When it works:
[image]

When it does not work it hangs indefinitely without any response, it has past 7 minutes at the time of writing this:
[image]

I would be able to get telemetry data coming in if I had connected to QGC or Mission Planner, disconnect it and connect via mavros.

in the case of stream rate call hanging, only able to echo /mavros/state and not other topics.

Is there a solution for setting stream rate?

### MAVROS version and platform

Mavros: 2.0.3
ROS: Foxy
Ubuntu: 18.04

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: 4.0.3

### Node logs

```
[INFO] [1637885590.807011997] [mavros_node]: Starting mavros_node container
[INFO] [1637885590.807160800] [mavros_node]: FCU URL: tcp://localhost:5760
[INFO] [1637885590.807176798] [mavros_node]: GCS URL: udp://@192.168.1.117
[INFO] [1637885590.807184537] [mavros_node]: UAS Prefix: /uas1
[INFO] [1637885590.807191724] [mavros_node]: Starting mavros router node
[INFO] [1637885591.014065182] [mavros_router]: Built-in SIMD instructions: SSE, SSE2
[INFO] [1637885591.014152440] [mavros_router]: Built-in MAVLink package version: 2021.10.10
[INFO] [1637885591.014179103] [mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV all development icarous matrixpilot paparazzi standard uAvionix ualberta
[INFO] [1637885591.014190500] [mavros_router]: MAVROS Router started
[INFO] [1637885591.014252918] [mavros_router]: Requested to add endpoint: type: 0, url: tcp://localhost:5760
[INFO] [1637885591.014315798] [mavros_router]: Endpoint link[1000] created
[INFO] [1637885591.029710371] [mavros_router]: link[1000] opened successfully
[INFO] [1637885591.030049807] [mavros_router]: Requested to add endpoint: type: 1, url: udp://@192.168.1.117
[INFO] [1637885591.030137365] [mavros_router]: Endpoint link[1001] created
[INFO] [1637885591.030377034] [mavros_router]: link[1001] opened successfully
[INFO] [1637885591.030431283] [mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[INFO] [1637885591.030457194] [mavros_router]: Endpoint link[1002] created
[INFO] [1637885591.031851626] [mavros_router]: link[1002] opened successfully
[INFO] [1637885591.031904712] [mavros_node]: Starting mavros uas node
[INFO] [1637885592.155947944] [mavros_router]: link[1000] detected remote address 1.1
[WARN] [1637885592.222138016] [mavros]: diagnostic_updater: No HW_ID was set. This is probably a bug. Please report it. For devices that do not have a HW_ID, set this value to 'none'. This warning only occurs once all diagnostics are OK. It is okay to wait until the device is open before calling setHardwareID.
[INFO] [1637885592.322537428] [mavros]: UAS Executor started
[INFO] [1637885593.056227582] [mavros]: Plugin actuator_control created
[INFO] [1637885593.056306540] [mavros]: Plugin actuator_control added to executor
[INFO] [1637885593.056342946] [mavros]: Plugin actuator_control initialized
[INFO] [1637885593.070531455] [mavros]: Plugin altitude created
[INFO] [1637885593.070601932] [mavros]: Plugin altitude added to executor
[INFO] [1637885593.070620646] [mavros]: Plugin altitude initialized
[INFO] [1637885593.081712058] [mavros]: Plugin command created
[INFO] [1637885593.081801842] [mavros]: Plugin command added to executor
[INFO] [1637885593.081831923] [mavros]: Plugin command initialized
[INFO] [1637885593.113814613] [mavros]: Plugin ftp created
[INFO] [1637885593.114010087] [mavros]: Plugin ftp added to executor
[... 424 lines cut ...]
- level: "\0"
  name: 'mavros: Heartbeat'
  message: Normal
  hardware_id: uas:///uas1
  values:
  - key: Heartbeats since startup
    value: '90'
  - key: Frequency (Hz)
    value: '0.900363'
  - key: Vehicle type
    value: Quadrotor
  - key: Autopilot type
    value: ArduPilot
  - key: Mode
    value: STABILIZE
  - key: System status
    value: STANDBY
---
```

### Check ID

```
No executable found
```

vooon, 2021-11-26:
Duplicate of #1588

closed 2021-11-26 by vooon

weicong96, 2021-11-27:
@vooon 
Fyi, It appears that the above situation is resolved after making set_stream_rate run and wait in the background before starting up the node. I now have telemetry streaming but calling the set_mode service does not have any response that comes back. The set_mode call:
  ```
  response = None
  req = SetMode.Request()
  req.custom_mode = 'GUIDED'

  set_mode_future = self.set_mode_client.call_async(req)
  while rclpy.ok():
    try:
      if set_mode_future.done():
        response = set_mode_future.result()
        break
    except Exception as e:
      self.log.error(str(e))
  return response
```

I also saw that in set_mode message definition([link](http://docs.ros.org/en/api/mavros_msgs/html/srv/SetMode.html)) , there is a base_mode field, i'm wondering if i'm doing something wrong? Also tried with base_mode set to one of the enums, none of them give a different behavior.