mavlink/mavros #1659: How to understand and set the connection url

dzywater, 2021-11-27:
Connection url support several schemas 

Serial: /path/to/serial/device[:baudrate]
Serial: serial:///path/to/serial/device[:baudrate][?ids=sysid,compid]
Serial with hardware flow control: serial-hwfc:///path/to/serial/device[:baudrate][?ids=sysid,compid]
UDP: udp://[bind_host][:port]@[remote_host][:port][/?ids=sysid,compid]
UDP broadcast until GCS discovery: udp-b://[bind_host][:port]@[:port][/?ids=sysid,compid]
UDP broadcast (permanent): udp-pb://[bind_host][:port]@[:port][/?ids=sysid,compid]
TCP client: tcp://[server_host][:port][/?ids=sysid,compid]
TCP server: tcp-l://[bind_port][:port][/?ids=sysid,compid]

For UDP, how to set and understand the two parts before and after @?

vooon, 2021-11-27:
First part is where you bind to, second optional part tells initial remote endpoint address.

closed 2021-11-27 by vooon

dzywater, 2021-11-28:
> First part is where you bind to, second optional part tells initial remote endpoint address.
I cannot understand yet.

What I thought:
fcu_url: first part means local address. What is the second part ? is it meaningful?
gcs_url: second part means remote address. Is the first part meanfully?

Besides, how to use the mavros2, are there some demos ?

vooon, 2021-11-28:
`fcu_url` provides connection to your Flight Control Unit, in other words, your autopilot. What connection method should be used is up to AP config. If your AP listens on 192.168.1.100:14550, your computer wants to listen on 192.168.1.200:14555, then your url: `udp://192.168.1.200:14555@192.168.1.100:14550`. But you may bind to 0.0.0.0:14555 (which is default host and port), so url will look as follows `udp://@192.168.1.100:14550`. But if your AP preconfigured to send packets to 192.168.1.200:14555, you may let connection to retrieve remote endpoint address, so url: `udp://@`.

`gcs_url` provides a proxy for Ground Control Station, it simply copy all messages between fcu and gcs links.
Please read about BSD sockets to understand `bind` and `listen` terms better.

dzywater, 2021-11-29:
OK, thanks! I should read something. Could you write some demos for ros2.
In the PX4 gazebo simulation, I want to try some simple commands.
I call service /mavros/cmd/arming, it works(the drone arms).
I call service /mavros/cmd/takeoff, it does not work(the drone does not act).
[image]
My connection string: ros2 run mavros mavros_node --ros-args --param fcu_url:=udp://:14540@ --param gcs_url:=udp://@10.42.0.11:14550
The QGroundControl on 10.42.0.11 work normally and can let takeoff.

vooon, 2021-11-29:
I strongly suggest you use mavros v1.11.1 trough ros1_bridge instead of v2 as it still have #1601 bug.
But i also suggest to look for `FCU: ...` log messages, as they come from autopilot, and usually have some useful notes why some action is prohibited.
But yes, usually, if drone arms, it should liftoff.