mavlink/mavros #1661: mavftp is not working

itzmejawad, 2021-11-30:
### Issue details

mavftp is not working with the following command

**rosrun mavros mavftp list**

need help!

[image]

ROS: ?Melodic?
Ubuntu: ?18.04?

vooon, 2021-11-30:
Service unavailable error. Did you run mavros?

itzmejawad, 2021-12-01:
@vooon 

Mavros is running... Also I am running Ardupilot in Pixhawk

[image]

in this mavros wiki , i see MAVFTP is for PX4

I have Logs in my Pixhawk's SD card

As u said... even if i run "rostopic list" ..... there are no topics shown for mavftp...

If the mavftp plugin is missing in my ros-melodic... how can i add ?

itzmejawad, 2021-12-01:
rostopic echo -n1 /diagnostics
header: 
  seq: 2138
  stamp: 
    secs: 1638349177
    nsecs: 621137835
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "/dev/ttyPixhawk:115200"
    values: 
      - 
        key: "Received packets:"
        value: "35315"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "243"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "991775"
      - 
        key: "Tx total bytes:"
        value: "949412"
      - 
        key: "Rx speed:"
        value: "566.000000"
      - 
        key: "Tx speed:"
        value: "500.000000"
  - 
    level: 2
    name: "mavros: GPS"
    message: "No satellites"
    hardware_id: "/dev/ttyPixhawk:115200"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "Unknown"
      - 
        key: "EPV (m)"
        value: "Unknown"
  - 
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "/dev/ttyPixhawk:115200"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "3061"
      - 
        key: "Frequency (Hz)"
        value: "0.962966"
      - 
        key: "Vehicle type"
        value: "Quadrotor"
      - 
        key: "Autopilot type"
        value: "ArduPilot"
      - 
        key: "Mode"
        value: "STABILIZE"
      - 
        key: "System status"
        value: "Standby"
  - 
    level: 0
    name: "mavros: System"
    message: "Normal"
    hardware_id: "/dev/ttyPixhawk:115200"
    values: 
      - 
        key: "Sensor present"
        value: "0x00000000"
      - 
        key: "Sensor enabled"
        value: "0x00000000"
      - 
        key: "Sensor health"
        value: "0x00000000"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 2
    name: "mavros: Battery"
    message: "No data"
    hardware_id: "/dev/ttyPixhawk:115200"
    values: 
      - 
        key: "Voltage"
        value: "-1.00"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "0.0"
  - 
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "/dev/ttyPixhawk:115200"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "4239"
      - 
        key: "Frequency (Hz)"
        value: "10.000025"
      - 
        key: "Last RTT (ms)"
        value: "6.637244"
      - 
        key: "Mean RTT (ms)"
        value: "5.324947"
      - 
        key: "Last remote time (s)"
        value: "5581.191723000"
      - 
        key: "Estimated time offset (s)"
        value: "1638343596.331534863"
---

itzmejawad, 2021-12-01:
CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
 * /mavros/fake_gps/satellites_visible: 6
 * /mavros/fake_gps/speed_accuracy: 0.0
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_hil_gps: True
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fake_gps/vert_accuracy: 0.5
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyPixhawk:1...
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
 * /mavros/global_position/use_relative_alt: True
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /mavros/imu/frame_id: base_link
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/landing_target/camera/fov_x: 2.0071286398
 * /mavros/landing_target/camera/fov_y: 2.0071286398
 * /mavros/landing_target/image/height: 480
 * /mavros/landing_target/image/width: 640
 * /mavros/landing_target/land_target_type: VISION_FIDUCIAL
 * /mavros/landing_target/listen_lt: False
 * /mavros/landing_target/mav_frame: LOCAL_NED
 * /mavros/landing_target/target_size/x: 0.3
 * /mavros/landing_target/target_size/y: 0.3
 * /mavros/landing_target/tf/child_frame_id: camera_center
[... 54 lines cut ...]
 * /mavros/vision_pose/tf/rate_limit: 10.0
 * /mavros/vision_speed/listen_twist: True
 * /mavros/vision_speed/twist_cov: True
 * /mavros/wheel_odometry/child_frame_id: base_link
 * /mavros/wheel_odometry/count: 2
 * /mavros/wheel_odometry/frame_id: map
 * /mavros/wheel_odometry/send_raw: True
 * /mavros/wheel_odometry/send_twist: False
 * /mavros/wheel_odometry/tf/child_frame_id: base_link
 * /mavros/wheel_odometry/tf/frame_id: map
 * /mavros/wheel_odometry/tf/send: True
 * /mavros/wheel_odometry/use_rpm: False
 * /mavros/wheel_odometry/vel_error: 0.1
 * /mavros/wheel_odometry/wheel0/radius: 0.05
 * /mavros/wheel_odometry/wheel0/x: 0.0
 * /mavros/wheel_odometry/wheel0/y: -0.15
 * /mavros/wheel_odometry/wheel1/radius: 0.05
 * /mavros/wheel_odometry/wheel1/x: 0.0
 * /mavros/wheel_odometry/wheel1/y: 0.15
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    mavros (mavros/mavros_node)

vooon, 2021-12-03:
What is in `/mavros/plugin_blacklist`?

itzmejawad, 2022-08-26:
@vooon 
in Ardupilot..........

[image]

the ftp is removed from blacklist but still

[image]
[image]

Rand0409, 2022-09-07:
The reasong might be this PX4 only...  I am using Ardupilot, seems you are using Ardu instead of PX as well. 

[image]

vooon, 2022-09-11:
@Rand0409 initially mavlink ftp was implemented only in PX4. Today APM also have it, but i didn't test it. Probably need to add some fixes to the plugin.

Rand0409, 2022-09-27:
Yeah, I got a lot Input/Output error using APM with mavros mavftp upload

closed 2026-10-29 by github-actions[bot]