mavlink/mavros #1662: read and send MAV_CMD_USER_* messages using mavros

tadteo, 2021-11-30:
### Issue details

Is it possible to read and send the MAV_CMD_USER_* messages using mavros? What is the best way to read and generate them?

### MAVROS version and platform

Mavros: Installed from apt ros-noetic-mavros
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

I am trying to send and receive MAV_CMD_USER_* messages from a ground station computer, to an onboard computer running mavros on the vehicle, through autopilot. I did not understand if those messages are available in mavros or not.

I'll explain a little bit better.

I'm using a ground computer with a telemetry connection to the vehicle. This ground computer uses pymavlink to read and generate the messages. On the vehicle, I have the autopilot and an onboard computer running mavros.

I want to use the MAV_CMD_USER_* messages to set up a particular landing protocol communication between the ground computer and the onboard computer on the vehicle.

Is this feasible? Or do you suggest another approach?

vooon, 2021-11-30:
You're meaning that you need to handle COMMAND_LONG coming to gcs_link? No, that's not possible with mavros v1.
Only way to do that - pass you're message addressed to 1.191 on fcu_link. Then it'll be possible to write a plugin to handle that message.
Possibly you may put mavlink-router between ap and mavros, and use it to provide gcs link.

tadteo, 2021-11-30:
No sorry, I think I was not clear.

The communication path of the messages can be seen here:

[MY_PROGRAM_ON_GROUND]<--->[PYMAVLINK]<--->[GROUND_STATION]<--telemetry-->[PX4]<--->[ONBOARD_COMPUTER]<--->[MAVROS]<--->[MY_PROGRAM_ON_THE_DRONE]

The ground computer is communicating directly with the autopilot using a telemetry link, and the onboard computer is then connected to the autopilot, and it is consuming the messages sent by the autopilot.

Essentially, I need to handle user-defined messages, from my two programs. One is running on the onboard computer on the drone, and the other is running on a computer on the ground. 

To do that, I was thinking to use the [MAV_CMD_USER_1](https://mavlink.io/en/messages/common.html#MAV_CMD_USER_1)  message. 

My first question is if this message can be read and sent by mavros. Since, I have also seen [here](http://docs.ros.org/en/api/mavros_msgs/html/msg/CommandCode.html) in the cog part, that the "USER" messages are inserted in the ``exception list of commands to not include``.
So I was not sure if mavros has enabled the handling of the MAV_CMD_USER_* messages.

The second question is if I can broadcast these messages received by the autopilot, directly to the onboard computer. (But I am not sure if this is a PX4 related question).

vooon, 2021-12-03:
1. You may pass any MAV_CMD code. It's just a field in [COMMAND_LONG](https://mavlink.io/en/messages/common.html#COMMAND_LONG) message. Command plugin emits COMMAND_LONG or COMMAND_INT and receives COMMAND_ACK, so it's not the thing you need. To be able to receive COMMAND_LONG (of certain type) you have to write custom plugin (which is easy, and can be placed in your package, look at dummy.cpp).
2. That depends on existence of mavlink router inside a firmware, so it can pass a message from one port (telemetry) to another (obc). I know that APM have it, but didn't look PX4 for a while.

tadteo, 2021-12-16:
Hi again,
Thank you for the previous information. 
I have been able to share the messages between the drone (FCU) and the computer on the ground (if someone is interested to know how you can look at [this issue](https://github.com/PX4/PX4-SITL_gazebo/issues/834)). 
However, I am not able to receive the messages sent from the computer on the ground to the offboard computer.
The  messages (MAV_CMD) arrive on the autopilot, however, these are not sent (or received) to the offboard computer (they do not arrive in the /mavlink/from topic)

tadteo, 2022-01-14:
> 1. You may pass any MAV_CMD code. It's just a field in [COMMAND_LONG](https://mavlink.io/en/messages/common.html#COMMAND_LONG) message. Command plugin emits COMMAND_LONG or COMMAND_INT and receives [...]

I'm now trying to implement a plugin to read specific COMMAND_INT messages. However, there is any guide for this?
And in general, do I need to compile the mavros repo from scratch with my plugin or I can write my plugin on another repo and import in mavros?

vooon, 2022-01-14:
Look at dummy.cpp, you just need to add handle method for message.

Pluginlib allow you to load a plugin from different packages. Check docs, it's just require .so, mavros_plugin.xml and special entry in package.xml.

tadteo, 2022-01-28:
In the end, I have implemented a new message type and the plugin related.
Now it is working.

closed 2022-01-28 by tadteo