mavlink/mavros #1665: fake_gps not affecting global_position, making offboard control impossible

michaelb-sm, 2021-12-01:
### Issue details

I am attempting to fly indoor in a GPS denied environment, and so am trying to use fake_gps to allow me to enter offboard mode. Currently I do not have a mocap/vision system available so I am publishing a static pose estimate to the /mavros/fake_gps/mocap/pose topic from a custom node. However even with this node running in the background, I don't see any data being published to /mavros/global_position/global or /mavros/hil/gps, and thus am unable to either set my home position or enter offboard mode.

I am using a Pixhawk 4 Mini with Jetson Nano 2GB as a companion computer, and QGroundControl to configure the Mini. I have MAV_USEHILGPS set to 1 in QGroundControl and am receiving data on GPS_RAW_INT in Mavlink Inspector, so I know that the Mini is receiving data of some kind. Connection to the Mini is established over the Jetson's GPIO pins, and I launch mavros with the following:

```
roslaunch mavros px4.launch fcu_url:=/dev/ttyTHS1:921600
```

### fake_gps plugin in px4_config.yaml

```
# fake_gps
fake_gps:
  # select data source
  use_mocap: true         # ~mocap/pose
  mocap_transform: false   # ~mocap/tf instead of pose
  use_vision: false       # ~vision (pose)
  use_hil_gps: true
  # origin (default: Zürich)
  geo_origin:
    lat: 43.663360          # latitude [degrees]
    lon: -79.393587           # longitude [degrees]
    alt: 97.336            # altitude (height over the WGS-84 ellipsoid) [meters]
  eph: 2.0
  epv: 2.0
  satellites_visible: 5   # virtual number of visible satellites
  fix_type: 3             # type of GPS fix (default: 3D)
  tf:
    listen: false
    send: false           # send TF?
    frame_id: "map"       # TF frame_id
    child_frame_id: "fix" # TF child_frame_id
    rate_limit: 10.0      # TF rate
  gps_rate: 5.0           # GPS data publishing rate
```

### Published message to /mavros/fake_gps/mocap/pose

```
header: 
  seq: 46
  stamp: 
    secs: 1638391613
    nsecs: 321658931
  frame_id: "map"
pose: 
  position: 
    x: 0.0
    y: 0.0
    z: 0.0
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 0.0
```

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

[... 483 lines cut ...]
        value: "9.919490"
      - 
        key: "Last RTT (ms)"
        value: "6.540828"
      - 
        key: "Mean RTT (ms)"
        value: "8.352056"
      - 
        key: "Last remote time (s)"
        value: "2094.117235000"
      - 
        key: "Estimated time offset (s)"
        value: "1638393984.283438683"
```

### Check ID

```
OK. I got messages from 1:1.

---
Received 5453 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 8, 140, 141, 147, 22, 30, 31, 36, 44, 74, 331, 77, 83, 230, 105, 111, 241, 245
```

michaelb-sm, 2021-12-02:
Edit: Added readout from MAVlink Inspector

vooon, 2021-12-03:
I do not see any trouble on mavros side. Probably you need to change some firmware parameters to select mocap position source.

michaelb-sm, 2021-12-03:
I'm inclined to agree, but I'm not sure which parameters in QGroundControl would help that I haven't already changed. Since I don't have a GPS module connected, GPS_1_CONFIG doesn't have any obvious port to be assigned. If I set it to TELEM1 (since technically I'm sending the GPS data over that port) then it blocks communication from mavros over that port. Outside of MAV_ or GPS_ parameters, of which I have looked into, I'm unaware of any other parameters which would specifically grab this data (since it's supposed to be read as GPS data using fake_gps).

I know EKF2 is used for state estimation on the Mini, but i'm not very familiar with it. Is it possible that it's having trouble reading this source?

vooon, 2021-12-03:
Sorry, but i never tried to setup it. So i mostly guessing. Perhaps you could find something useful here: https://docs.px4.io/v1.12/en/ros/external_position_estimation.html

Just one common thing to check: do you update header.stamp?

michaelb-sm, 2021-12-03:
Thanks for the resource, looking into it now but I'm not sure if this will properly substitute a GPS signal. 

I am updating header.stamp with ros::Time::now() every time it's published

bresch, 2021-12-16:
Hi @michaelb-sm ,

Not sure if this is the only issue blocking you, but the data sent by your fake_gps doesn't pass the EKF requirements. This is a reason why it is not using the data (e.g.: number of satellites is 5 while the min specified by the parameter is 6 by default, see https://docs.px4.io/master/en/advanced_config/tuning_the_ecl_ekf.html#gnss-performance-requirements).
So you can change the data you send using your fake GPS or relax the EKF requirements (`EKF2_REQ_xxx`) or disable some checks using `EKF2_GPS_CHECK`.

closed 2026-10-29 by github-actions[bot]