mavlink/mavros #1673: Not receving any data by /mavros/imu/data_raw

bhaskar-glitch, 2021-12-10:
Hi,
I'm using pixhawk 2.4.8 with px4 latest firmware now while launching the px4.launch file I'm not getting any data with topic **/mavros/imu/data_raw**(rostopic echo /mavros/imu/data_raw) although when I echo the **/mavros/imu/data** then it's showing the value of gyro only not the accelero and I've use the data of both gyro and accelro to sync and calibrate my camera with it..PLease help

vooon, 2021-12-10:
Please follow https://github.com/mavlink/mavros/blob/ros2/.github/ISSUE_TEMPLATE because without logs and diag only fortune teller can guess what's wrong in your setup.

bhaskar-glitch, 2021-12-10:
> Please follow https://github.com/mavlink/mavros/blob/ros2/.github/ISSUE_TEMPLATE because without logs and diag only fortune teller can guess what's wrong in your setup.

**ISSUE:**
Hi,
I'm using pixhawk 2.4.8 with px4 latest firmware now while launching the px4.launch file I'm not getting any data with topic /mavros/imu/data_raw(rostopic echo /mavros/imu/data_raw) although when I echo the /mavros/imu/data then it's showing the value of gyro only not the accelero and I've use the data of both gyro and accelro to sync and calibrate my camera with it..

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Melodic?
Ubuntu: ?Jetson xavier 4.6?

### Autopilot type and version
[  ] ArduPilot
[X] PX4

Version: ?3.7.1?
### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```
header: 
  seq: 106
  stamp: 
    secs: 1639149305
    nsecs: 590284252
  frame_id: ''
status: 
  - 
    level: 1
    name: "camera/realsense2_camera_manager_infra2: Frequency Status"
    message: "Frequency too low."
    hardware_id: "923322071722"
    values: 
      - 
        key: "Events in window"
        value: "66"
      - 
        key: "Events since startup"
        value: "628"
      - 
        key: "Duration of window (s)"
        value: "5.372158"
      - 
        key: "Actual frequency (Hz)"
        value: "12.285567"
      - 
        key: "Target frequency (Hz)"
        value: "60.000000"
      - 
        key: "Minimum acceptable frequency (Hz)"
        value: "54.000000"
      - 
        key: "Maximum acceptable frequency (Hz)"
        value: "66.000000"
---

### Check ID

```
rosrun mavros checkid
```
OK. I got messages from 1:1.

---
Received 362 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 36, 230, 147, 32, 8, 74, 77, 111, 141, 241, 83, 245, 4, 30

vooon, 2021-12-11:
Sorry, but you catch diagnostics for another node, a realsense one, not mavros. And there still no logs.

lidorshimoni, 2022-08-07:
My problem was in the DDS QoS profile. I did not see any traffic in rqt but in ros2 topic echo it worked

deltaRing, 2023-04-28:
have you set rosrun mavros mavsys rate --all 100 ? I followed the offcial website and get all imu data well. https://ardupilot.org/dev/docs/ros-data-from-ap.html

RohithReddy-1402, 2026-01-20:
i am not even getting data from topics /imu_raw or /imu , what can be the reason . i have connected pixhawk directly to laptop via serial connection(usb) .

closed 2026-10-28 by github-actions[bot]