mavlink/mavros #1683: Invalid topic names

mtesseracttech, 2021-12-20:
### Issue details

When trying to initialize mavros, while mavros_extras is installed (as we want to use the fake_gps plugin), it encounters invalid topic names (we see one, but there could be more):

### MAVROS version and platform

Mavros: 2.0.5-1focal.20211130.103712
ROS: Foxy
Ubuntu: 20.04

### Steps to reproduce

apt install ros-foxy-mavros ros-foxy-mavros-extras

apm_config.yaml (we tried to exclude packages that might cause problems, to no avail): 
```
mavros:
  ros__parameters: {}
mavros_router:
  ros__parameters: {}
mavros_node:
  ros__parameters:	
    fcu_url: serial:///dev/ttyTHS1:57600
```

Run `ros2 run mavros mavros_node --ros-args --params-file ~ros2_ws/src/motor_control/launch/apm_config.yaml`

### Logs
```
[INFO] [1640020333.809791788] [mavros_node]: Starting mavros_node container
[INFO] [1640020333.810050492] [mavros_node]: FCU URL: serial:///dev/ttyTHS1:57600
[INFO] [1640020333.810098097] [mavros_node]: GCS URL: 
[INFO] [1640020333.810134504] [mavros_node]: UAS Prefix: /uas1
[INFO] [1640020333.810168151] [mavros_node]: Starting mavros router node
[INFO] [1640020333.820001701] [mavros_router]: Built-in SIMD instructions: ARM NEON
[INFO] [1640020333.820105662] [mavros_router]: Built-in MAVLink package version: 2021.11.11
[INFO] [1640020333.820163944] [mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV all development icarous matrixpilot paparazzi standard uAvionix ualberta
[INFO] [1640020333.820205612] [mavros_router]: MAVROS Router started
[INFO] [1640020333.820301135] [mavros_router]: Requested to add endpoint: type: 0, url: serial:///dev/ttyTHS1:57600
[INFO] [1640020333.820420929] [mavros_router]: Endpoint link[1000] created
[INFO] [1640020333.822104039] [mavros_router]: link[1000] opened successfully
[INFO] [1640020333.822283418] [mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[INFO] [1640020333.822376910] [mavros_router]: Endpoint link[1001] created
[INFO] [1640020333.824025071] [mavros_router]: link[1001] opened successfully
[INFO] [1640020333.824182731] [mavros_node]: Starting mavros uas node
[INFO] [1640020333.998091206] [mavros]: UAS Executor started
[INFO] [1640020334.049880370] [mavros]: Plugin actuator_control created
[INFO] [1640020334.050032352] [mavros]: Plugin actuator_control added to executor
[INFO] [1640020334.050116365] [mavros]: Plugin actuator_control initialized
[INFO] [1640020334.101630783] [mavros]: Plugin adsb created
[INFO] [1640020334.101754171] [mavros]: Plugin adsb added to executor
[INFO] [1640020334.101828652] [mavros]: Plugin adsb initialized
[INFO] [1640020334.112102473] [mavros]: Plugin altitude created
[INFO] [1640020334.112224350] [mavros]: Plugin altitude added to executor
[INFO] [1640020334.112300654] [mavros]: Plugin altitude initialized
[INFO] [1640020334.125816421] [mavros]: Plugin cam_imu_sync created
[INFO] [1640020334.125955747] [mavros]: Plugin cam_imu_sync added to executor
[INFO] [1640020334.126041583] [mavros]: Plugin cam_imu_sync initialized
terminate called after throwing an instance of 'rclcpp::exceptions::InvalidTopicNameError'
  what():  Invalid topic name: topic name must not have a tilde '~' that is not followed by a forward slash '/':
  '~image_captured'
    ^
```

vooon, 2021-12-20:
Bug fixed in current head. Please use source install as ros2 port far from complete.

vooon, 2021-12-20:
Fix: https://github.com/mavlink/mavros/commit/58296798ee98f069d02c324adf6fbf85ba8d9115

closed 2021-12-20 by vooon