mavlink/mavros #1684: Issue when publishing to topic

smokranova, 2021-12-21:
### Issue details

When running `ros2 run mavros mavros_node` and then publishing to the topic `/mavros/fake_gps/mocap/tf` with `ros2 topic pub /mavros/fake_gps/mocap/tf geometry_msgs/msg/TransformStamped "{}"` the mavros_node crashes and prints `/path/to/ros2_ws/install/mavros/lib/mavros/mavros_node: symbol lookup error: /path/to/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZN3tf27fromMsgIN13geometry_msgs3msg10Transform_ISaIvEEEN5Eigen9TransformIdLi3ELi2ELi0EEEEEvRKT_RT0_` (you can see it in the node logs below as well). 

### MAVROS version and platform

Mavros: 2.0.5
ROS: Foxy
Ubuntu: 20.04

### Autopilot type and version

[X] ArduPilot

### Node logs

```
[INFO] [1640108289.639148009] [mavros_node]: Starting mavros_node container
[INFO] [1640108289.640191186] [mavros_node]: FCU URL: 
[INFO] [1640108289.640298038] [mavros_node]: GCS URL: 
[INFO] [1640108289.640312299] [mavros_node]: UAS Prefix: /uas1
[INFO] [1640108289.640321851] [mavros_node]: Starting mavros router node
[INFO] [1640108289.673971611] [mavros_router]: Built-in SIMD instructions: SSE, SSE2
[INFO] [1640108289.674252463] [mavros_router]: Built-in MAVLink package version: 2021.12.12
[INFO] [1640108289.674270025] [mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV all development icarous matrixpilot paparazzi standard uAvionix ualberta
[INFO] [1640108289.674280531] [mavros_router]: MAVROS Router started
[INFO] [1640108289.674433133] [mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[INFO] [1640108289.674812693] [mavros_router]: Endpoint link[1000] created
[INFO] [1640108289.677846828] [mavros_router]: link[1000] opened successfully
[INFO] [1640108289.678352667] [mavros_node]: Starting mavros uas node
[INFO] [1640108289.868841295] [mavros]: UAS Executor started
[INFO] [1640108289.938348793] [mavros]: Plugin actuator_control created
[INFO] [1640108289.939383139] [mavros]: Plugin actuator_control added to executor
[INFO] [1640108289.939587302] [mavros]: Plugin actuator_control initialized
[INFO] [1640108290.004855475] [mavros]: Plugin adsb created
[INFO] [1640108290.006595873] [mavros]: Plugin adsb added to executor
[INFO] [1640108290.006803623] [mavros]: Plugin adsb initialized
[INFO] [1640108290.068781495] [mavros]: Plugin altitude created
[INFO] [1640108290.070317723] [mavros]: Plugin altitude added to executor
[INFO] [1640108290.070682138] [mavros]: Plugin altitude initialized
[INFO] [1640108290.103444108] [mavros]: Plugin cam_imu_sync created
[INFO] [1640108290.105389448] [mavros]: Plugin cam_imu_sync added to executor
[INFO] [1640108290.105516531] [mavros]: Plugin cam_imu_sync initialized
[INFO] [1640108290.145918432] [mavros]: Plugin camera created
[INFO] [1640108290.146455053] [mavros]: Plugin camera added to executor
[INFO] [1640108290.146604771] [mavros]: Plugin camera initialized
[INFO] [1640108290.259346653] [mavros]: Plugin command created
[INFO] [1640108290.259774373] [mavros]: Plugin command added to executor
[INFO] [1640108290.259994000] [mavros]: Plugin command initialized
[INFO] [1640108290.296456956] [mavros]: Plugin companion_process_status created
[INFO] [1640108290.296769556] [mavros]: Plugin companion_process_status added to executor
[INFO] [1640108290.297451101] [mavros]: Plugin companion_process_status initialized
[INFO] [1640108290.355990482] [mavros]: Plugin debug_value created
[INFO] [1640108290.357817271] [mavros]: Plugin debug_value added to executor
[INFO] [1640108290.363665542] [mavros]: Plugin debug_value initialized
[INFO] [1640108290.407396561] [mavros.distance_sensor]: DS: Plugin not configured!
[INFO] [1640108290.408741238] [mavros]: Plugin distance_sensor created
[INFO] [1640108290.410084585] [mavros]: Plugin distance_sensor added to executor
[INFO] [1640108290.410335411] [mavros]: Plugin distance_sensor initialized
[INFO] [1640108290.494295503] [mavros]: Plugin esc_status created
[INFO] [1640108290.494997640] [mavros]: Plugin esc_status added to executor
[INFO] [1640108290.495249964] [mavros]: Plugin esc_status initialized
[INFO] [1640108290.534895948] [mavros]: Plugin esc_telemetry created
[INFO] [1640108290.536222544] [mavros]: Plugin esc_telemetry added to executor
[INFO] [1640108290.536602523] [mavros]: Plugin esc_telemetry initialized
[INFO] [1640108290.618538841] [mavros]: Plugin fake_gps created
[INFO] [1640108290.618809402] [mavros]: Plugin fake_gps added to executor
[INFO] [1640108290.621841029] [mavros]: Plugin fake_gps initialized
[INFO] [1640108290.717256155] [mavros]: Plugin ftp created
[INFO] [1640108290.718065705] [mavros]: Plugin ftp added to executor
[... 113 lines cut ...]
[INFO] [1640108295.067284812] [mavros]: Plugin vibration created
[INFO] [1640108295.067948454] [mavros]: Plugin vibration added to executor
[INFO] [1640108295.070545712] [mavros]: Plugin vibration initialized
[INFO] [1640108295.208849422] [mavros]: Plugin vision_pose created
[INFO] [1640108295.209460253] [mavros]: Plugin vision_pose added to executor
[INFO] [1640108295.211558621] [mavros]: Plugin vision_pose initialized
[INFO] [1640108295.419875540] [mavros]: Plugin vision_speed created
[INFO] [1640108295.420188727] [mavros]: Plugin vision_speed added to executor
[INFO] [1640108295.429914825] [mavros]: Plugin vision_speed initialized
[INFO] [1640108295.610498091] [mavros]: Plugin waypoint created
[INFO] [1640108295.611668587] [mavros]: Plugin waypoint added to executor
[INFO] [1640108295.611971859] [mavros]: Plugin waypoint initialized
[INFO] [1640108295.786260177] [mavros]: Plugin wheel_odomotry created
[INFO] [1640108295.788097997] [mavros]: Plugin wheel_odomotry added to executor
[INFO] [1640108295.788369972] [mavros]: Plugin wheel_odomotry initialized
[INFO] [1640108295.915061236] [mavros]: Plugin wind_estimation created
[INFO] [1640108295.915846357] [mavros]: Plugin wind_estimation added to executor
[INFO] [1640108295.922115084] [mavros]: Plugin wind_estimation initialized
[INFO] [1640108295.933549163] [mavros]: Built-in SIMD instructions: SSE, SSE2
[INFO] [1640108295.933757289] [mavros]: Built-in MAVLink package version: 2021.12.12
[INFO] [1640108295.933854964] [mavros]: Known MAVLink dialects: common ardupilotmega ASLUAV all development icarous matrixpilot paparazzi standard uAvionix ualberta
[INFO] [1640108295.933939645] [mavros]: MAVROS UAS via /uas1 started. MY ID 1.191, TARGET ID 1.1
/path/to/ros2_ws/install/mavros/lib/mavros/mavros_node: symbol lookup error: /path/to/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZN3tf27fromMsgIN13geometry_msgs3msg10Transform_ISaIvEEEN5Eigen9TransformIdLi3ELi2ELi0EEEEEvRKT_RT0_

```

vooon, 2021-12-22:
Looks like i forgot to add some dependency to extras package. Seems it's for `tf2_ros::fromMsg()`.
Could you please use source install? ROS2 port still alpha, and more importantly we still having #1588.

smokranova, 2021-12-22:
I did source install, which is when the issue appears.

vooon, 2022-01-17:
@smokranova i've checked cmake, dependencies are set. Will try to reproduce later today.

closed 2026-10-28 by github-actions[bot]