mavlink/mavros #1685: Low publish rate on Mavros topics

bhaskar-glitch, 2022-01-04:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hii,
I'm working on a indoor autonomous drone project and recently I found an issue in
the topics of MAVROS because the topics are being updated in a very low rate (at 1Hz) the topics i've noticed of having low update rate are: /mavros/local_position/odom , /mavros/local_position/pose
Now let me tell you the full procedure i'm doing...I'm using VINS_FUSION as odometry with d453i camera and using VIO(realsense bridge)to connect VINS and MAVROS(FCU)now the issue is as the drone doesn't know its correct altitude of the real time coz of the slow update rate i'm getting some sort of disturbance in my autonomous flight also i'm using JETSON XAVIER on edge of drone to process everything.

### MAVROS version and platform

Mavros: ?latest?
ROS: ?melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: ?latest?

### Node logs

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /copter_visualization/child_frame_id: fcu
 * /copter_visualization/fixed_frame_id: map
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
[... 358 lines cut ...]
        value: "2.107066"
      - 
        key: "Mean RTT (ms)"
        value: "1.977769"
      - 
        key: "Last remote time (s)"
        value: "10855.806426000"
      - 
        key: "Estimated time offset (s)"
        value: "1641277360.128325224"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 592 messages, from 1 addresses
sys:comp   list of messages
  1:1     32, 1, 36, 230, 0, 8, 74, 141, 77, 111, 241, 83, 245, 4, 30, 147

```

vooon, 2022-01-09:
Did you try to use set_stream_rate / set_message_interval?

bhaskar-glitch, 2022-01-10:
> Did you try to use set_stream_rate / set_message_interval?

Yah i have tried using stream rate but nothing helped...(rosservice call /mavros/set_stream_rate 1 200 1)

vooon, 2022-01-10:
https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/mavlink/mavlink_receiver.h

Seems PX4 doesn't support set_stream_rate anymore, could you please try `rosrun mavros mavsys message_interval --id=31 --rate=200` ([ATTITUDE_QUATERNION=31](https://mavlink.io/en/messages/common.html#ATTITUDE_QUATERNION), [HIGHRES_IMU=105](https://mavlink.io/en/messages/common.html#HIGHRES_IMU)).

bhaskar-glitch, 2022-01-11:
> 

I tried this but their is no effect on the publish rate yet..It's 1hz as before..
Please help
[image]

bhaskar-glitch, 2022-01-11:
Hey thanks for help!
**rosrun mavros mavsys message_interval --id=32 --rate=200**
solved my issue :)))))

bhaskar-glitch, 2022-04-14:
Hi @vooon I facing the same issue again!! But this time the topic /mavros/local_positon/odom and pose is working fine on 100hz but in QGC mavlink inspector the rate is again very slow.. it's 1hz there.
Please help me out!!

mengchaoheng, 2023-04-22:
@bhaskar-glitch @vooon can you help me please! https://github.com/PX4/PX4-Autopilot/issues/21504

ZhongmouLi, 2023-11-07:
maybe you can try 
`rosrun mavros mavsys rate --all 100`

closed 2026-10-28 by github-actions[bot]