mavlink/mavros #1686: Can't connect through Router

WARGRaph, 2022-01-04:
### Issue details
Problem: Mavlink Connection through router doesn't work/only works partially. 

Setup: Ubuntu-PC (running Docker) -> Router -> OpenHD -> Ardupilot (4.1.2)
Replacing the Ubuntu-PC with a Windows laptop and running Mission Planner, the connection works. 
Replacing the Router with a straight wire, the connection works
Thus, I suspect the problem to lie somewhere in mavros or my configuration of it. 
Connection URL: 
`<arg name="fcu_url" default="udp://:14550@192.168.1.238" />`

### MAVROS version and platform

Mavros: 1.8.0-1bionic.20210708.003326
ROS: Melodic
Ubuntu: Ubuntu 18.04.5 LTS

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: 4.1.2

### Node logs

```
root@server:/control_center_ws# roslaunch drone_control drone_control.launch
... logging to /root/.ros/log/8ebcf2a6-6d5b-11ec-82dd-b42e99464746/roslaunch-server-28715.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://server:37991/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /flight_params/mission_alt: 15.0
 * /flight_params/mission_speed: 5.0
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.1
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.1
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
[... 443 lines cut ...]
      - 
        key: "Last RTT (ms)"
        value: "44.548126"
      - 
        key: "Mean RTT (ms)"
        value: "40.975510"
      - 
        key: "Last remote time (s)"
        value: "2338.918634000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
```

### Check ID

```
root@wargdronesserver:/control_center_ws# rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 738 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 44, 77, 111, 22, 253
```

vooon, 2022-01-04:
But you actually got connection:
```
[ INFO] [/mavros]: udp0: Remote address: 192.168.1.238:37443
[ INFO] [/mavros]: CON: Got HEARTBEAT, connected. FCU: ArduPilot
```

WARGRaph, 2022-01-05:
Oops, meant to close the ticket yesterday:
The problem was that there was a connection, but no telemetry came from the drone to the ground. Switching out the drone fixed the problem, why one drone was sending telemetry data and the other not is yet to be determined.

closed 2022-01-05 by WARGRaph