mavlink/mavros #1691: Python3 support

PeymanAmirii, 2022-01-10:
Hi,
Does MAVROS officially support and be able to work with Python 3 correctly or we have to use just Python 2 to work safely with MAVROS?
Thanks

vooon, 2022-01-10:
That depends on what version do you use. Version 1.12 on Noetic should be mostly fine, but there are some fixes in current master.
Anyway if you're going to use ROS API, then it is enough to use ROS Noetic. Main node is written in C++, so py3 mostly comes to rospy support.

PeymanAmirii, 2022-01-10:
Thanks for your quick reply. We want to use MAVROS APIs for PX4. We know that ROS Noetic is ROS1, but it is recommended to use ROS2 for PX4. If we use ROS2, will MAVROS work with Python 3 (3.7 or 3.8) correctly?
Many Thanks

vooon, 2022-01-11:
ROS1 vs ROS2 is completely different story. ROS2 has huge overhaul of all things. MAVROS v2 (version for ROS2) is on it's alpha stage and still having some critical bugs. I cannot recommend to use it now.

PX4 has their own bridge node to ROS2, which gives access to uORB, an internal pubsub system of the firmware.
That's counterpart to mavros, which talks to mavlink app, which then interacts with uORB.

PeymanAmirii, 2022-01-11:
Thanks a bunch. So we can use ROS2 for PX4 without using MAVROS?

vooon, 2022-01-12:
Yes, you can.

closed 2022-01-12 by vooon