mavlink/mavros #1692: ArUco marker pose estimation & following

ParkYoungHoo1, 2022-01-11:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[ O ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

Hello, I want to use ROS to estimate the pose of a ArUco marker using camera and make a drone which follows the marker.
Tracking for the position is performed well using a /mavros/landing_target/raw, but I can not perform pose tracking...
I can know about all of the marker's information(position, orientation, id etc..)
Can you recommend a mavros node that I can use ??

Thank you!

vooon, 2022-01-11:
Try to use https://github.com/CopterExpress/clover , as i remember they has marker tracking as one of their student training.

ParkYoungHoo1, 2022-01-11:
OMG! 
It must be helpful to me absolutely.
Thank you!

closed 2022-01-12 by vooon