mavlink/mavros #1693: MAVlink CMD receive in MAVROS

LightningPORTO, 2022-01-14:
----

### Issue details

Hello,

I'm currently developing a basic implementation of the MAVlink camera protocol [here](https://mavlink.io/en/services/camera.html) as a plugin for MAVROS.
I have managed to receive incoming msgs, however, I don't know how to receive MAVink commands.

Are there any examples I can get? I haven't found any. 

Thank you in advance

### MAVROS version and platform

Mavros: 0.18.4
ROS: NOETIC
Ubuntu: 20.04

### Autopilot type and version

[x] ArduPilot
[  ] PX4

vooon, 2022-01-14:
To send commands you should use command plugin services, so you just need to create service client and perform a call.

https://github.com/mavlink/mavros/blob/ed762be1ea6f53e4702f49bd4e22644f6347040f/mavros/src/plugins/sys_status.cpp#L1184-L1211

LightningPORTO, 2022-01-17:
Hello Vooon, thanks for your answer.

I am familiar with the sending protocol that MAVROS has implemented, I would like to receive a MAVlink command, not send.

Are there any additional examples you can provide ? Thank you again.

vooon, 2022-01-17:
Then simply make handler and provide it to get_subscriptions():
```
  Subscriptions get_subscriptions() override
  {
    return {
      make_handler(&CustomPlugin::handle_command_long)
    };
  }
```

LightningPORTO, 2022-01-21:
Thanks for the help, I will attempt the implementation.

Closing the issue, for now.

closed 2022-01-21 by LightningPORTO

LightningPORTO, 2022-01-21:
@vooon 

I'm sorry for reopening this, but, following your example on the handler, can you please give me an example of the callback function? 

Any callback example that receives a MAVlink command would do.

Thank you

vooon, 2022-01-21:
```cpp
void handle_command_long(const mavlink::mavlink_message_t *msg, mavlink::common::msg::COMMAND_LONG &cmd)
{
    if (!m_uas->is_my_target(cmd))
        return;

    if (cmd.command != 123) {
        // not my command
        retrun;
    }

    // do stuff
}
```