mavlink/mavros #1694: Mavros publishing topics out of sync and with ridiculously high fake publishing rates (thousands of Hz)

evbernardes, 2022-01-14:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I managed to install Mavros both with the binaries and the recent github version on my ROS Noetic system, and I've been having the following problems (that might or not be related):

The messages seem to be completely out of sync. I scripted a pretty simple subscriber that reads `mavros/state/armed` bool and plots either "armed" or "unarmed" only when the state changes: 

`def callback_status(data):
	global armed
	if armed != data.armed:
		if data.armed:
			rospy.loginfo("Armed")
		else:
			rospy.loginfo("Disarmed")
	armed = data.armed`

However, when I arm the vehicle, instead of only showing one "armed", it quickly goes back and forth between armed and unarmed:
`[INFO] [1642170285.023291]: Disarmed
[INFO] [1642170285.032543]: Armed
[INFO] [1642170285.041282]: Disarmed
[INFO] [1642170285.052351]: Armed
[INFO] [1642170285.060567]: Disarmed
[INFO] [1642170285.067647]: Armed
[INFO] [1642170285.074055]: Disarmed
[INFO] [1642170285.081127]: Armed
[INFO] [1642170285.086995]: Disarmed
[INFO] [1642170285.094041]: Armed
[INFO] [1642170285.100504]: Disarmed
[INFO] [1642170285.108004]: Armed
[INFO] [1642170285.114542]: Disarmed
[INFO] [1642170285.122030]: Armed
[INFO] [1642170285.130380]: Disarmed
[INFO] [1642170285.141082]: Armed
[INFO] [1642170285.146716]: Disarmed
[INFO] [1642170285.154045]: Armed
[INFO] [1642170285.160301]: Disarmed
[INFO] [1642170285.168443]: Armed
[INFO] [1642170285.174759]: Disarmed
[INFO] [1642170285.182258]: Armed
[INFO] [1642170285.188843]: Disarmed
[INFO] [1642170285.197310]: Armed
[INFO] [1642170285.203860]: Disarmed
[INFO] [1642170285.210826]: Armed
[INFO] [1642170285.216850]: Disarmed
[INFO] [1642170285.223525]: Armed
[INFO] [1642170285.231777]: Disarmed
[INFO] [1642170285.242224]: Armed
[INFO] [1642170285.250873]: Disarmed
[INFO] [1642170285.259866]: Armed
[INFO] [1642170285.265799]: Disarmed
[INFO] [1642170285.272762]: Armed
[INFO] [1642170285.278491]: Disarmed
`
I then checked the same topic on RQT and I can see it is indeed going back and forth. 

Another thing caught my eye though: The publish frequencies of the topics are all over the place. Some topics show a 2000Hz publish frequency, some 1000Hz, some 8000Hz and all of this oscillating very fast. Does anyone else have the same problem?

I also noticed a second possibly related thing. When I ran mavros, I got multiple warnings like this:
`[ WARN] [1642170493.138885006]: TM : RTT too high for timesync: 6213.76 ms.
[ WARN] [1642170493.139604077]: TM : RTT too high for timesync: 7814.40 ms.`

So I changed the ratesync parameter to 0 and it stopped. The "arm" sync and frequency problems described earlier all continued to happen though.

### MAVROS version and platform

Mavros: both binaries from Ubuntu and recent git version
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

vooon, 2022-01-14:
Rx speed and HEARTBEAT at 511 Hz looks too high.
Could you please add a log?

evbernardes, 2022-01-16:
> Rx speed and HEARTBEAT at 511 Hz looks too high. Could you please add a log?

Thanks for your input, do you mean the output from the console? Or the log file from ROS?

vooon, 2022-01-16:
Output from console to see startup and few plugin log lines like RTT.

evbernardes, 2022-01-17:
Just put a console log, I let the timesync with its default value so you can also see the high RTT warnings. 

Also, note how at the end it starts to loop these messages (this happens even when opting out of timesync):
```
[ INFO] [1642410819.431777032]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1642410819.431821560]: VER: 1.1: Flight software:     010d0000 (e835a7c4ea000000)
[ INFO] [1642410819.431836038]: VER: 1.1: Middleware software: 010d0000 (e835a7c4ea000000)
[ INFO] [1642410819.431853150]: VER: 1.1: OS software:         080200ff (0a5f0d0b6cd238ad)
[ INFO] [1642410819.431866824]: VER: 1.1: Board hardware:      00000012
[ INFO] [1642410819.431880834]: VER: 1.1: VID/PID:             26ac:0012
[ INFO] [1642410819.431900715]: VER: 1.1: UID:                 3039510f36373532
```

vooon, 2022-01-17:
```
[ INFO] [1642410810.592782176]: FCU URL: udp://:14550@192.168.0.101:14557
[ INFO] [1642410810.594874419]: udp0: Bind address: 0.0.0.0:14550
[ INFO] [1642410810.594987261]: udp0: Remote address: 192.168.0.101:14557
[ INFO] [1642410810.595183273]: GCS URL: udp://@localhost
[ INFO] [1642410810.595313995]: udp1: Bind address: 0.0.0.0:14555
[ INFO] [1642410810.595431697]: udp1: Remote address: 127.0.0.1:14550
```

Look at the ports. You've made a loop.

evbernardes, 2022-01-17:
The only address I set was 192.168.0.101 in the launch file though (the Pixracer wifi module's IP on my network). These are the original lines:
```
<arg name="fcu_url" default="/dev/ttyACM0:57600" />
<arg name="gcs_url" default="" />
```
And I changed both the fcu to the module's IP adress and the gcs to my computer's IP address. 

Do I have to change something else in this case? What should I change these bind and remote adresses to (and where exactly)?

Also, not that I do have some data coming from MAVROS, they just seem to be completely out of sync.

vooon, 2022-01-17:
Ok, i'll explain what's going on: by setting `udp://:14550@192.168.0.101:14557` you told to bind on IPV4_ANY port 14550. Then by setting `udp://@localhost` you told system to send to localhost:14550. It's address pair listened by fcu port, so when first message come from real FCU it then was copied to GCS link, which sends it to FCU link again. That's infinite loop, which results in high rates.

Moreover udp link automatically switches to sender of each message, that's kind of autodetection, which also makes things worse in that case (mavlink-router have better solution...).

So i suppose to use fcu url `udp://:14540@192.168.0.101:14557` and gcs url just `udp://@`.
GCSes often tries to connect to default port on localhost, so that should be fine.

evbernardes, 2022-01-17:
Oh right, I think I see it now! Thanks for your help, I'll check that tomorrow!

evbernardes, 2022-01-18:
Changing only gcs url to `udp://@` makes it work, apparently! Thanks a lot for your help :)

closed 2022-01-18 by vooon