mavlink/mavros #1697: GPS_RTK using NTRIP

ryanyej, 2022-01-27:
when using NTRIP to get rtcm, then copy rtcm data to /mavros/gps_rtk/send_rtcm topic.
I am not sure to using rtcm data directly received from NTRIP caster 
or I need to process these rtcm data first ,then convert to /mavros/gps_rtk/send_rtcm?

vooon, 2022-01-27:
No, you don't have to parse RTCM packets, see [RTCM.msg](https://github.com/mavlink/mavros/blob/master/mavros_msgs/msg/RTCM.msg) - you just need to copy it to data[] and use send_rtcm.
Note that one RTCM packet may be fragmented up to 4 chunks of 180 byte long. So your message must be smaller than 720 bytes.

ryanyej, 2022-01-27:
Hi,@vooon  .Thank you for the answer. It helped a lot.
BTW, is there a requirement for the system receiving frequency?
Like how often do we need to send /mavros/gps_rtk/send_rtcm to get rtk float/fixed?

vooon, 2022-01-27:
I do not know, BTW don't forget to update header.stamp.

vooon, 2022-02-03:
@ryanyej any updates?

ryanyej, 2022-02-07:
It worked on my CUAV X7+.  However sometimes received RTCM packets may be larger than 720. 
I discarded the excess part.
For now , RTK status(5/6) seems good.
Thanks.

closed 2022-02-07 by ryanyej