mavlink/mavros #1699: Can not takeoff with mavros and can not arm in loiter (non-gps)

bilalkah, 2022-02-03:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: git tag 1.13.0
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: 4.1.3

### Node logs

```
... logging to /home/udoo/.ros/log/f72d26b8-84ca-11ec-bb83-04d9f5f0a80a/roslaunch-polocomp-udoo-8034.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://polocomp-udoo:37255/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
 * /mavros/fake_gps/satellites_visible: 6
 * /mavros/fake_gps/speed_accuracy: 0.0
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
[... 491 lines cut ...]
 * /roll_cam: 0
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /source_frame_id: /camera_link
 * /target_frame_id: /camera_odom_frame
 * /yaw_cam: 0

NODES
  /
    t265_to_mavros (vision_to_mavros/vision_to_mavros_node)

ROS_MASTER_URI=http://localhost:11311

process[t265_to_mavros-1]: started with pid [8252]
[ INFO] [1643876819.561642234]: Get target_frame_id parameter: /camera_odom_frame
[ INFO] [1643876819.567613605]: Get source_frame_id parameter: /camera_link
[ INFO] [1643876819.569104976]: Get output_rate parameter: 30.000000
[ INFO] [1643876819.570539324]: Get gamma_world parameter: -1.570796
[ INFO] [1643876819.572207576]: Get roll_cam parameter: 0.000000
[ INFO] [1643876819.573635712]: Get pitch_cam parameter: 0.000000
[ INFO] [1643876819.574943627]: Get yaw_cam parameter: 0.000000
[ INFO] [1643876819.576457434]: Precision landing disabled by default
```

I try to accomplish non-GPS navigation in indoor environment. I launch the mavros node, rs_t265.launch and vision_tf_to_mavros.launch, after I set EKF with Mission planner. I also configurated parameters for use realsense t265 camera and EKF3 source for external navgiation source. I also use hereflow optical sensor and configurate optical flow and rangefinder parameters as well. I can see change position and heading from miision planner. Also, I can see /mavros/vision_pose/pose and /mavros/local_position/pose are publishing. But I cannot see VISION_POSITION_ESTIMATE from mavlink inspector. Also, I can not arm copter while using loiter mode. It says bad gps position error but I already disabled gps and changed EKF3 sources. Also I can not takeoff with using /mavros/cmd/takeoff service. I put copter to GUIDED_NOGPS mode when I try to takeoff with mavros. Can  you help mw with this problem?

vooon, 2022-02-03:
First you need to Arm your drone. It won't takeoff without it.

VISION_POSITION_ESTIMATE sent to the copter, so you can't hear what mavros sends on gcs side. Only possibility to check - use wireshark, or mavlink-router between FCU and mavros.

bilalkah, 2022-02-03:
Thank you for your reply,
I armed the copter, below the source code.
When I used this source code takeoff request returned 1 but nothing happened. Also when I try /mavros/cmd/takeoff service it return false whether armed or not. Where is the problem?
```
#include <cstdlib>
#include <ros/ros.h>
#include <mavros_msgs/CommandBool.h>
#include <mavros_msgs/CommandTOL.h>
#include <mavros_msgs/SetMode.h>
#include <mavros_msgs/Waypoint.h>
#include <mavros_msgs/WaypointPush.h>
#include <geometry_msgs/PoseStamped.h>
#include <std_msgs/Header.h>
#include <geometry_msgs/Pose.h>
#include <geometry_msgs/Point.h>
#include <geometry_msgs/Quaternion.h>
#include <stdlib.h>

//Current position of the drone
double x_position;
double y_position;
double z_position;

void position_call_back(const geometry_msgs::PoseStamped::ConstPtr &msg){
    geometry_msgs::PoseStamped position = *msg;
    x_position = position.pose.position.x;
    y_position = position.pose.position.y;
    z_position = position.pose.position.z;
}

bool is_greater_then_threshold(geometry_msgs::PoseStamped destination){
	//Is drone close enough to the target
    double distance = sqrt(pow(destination.pose.position.x - x_position, 2) + pow(destination.pose.position.y - y_position, 2));
    return distance> 0.8; //0.8 is just a threshold
}

//function that make drone move by publishing destination.
void travel(geometry_msgs::PoseStamped destination, ros::Publisher local_pos_pub  ){
    while(is_greater_then_threshold(destination)){
        local_pos_pub.publish(destination);
        sleep(1);
    }
    return;
}

int main(int argc, char **argv){
    int rate = 30;
    ros::init(argc, argv, "ardu_takeoff");
    ros::NodeHandle n;
    ros::Rate r(rate);
    //start callbacks
    ros::AsyncSpinner spinner(1);
    spinner.start();

    /*********** GUIDED ***********/
    ros::ServiceClient cl = n.serviceClient<mavros_msgs::SetMode>("/mavros/set_mode");
    mavros_msgs::SetMode srv_setMode;
    srv_setMode.request.base_mode = 0;
    srv_setMode.request.custom_mode = "GUIDED_NOGPS";
    if(cl.call(srv_setMode)){
        ROS_INFO("setmode send ok %d value:", srv_setMode.response.mode_sent);
    }else{
        ROS_ERROR("Failed SetMode");
        return -1;
    }

    /***********ARM***************/
    ros::ServiceClient arming_cl = n.serviceClient<mavros_msgs::CommandBool>("/mavros/cmd/arming");
    mavros_msgs::CommandBool srv;
    srv.request.value = true;
    if(arming_cl.call(srv)){
        ROS_INFO("ARM send ok %d", srv.response.success);
    }else{
        ROS_ERROR("Failed arming or disarming");
    }
    sleep(2);
    /**********TAKE OFF***********/
    ros::ServiceClient takeoff_cl = n.serviceClient<mavros_msgs::CommandTOL>("/mavros/cmd/takeoff");
    mavros_msgs::CommandTOL srv_takeoff;
    srv_takeoff.request.altitude = 39;
    srv_takeoff.request.latitude = 0;
    srv_takeoff.request.longitude = 0;
    srv_takeoff.request.min_pitch = 0;
    srv_takeoff.request.yaw = 0;
    if(takeoff_cl.call(srv_takeoff)){
        ROS_INFO("srv_takeoff send ok %d", srv_takeoff.response.success);
    }else{
        ROS_ERROR("Failed Takeoff");
    }
    ROS_INFO("after takeoff");
    /************TASK****************/
    sleep(10);
    
    ros::Publisher local_pos_pub = n.advertise<geometry_msgs::PoseStamped>("mavros/setpoint_position/local", 1000);
    ros::Subscriber position_sub = n.subscribe<geometry_msgs::PoseStamped>("/mavros/local_position/pose", 10, position_call_back);
    
    // //Declare destination, change it when reached.
    // geometry_msgs::PoseStamped destination;
    // destination.pose.position.x = 3;
    // destination.pose.position.y = 0;
    // destination.pose.position.z = 10;
    // travel(destination, local_pos_pub);
	// //second point of the square 
    // destination.pose.position.x = 3;
    // destination.pose.position.y = 3;
    // destination.pose.position.z = 10;
    // travel(destination, local_pos_pub);
    // //third point of the square
    // destination.pose.position.x = 0;
    // destination.pose.position.y = 3;
    // destination.pose.position.z = 10;
    // travel(destination, local_pos_pub);
    // //final point 
    // destination.pose.position.x = 0;
    // destination.pose.position.y = 0;
    // destination.pose.position.z = 10;
    // travel(destination, local_pos_pub);
    // sleep(3);
    
    /***********LAND***********/
    ros::ServiceClient land_cl = n.serviceClient<mavros_msgs::CommandTOL>("/mavros/cmd/land");
    mavros_msgs::CommandTOL srv_land;
    srv_land.request.altitude = 1;
    srv_land.request.latitude = 0;
    srv_land.request.longitude = 0;
    srv_land.request.min_pitch = 0;
    srv_land.request.yaw = 0;
    if(land_cl.call(srv_land)){
        ROS_INFO("srv_land send ok %d", srv_land.response.success);
    }else{
        ROS_ERROR("Failed Land");
    }

    return 0;
}
```

bilalkah, 2022-02-04:
Here is the output of previous code. First drone is armed after 5 seconds it is disarmed.
[image]  

Here is the mavros output. 
[image]

Here is the /mavros/local_position/pose
[image]

Here is the /mavros/vision_pose/pose  
[image]

bilalkah, 2022-03-04:
Drone mod must be in GUIDED no GUIDED_NOGPS. I also ran set_origin.py in vision to mavros repo. I flied perfectly. Actualy it is really confusing GUIDED_NOGPS just for attitude controlling. When I try to do some indoor work, I would feeling that GUIDED_NOGPS must be the way to do it. Dont be misleading, you should use GUIDED mode...

closed 2022-03-04 by bilalkah

Nikhiljohnson21, 2023-08-30:
Hi i am also facing a similar problem like this, I'm trying to send commands via mavros to my pixhawk flight controller to control the drone. For now I'm testing with 1 motor.

Mavros is connected to pixhawk and I'm able to arm the motor
"rosrun mavros mavsafety arm"

Now I'm trying to take-off the drone using the command
"rosrun mavros mavcmd takeoff (with the yaw, pitch roll, latitude, longitude)"

But I'm getting "request failed. Check mavros logs. ACK:4"

I followed the steps instructed in mavros/utility commands documentation.