mavlink/mavros #1715: /mavros/local_positon/odom and pose giving wrong values

bhaskar-glitch, 2022-02-19:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hi, @vooon I'm facing a problem while using odometry for the position of the drone..I'm using realsense d435i, jetson xavier nx, and pixhawk 2.4.8 with px4 firmware....I'm using a bridge VIO to provide odometry data to px4..how the issue I'm facing right now is that bridge is sending data to /mavros/odometry/out and then odometry out is sending those data to mavros and further those data get into /mavros/local_position/odom and pose...but the position data (X,y,z) from the VINS(odometry)is correct till /mavros/odometry/out(topic) for after that when I'm doing rostopic echo of /mavros/local_positon/odom and pose it is giving some random wrong values...I don't know why because at /odometry/out topic values are very fine but after going to mavros node it gets wrong!! also, I changed all the parameters which is required for vision pose estimate like(EKF2_AID_MASK=24, EKF2_HGT_MODE=VISION and turned off the barometer)!!
Please help regarding the issue that why I'm getting that random wrong values.

### MAVROS version and platform

Mavros: ?latest?
ROS: ?Melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: ?latest?

### Node logs
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /copter_visualization/child_frame_id: fcu
 * /copter_visualization/fixed_frame_id: map
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
[... 356 lines cut ...]
        key: "Mean RTT (ms)"
        value: "1.977769"
      - 
        key: "Last remote time (s)"
        value: "10855.806426000"
      - 
        key: "Estimated time offset (s)"
        value: "1641277360.128325224"
---

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID
OK. I got messages from 1:1.

---
Received 592 messages, from 1 addresses
sys:comp   list of messages
  1:1     32, 1, 36, 230, 0, 8, 74, 141, 77, 111, 241, 83, 245, 4, 30, 147
```
rosrun mavros checkid
```

rishabhdevyadav, 2024-07-10:
Hi, anyone find the solution for this. I am also facing same issue.
I launched:
cd PX4-Autopilot
source ~/catkin_ws/devel/setup.bash
source Tools/simulation/gazebo-classic/setup_gazebo.bash

(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/simulation/gazebo-classic/sitl_gazebo-classic
roslaunch px4 mavros_posix_sitl.launch

And I can see the drone in gazebo, till here everything is completely normal. I am printing mavros/local_position/pose and after 4-5 seconds as soon as I am getting wrong z: position.
Quadrotor is on ground and it is printing
x: 0.011687700636684895
y: 0.004684464540332556
z: -1.1767728328704834

Slowly z is drifting slowly. However 'base_link' position is correct. Any solution ?
I am using docker. Ubuntu 20, Noetic.

bhaskar-glitch, 2024-07-11:
In our case we implemented it on hardware, so the drift was because of the vibration of imu, when we calibirated the camera and imu together the issue was resolved.

bhaskar-glitch, 2024-07-11:
> Hi, anyone find the solution for this. I am also facing same issue. I launched: cd PX4-Autopilot source ~/catkin_ws/devel/setup.bash source Tools/simulation/gazebo-classic/setup_gazebo.bash [...]

In our case we implemented it on hardware, so the drift was because of the vibration of imu, when we calibirated the camera and imu together the issue was resolved.

closed 2026-10-27 by github-actions[bot]