mavlink/mavros #1718: How to use mavros2

dzywater, 2022-03-01:
I don't how to use mavros2 and want to get some tutorials like "how to connect/arm/land/takeoff/mission/offboard".

Currently in PX4 gazebo simulation
(1) ros2 run mavros mavros_node --ros-args --param fcu_url:=udp://:14540@
       [ERROR] [1646107611.895603244] [mavros.odometry]: ODOM: Ex: Could not find a connection between 'map' and 'base_link_frd' because they are not part of the same tree.Tf has two or more unconnected trees.
(2) ros2 service call /mavros/cmd/arming mavros_msgs/srv/CommandBool '{value: True}'
      iris can arm.
(3)ros2 service call  /mavros/cmd/takeoff mavros_msgs/srv/CommandTOL '{min_pitch: 10, yaw: 10, latitude: 8, longitude: 47, altitude: 550}'
    iris don't takeoff.

vooon, 2022-03-01:
What version do you use? Try to build current ros2 branch, as it might have some important fixes.
TF error shouldn't affect takeoff, firmware do not aware about tf (but i'm unsure how gazebo simulates position).

Firmware usually tells something when it rejects to fly, so maybe you have some `FCU: ...` on (1)?

Rainerino, 2022-03-02:
Which flight controller are you using? Assume it's PX4 (since you are using iris from gazebo?), I managed to takeoff by 
1. send setpoint to one of the /mavros/setpoint nodes 
2. change into offbaord mode
3. it takes off 

I think for Ardupliot you have to use `CommandTOL` as you did, but I couldn't get it to take off either. 

I have that connection error too, would love to know what's causing it. My guess is it has something to do with the parameter file, which I am currently working on in #1564

dzywater, 2022-03-04:
I use v2.1.0

Mark-Beaty, 2022-03-10:
@dzywater 
**(1)** I believe that error is related to the mavros_extras package being installed which is still being worked on to be fully supported, I'm currently working on getting mavros_extras working in my system as I need some of the plugins to hopefully simplify my development on a project, but having it installed launches all the plugins by default which I'm working on adjusting currently (hopefully).

If you uninstall the mavros_extras package that will fix that error printing repeatedly, and I assume you likely don't need the plugins for your implementation (at least not yet):
`sudo apt-get uninstall ros-[DISTRO]-mavros-extras`
I'm using ros2 foxy so for me this would look like `sudo apt-get uninstall ros-foxy-mavros-extras`.  

**(3)** As for your commands, its possible that you giving the lat and long coordinates for takeoff is causing it to reject the command for being out of range of the drone.  You can work on adjusting this by looking at the output of the gps:
`ros2 topic echo /mavros/global_position/global`
and adjusting your lat and long given in your takeoff command to match or be very close (as in, probably within 0.0001 of current location for lat and long, though you can play with it once you actually get it off the ground)

Or you can just leave those out, which will use the drone's current location from the gps to set takeoff location, though you need absolute altitude rather than relative to home for this command I believe:
`ros2 service call /mavros/cmd/takeoff mavros_msgs/srv/CommandTOL "{min_pitch: 0, yaw: 90, altitude: 550}"`

With regard to your request for some tutorials, I can pass along some of the other commands I've worked out for doing some useful stuff.  

- Arming:
`ros2 service call /mavros/cmd/arming mavros_msgs/srv/CommandBool "{value: True}"`
- Disarming:
`ros2 service call /mavros/cmd/arming mavros_msgs/srv/CommandBool "{value: False}"`

- Adjusting velocity: (to understand what's happening here look at the CommandLong message, then see [here](https://mavlink.io/en/services/mission.html) and [here](https://mavlink.io/en/messages/common.html#mav_commands), as well as what command 178 means [here](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_CHANGE_SPEED)
`ros2 service call /mavros/cmd/command mavros_msgs/srv/CommandLong "{command: 178, param1: 0, param2: 2, param3: -1, param4: 0}"`

- Creating a waypoint mission and starting it (with velocity adjustments, which are in frame 2, meaning its a mission item rather than a waypoint in a specific world frame):
  - Note that this is originally for a jmavsim environment I believe, not gazebo, so lat's and long's might need to be adjusted to work for you.  
  - Also worth noting that the mission structure doesn't require you to specifically arm the drone or declare a takeoff command, arming is automatic through mission start and the first waypoint can act as your takeoff waypoint.  Also note that I'm using frame 6 for my waypoints, which is altitude relative to the drone's home location rather than absolute altitude with respect to sea level.
  - Also see command [16](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_WAYPOINT) and know that you can find some reference to what commands codes there are in /mavros/mavros_msgs/msg/CommandCode.msg

```
ros2 service call /mavros/mission/push mavros_msgs/srv/WaypointPush "{waypoints: [{frame: 2, command: 178, is_current: True, autocontinue: True, param1: 2, param2: 2, param3: NaN, param4: 0, x_lat: NaN, y_long: NaN, z_alt: NaN}, \
{frame: 6, command: 16, is_current: False, autocontinue: True, x_lat: 47.3977419, y_long: 8.5455938, z_alt: 13}, \
{frame: 2, command: 178, is_current: False, autocontinue: True, param1: 2, param2: 0.001, param3: NaN, param4: 0, x_lat: NaN, y_long: NaN, z_alt: NaN}, \
{frame: 6, command: 16, is_current: False, autocontinue: True, x_lat: 47.3977419, y_long: 8.5455938, z_alt: 15}, \
{frame: 6, command: 16, is_current: False, autocontinue: True, x_lat: 47.3977419, y_long: 8.5455938, z_alt: 5}]}"
```

  - Then start the mission: [300](https://mavlink.io/en/messages/common.html#MAV_CMD_MISSION_START)
`ros2 service call /mavros/cmd/command_int mavros_msgs/srv/CommandInt "{command: 300}"`

- I'm fully aware of how ugly and horrible these service calls are to keep track of and make from the command line, so I'd recommend working on building a python node to construct and push a waypoint list for you, something like:
```
import rclpy
from rclpy.node import Node
from mavros_msgs.srv import WaypointPush
from mavros_msgs.msg import Waypoint
from mavros_msgs.msg import WaypointList

class WaypointPushNode(Node):
    def __init__(self):
        super().__init__('waypoint_push_node')
        waypoint_push_client = self.create_client(WaypointPush, '/mavros/mission/push')
        
        # See /mavros/mavros_msgs/msg/Waypoint.msg and /mavros/mavros_msgs/msg/WaypointList.msg
        waypoint_list = WaypointList()
        waypoint = Waypoint()

        waypoint.frame = 6 # frame 6 = Global relative altitude (to home position)
        waypoint.command = 16 # command 16 = MAV_CMD_NAV_WAYPOINT -> go to waypoint
        waypoint.is_current = False
        waypoint.autocontinue = True
        waypoint.param1 = float(0) # command 16 -> Hold time
        waypoint.param2 = float(0) # command 16 -> Accept Radius (how close you need to be to mark waypoint as reached)
[... cut ...] mmand line example) as things other than strictly waypoints, depending on what you set your command and frame values to.  They are more akin to mission items.  Also I believe that the default setting when the PX4 completes a mission it will enter recovery mode or something similar and will land itself (I believe back at the home location), so you don't need to account for getting your drone back home in your early mission experiments.  Let me know if I can provide any more clarification or help!

vooon, 2022-03-11:
@Mark-Beaty please note that new python library may hide some interactions with rclpy:

https://github.com/mavlink/mavros/blob/ros2/mavros/mavros/__init__.py#L17 - mavros.Client (node class) provides interfaces to params, missions and others.
It's used by `mav` script.

```
import mavros

client = mavros.Client()

client.start_spinner()  # note: instead you may add `client` to node spinner, that method creates thread

client.system.wait_fcu_connection(10.0)

clinet.system.subscribe_state(lambda msg: print(msg))

print(client.waypoint.points)

vooon, 2022-03-11:
Also note that plugin allow and deny lists only needed for UAS node (by default = `/mavros`).

vooon, 2022-03-11:
`mavros_node` - is a container that runs Router and UAS nodes so it acts more like mavros_node from ROS1.

All the plugins now runs inside UAS as a subnode. So each plugin has it's own parameter server, it's own resources.

Rezenders, 2022-05-10:
Hello,

I am having some troubles with mavros2. I am trying to call the arming service as you described but it keeps waiting for it to be available, but I am sure it is available, see below:

```
gus@alien:~/tudelft_hackathon_ws$ ros2 service list -t | grep cmd
/mavros/cmd/arming [mavros_msgs/srv/CommandBool]
/mavros/cmd/command [mavros_msgs/srv/CommandLong]

gus@alien:~/tudelft_hackathon_ws$ ros2 service call /mavros/cmd/arming mavros_msgs/srv/CommandBool '{value: True}'
waiting for service to become available...
```

Any ideas of what might be happening?
I am using ubuntu 20.04 with ros foxy and I tested with mavros installing directly from apt and with the version on the ros2 branch of this repo

I am launching it with this launchfile:

```
<launch>
  <!-- SITL, 255 is the sysid and 240 is the component id -->
  <!-- <arg name="fcu_url" default="udp://:14551@:14555/?ids=255,240"/> -->
  <arg name="fcu_url" default="udp://:14551@:14555"/>
  <arg name="gcs_url" default="udp://:14549@:14548"/>
  <arg name="system_id" default="255"/>
  <arg name="component_id" default="240"/>
  <arg name="tgt_system" default="1"/>
  <arg name="tgt_component" default="1"/>

  <node name="mavros" pkg="mavros" exec="mavros_node">
    <param name="fcu_url" value="$(var fcu_url)"/>
    <param name="gcs_url" value="$(var gcs_url)"/>
    <param name="system_id" value="$(var system_id)"/>
    <param name="component_id" value="$(var component_id)"/>
    <param name="target_system_id" value="$(var tgt_system)"/>
    <param name="target_component_id" value="$(var tgt_component)"/>
  </node>

</launch>
```

By the way, other services work, such as `mavros/set_mode`

matthewoots, 2022-07-16:
@vooon and @dzywater thank you so much for this thread (only thread that is getting ros2 mavros setup for SITL and potentially any system). Several questions related to porting over to ros2.
1. There is no need for `px4_config.yaml` and `px4_pluginlists.yaml`? I've tried to run a launch file like ros1 px4.launch and got stuck at the YAML parser issue https://github.com/ros2/rcl/issues/463, are the configs hardcoded in ros2?
2. I haven't tested on a platform yet (will try) but is vision pose plugin up and stable for streaming vision pose data over to the FC?
3. Is DDS-RTPS (comms to uorb messages) the way to go for ros2 with PX4 and not mavros anymore? Will there be documentations for starting at least SITL with ros2 mavros (becasue i think people will have a hard time if they have not tried mavros before and starting on ros2)?

Sorry to tap on this thread to ask these questions!

vooon, 2022-07-17:
1) They are not ported yet #1747
2) Those plugins mostly the same as on v1
3) If someone will write them...

mzahana, 2023-09-05:
Hi. I keep going into loops through the issues related to ros2. Is there a working example of a launch file for PX4?

ZhongmouLi, 2023-11-11:
Hi, @mzahana @dzywater .

I find this thread really important, as little resource exist talking about how to use mavros2 in ros2.

Could you please help me connect mavros2 and my Ardupilot?

Here is what I did:

1. run `sim_vehicle.py -v ArduCopter -f gazebo-iris --model JSON --map --console`
```
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/robot/RobotFirmware/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Setting top to                           : /home/robot/RobotFirmware/ardupilot 
Setting out to                           : /home/robot/RobotFirmware/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Using toolchain                          : native 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
CXX Compiler                             : g++ 9.4.0 
Checking for need to link with librt     : not necessary 
Checking for feenableexcept              : yes 
Enabled OpenDroneID                      : no 
Enabled firmware ID checking             : no 
GPS Debug Logging                        : no 
Enabled custom controller                : yes 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for HAVE_MEMRCHR                      : yes 
Configured VSCode Intellisense:                : no 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 3.8.10 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 3.8.10 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Scripting                                      : enabled 
Scripting runtime checks                       : enabled 
Debug build                                    : disabled 
Coverage build                                 : disabled 
SITL 32-bit build                              : disabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (0.539s)
{'waf_target': 'bin/arducopter', 'default_params_filename': ['default_params/copter.parm', 'default_params/gazebo-iris.parm'], 'external': True, 'model': 'JSON', 'sitl-port': True}
SIM_VEHICLE: Building
SIM_VEHICLE: "/home/robot/RobotFirmware/ardupilot/modules/waf/waf-light" "build" "--target" "bin/arducopter"
Waf: Entering directory `/home/robot/RobotFirmware/ardupilot/build/sitl'
Embedding file locations.txt:Tools/autotest/locations.txt
Embedding file models/Callisto.json:Tools/autotest/models/Callisto.json
Waf: Leaving directory `/home/robot/RobotFirmware/ardupilot/build/sitl'

BUILD SUMMARY
Build directory: /home/robot/RobotFirmware/ardupilot/build/sitl
Target          Text (B)  Data (B)  BSS (B)  Total Flash Used (B)  Free Flash (B)
[... 79 lines cut ...]
[INFO] [1699715383.069463414] [mavros]: Plugin setpoint_trajectory created
[INFO] [1699715383.069493419] [mavros]: Plugin setpoint_trajectory initialized
[INFO] [1699715383.073369760] [mavros]: Plugin setpoint_velocity created
[INFO] [1699715383.073424540] [mavros]: Plugin setpoint_velocity initialized
[INFO] [1699715383.080053952] [mavros]: Plugin sys_status created
[INFO] [1699715383.080114865] [mavros]: Plugin sys_status initialized
[INFO] [1699715383.084392168] [mavros.time]: TM: Timesync mode: MAVLINK
[INFO] [1699715383.085738873] [mavros]: Plugin sys_time created
[INFO] [1699715383.085779268] [mavros]: Plugin sys_time initialized
[INFO] [1699715383.092068584] [mavros]: Plugin waypoint created
[INFO] [1699715383.092133376] [mavros]: Plugin waypoint initialized
[INFO] [1699715383.096168514] [mavros]: Plugin wind_estimation created
[INFO] [1699715383.096200355] [mavros]: Plugin wind_estimation initialized
[INFO] [1699715383.096749920] [mavros]: Built-in SIMD instructions: SSE, SSE2
[INFO] [1699715383.096765440] [mavros]: Built-in MAVLink package version: 2022.12.30
[INFO] [1699715383.096772137] [mavros]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[INFO] [1699715383.096778961] [mavros]: MAVROS UAS via /uas1 started. MY ID 1.191, TARGET ID 1.1
[INFO] [1699715384.078422027] [mavros_router]: link[1001] detected remote address 1.191
[WARN] [1699715385.084221364] [mavros.sys]: VER: broadcast request timeout, retries left 4
[WARN] [1699715386.084006698] [mavros.sys]: VER: broadcast request timeout, retries left 3
[WARN] [1699715387.084127859] [mavros.sys]: VER: unicast request timeout, retries left 2
[WARN] [1699715388.083773872] [mavros.sys]: VER: unicast request timeout, retries left 1
[WARN] [1699715389.084158020] [mavros.sys]: VER: unicast request timeout, retries left 0
[WARN] [1699715390.081450652] [mavros.sys]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
```

Rezenders, 2023-11-12:
@ZhongmouLi I recommend running mavros with the mavros launchfile, since it includes all the required params.

You can find an example of a launchfile for adupilot here: https://github.com/kas-lab/suave/blob/835b8a0fa5a24fed66a031ccb7eaf36b92de42c7/suave/launch/simulation.launch.py#L34

```
    mavros_path = get_package_share_directory('mavros')
    mavros_launch_path = os.path.join(
        mavros_path, 'launch', 'apm.launch')
    mavros_node = IncludeLaunchDescription(
        AnyLaunchDescriptionSource(mavros_launch_path),
        launch_arguments={
            'fcu_url': 'udp://127.0.0.1:14551@14555',
            'gcs_url': 'udp://@localhost:14550',
            'system_id': '255',
            'component_id': '240',
            'target_system_id': '1',
            'target_component_id': '1'
            }.items()
        )
```

ZhongmouLi, 2023-11-26:
@Rezenders  Thanks so much for your reply.

But I am using ros2 foxy. Do you mean I run mavros instead of mavros2 in foxy?

Rezenders, 2023-11-27:
> @Rezenders Thanks so much for your reply.
> 
> But I am using ros2 foxy. Do you mean I run mavros instead of mavros2 in foxy?

Hi @ZhongmouLi, I am not really sure which mavros version is out for foxy. But what I am doing for ros2 Humble is that I am getting mavros from this github repo instead of the version in apt. The version in the repo is working better for me (at least the last time I tried).

ZhongmouLi, 2023-12-16:
@Rezenders Thanks so much, after reading your reply. I finally found the solution.

In fact, it should be `'fcu_url': 'udp://127.0.0.1:14551@14555'`

iftahnaf, 2024-02-04:
Hi all,

Has someone implemented a service call for arming and setting mode to offboard in rclpy?

For some reason the arm command freezes at the ` rclpy.spin_until_future_complete(self, future, timeout_sec=1.0)`
```python
    def arm(self):
        while not self.arming_client.wait_for_service(timeout_sec=1.0):
            self.get_logger().info('Arming service not available, waiting again...')
        request = CommandBool.Request()
        request.value = True
        future = self.arming_client.call_async(request)
        rclpy.spin_until_future_complete(self, future, timeout_sec=1.0)
        if future.result() is not None:
            self.arming_state = True
            self.get_logger().info('Vehicle armed')
        else:
            self.get_logger().info('Request failed')
```

And in the setting mode I having error when using

```python
from mavros_msgs.msg import State
self.state_sub = self.create_subscription(State, '/mavros/state', self.cb_state, 10)
```

which returns:

```bash
AttributeError: type object 'EnumMeta' has no attribute '_TYPE_SUPPORT' This might be a ROS 1 message type but it should be a ROS 2 message type. Make sure to source your ROS 2 workspace after your ROS 1 workspace
```

even when ROS 1 is not installed inside the container.

duguguang, 2024-07-10:
Also you can create the config file, which names mavros_param.yaml, then run the command:
 `ros2 run mavros mavros_node --ros-args --params-file mavros_parm.yaml `

```
# mavros_param.yaml
mavros:
  ros__parameters: {}

mavros_router:
  ros__parameters: {}

mavros_node:
  ros__parameters:
    fcu_url: /dev/ttyACM0:57600
    gcs_url: udp://@192.168.3.199:14550
    tgt_system: 3
    tgt_component: 1
    log_output: screen
    fcu_protocol: v2.0
    respawn_mavros: false
    pluginlists_yaml: path/to/mavros/launch/px4_pluginlists.yaml
    config_yaml: path/to/mavros/launch/px4_config.yaml

```

duguguang, 2024-07-10:
> I use v2.1.0 [image]
> 
> It seems that takeoff cannot be taken. I'm sure I call takeoff service in time before disarming.
> 
> Moveover After run: ros2 run mavros mavros_node --ros-args --param fcu_url:=udp://:14540@ Always shows in the terminal [image]

Have you ever solve the ODOM:Ex .. problem ?

Pratikw369, 2025-03-18:
i have pixhawk 4 connected to my laptop by usb i can arm it from qgc when virtual throttle is down but i want to arm it using MAVROS when i tried to do using mavros 

prot:~/Downloads$ ros2 service call /mavros/cmd/arming mavros_msgs/srv/CommandBool "{value: True}"
waiting for service to become available...
requester: making request: mavros_msgs.srv.CommandBool_Request(value=True)

response:
mavros_msgs.srv.CommandBool_Response(success=False, result=1)

closed 2026-10-27 by github-actions[bot]