mavlink/mavros #1724: Error while plugin loading

BurakHocaoglu, 2022-03-15:
### Issue details

Hello,

I'm using Mavros 1.13.0 Release to use `camera` plugin for `gazebo-iris` model of ArduCopter in Gazebo. When I roslaunch, Mavros directly fails without trying to connect to an FCU at all. I added `gdb` in `node.launch` to see where it fails and it shows that a segfault happens when loading plugins in `mavros` / `mavros_extras`.

### MAVROS version and platform

Mavros: 1.13.0 Release
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ X ] ArduPilot
[  ] PX4

Version: Copter 4.2.0 Release

### Node logs

This is the output when I used gdb in launch prefix of `node` tag (`launch-prefix="gdb -ex run --args"`):
```
... logging to /home/burak/.ros/log/84a1aacc-a439-11ec-ade0-4074e0a985bc/roslaunch-burak-GL65-9SD-9738.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://burak-GL65-9SD:38225/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
 * /mavros/fake_gps/satellites_visible: 6
 * /mavros/fake_gps/speed_accuracy: 0.0
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
[... 372 lines cut ...]
[DEBUG] [1647339883.155647355]: Publishing message on topic [/rosout] with sequence number [7]
[DEBUG] [1647339883.155667028]: Publishing message on topic [/rosout] with sequence number [7]
[DEBUG] [1647339883.155678073]: Publishing message on topic [/rosout] with sequence number [7]
[DEBUG] [1647339883.155688369]: Publishing message on topic [/rosout] with sequence number [7]
================================================================================REQUIRED process [mavros-2] has died!
process has died [pid 12837, exit code -11, cmd /home/burak/damadama_ws/devel/lib/mavros/mavros_node __name:=mavros __log:=/home/burak/.ros/log/20a4bd3c-a44a-11ec-ade0-4074e0a985bc/mavros-2.log].
log file: /home/burak/.ros/log/20a4bd3c-a44a-11ec-ade0-4074e0a985bc/mavros-2*.log
Initiating shutdown!
================================================================================
[mavros-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

### Diagnostics

Can't run due to incomplete roslaunch.

### Check ID

Can't run due to incomplete roslaunch.

vooon, 2022-03-15:
Looks like double-free. While it's unclear where exactly.

Could you please try to build latest master?

BurakHocaoglu, 2022-03-15:
I tried, but the result is the same.

I copied only the camera plugin of latest `mavros_extras` into an older version of `mavros`, where the camera plugin isn't included. Without it, the old version works fine, but with it, the same problem happens.

vooon, 2022-03-15:
It's only apply to camera plugin? What if you blacklist it? (I want to locate troublesome plugin).

BurakHocaoglu, 2022-03-16:
I don't think my issue is related to a specific plugin, but rather to the double-free you mentioned in the first place. I tried blacklisting everything in `mavros` and `mavros_extras` and nothing changed. The debug level output of ROS suggests that it does try to load the `mavros_plugins.xml` files in both packages, but none of the plugins actually make it to the loading stage.

I created and built a new workspace and instead of `gdb`, I used `valgrind` to check whether something different comes out.

The output with Valgrind:
```
... logging to /home/burak/.ros/log/9ecadfe6-a500-11ec-835b-4074e0a985bc/roslaunch-burak-GL65-9SD-18059.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://burak-GL65-9SD:40077/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
 * /mavros/fake_gps/satellites_visible: 6
 * /mavros/fake_gps/speed_accuracy: 0.0
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_hil_gps: True
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fake_gps/vert_accuracy: 0.5
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://127.0.0.1:1...
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
 * /mavros/global_position/use_relative_alt: True
[... 267 lines cut ...]
==18102==    by 0x768E51C: clone (clone.S:109)
==18102== 
==18102== HEAP SUMMARY:
==18102==     in use at exit: 22,222,736 bytes in 27,529 blocks
==18102==   total heap usage: 86,504 allocs, 59,025 frees, 43,753,363 bytes allocated
==18102== 
==18102== LEAK SUMMARY:
==18102==    definitely lost: 2,672 bytes in 50 blocks
==18102==    indirectly lost: 0 bytes in 0 blocks
==18102==      possibly lost: 3,200 bytes in 10 blocks
==18102==    still reachable: 22,216,864 bytes in 27,469 blocks
==18102==         suppressed: 0 bytes in 0 blocks
==18102== Rerun with --leak-check=full to see details of leaked memory
==18102== 
==18102== For counts of detected and suppressed errors, rerun with: -v
==18102== Use --track-origins=yes to see where uninitialised values come from
==18102== ERROR SUMMARY: 200 errors from 8 contexts (suppressed: 0 from 0)
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I think Valgrind caught the segmentation fault and instead of terminating, let the rest of the program run as normal; hence, the output looking like mavros is actually working. I'll open an issue for `pluginlib` repo as well.

closed 2026-10-27 by github-actions[bot]