mavlink/mavros #1728: /mavros/landing_target/raw issue

ParkYoungHoo1, 2022-04-09:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: latest
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[ O ] ArduPilot
[  ] PX4

Version: 4.2.0

### Node logs

```
roslaunch mavros apm.launch fcu_url:=/dev/ttyUSB0:921600
```

### Diagnostics

```
place here result of:
rostopic echo /mavros/landing_target/raw
```

### Check ID

```
rosrun mavros checkid
```

Hello!
I am researching using the landing_target topic using mavros.
The SITL simulation using Gazebo has been completed using the /mavros/landing_target/raw topic. 
However, for the outdoor test, we tried to use the same topic by connecting Jetson TX2 to Pixhawk, but I did not see the /mavros/landing_target/raw topic on the rqt_graph. 
It has not been resolved why the topic is not visible. 
I wonder if it would be good to update to the latest version or if there is another way to solve it.

You can see below that /mission_node publishes the /mavros/landing_target/raw topic as the latest version of mavros in Ubuntu 18.04 but is not visible in the rqt_graph.
[image]
Thank you!
.

tugbakara, 2022-06-14:
Could you explain how to test in Gazebo this test because I ddn't tested it yet evven if I searched about it, it was confusing to me.
 I cannot see too this topic in rviz , did you solve that issue? When I make rostopic echo I didn't see ../landing_target/raw topic instead of it there is ../landing_target/pose_in (.. means mavros)

vooon, 2022-06-14:
https://github.com/mavlink/mavros/blob/cbe3aefb356b70c463be0ee1027ba4438efbd805/mavros_extras/src/plugins/landing_target.cpp#L64

What is the value of that parameter (`/mavros/landing_target/listen_lt`)?

tugbakara, 2022-06-15:
Isn't it enough to change in apm_config.yaml? I will change it now but my firmware is arduplane so if there is another restriction about the algorithm I have to change it also.

ParkYoungHoo1, 2022-06-15:
When I simulated the multi-copter in Gazebo, it is enough to change the landing_target parameter false to true.
But, now I solved the Jetson TX2 case in the same way as above.
I dont have an idea about the arduplane...

tugbakara, 2022-06-15:
Thanks! I will search about the simulation in gazebo too!

ParkYoungHoo1, 2022-06-15:
You are welcome!

tugbakara, 2022-06-15:
I changed the false to true and then catkin build but I don't see raw topic still instead ../landing_target/lt_marker topic became visable.

> https://github.com/mavlink/mavros/blob/cbe3aefb356b70c463be0ee1027ba4438efbd805/mavros_extras/src/plugins/landing_target.cpp#L64
> 
> What is the value of that parameter (`/mavros/landing_target/listen_lt`)?

tejalbarnwal, 2023-08-31:
@tugbakara did you manage to solve it?
Even when I enable false to true, I don't see the `/mavros/landing_target/raw` topic

closed 2026-10-27 by github-actions[bot]