mavlink/mavros #1729: How can I set mode to Offboard mode??

sumeyrakarsavran, 2022-04-10:
Hi guys,

I'm trying to control a drone in gazebo via mavros. I try to run a test node which set the mode to offboard, arm the drone and draw a square. When I run the node the drone takeoff and hold just a while then land. I cant set mode to offboard mode. 
Px4 and QGC show this warning '`WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported`'. I couldn't find what the problem exactly is.	 
I use ros melodic, gazebo9,vmware ubuntu 18.04.

THIRD TERMINAL IS THAT

   ```
 sumeyra@ubuntu:~/PX4-Autopilot$ rosrun tulpar kare.py 
    Waiting for arming...
    Waiting for arming...
    Waiting for arming...
    Waiting for arming...
    Waiting for arming...
    Waiting for arming...
    Waiting for arming...
    Waiting for arming...
    MAIN: SET OFFBOARD
```

(It just print 'set offboard' because of that I wrote this message on my code)
But while running my code I did '`rostopic echo /mavros/state`' which results in

    ```
sumeyra@ubuntu:~/PX4-Autopilot$ rostopic echo /mavros/state
    header: 
      seq: 44
      stamp: 
        secs: 1648394216
        nsecs: 419879279
      frame_id: ''
    connected: True
    armed: False
    guided: True
    manual_input: False
    mode: "AUTO.LOITER"
    system_status: 3
    ---
    header: 
      seq: 54
      stamp: 
        secs: 1648394230
        nsecs: 332277808
      frame_id: ''
    connected: True
    armed: True
    guided: True
    manual_input: False
    mode: "AUTO.TAKEOFF"
    system_status: 4
    ---
    header: 
      seq: 62
      stamp: 
        secs: 1648394246
        nsecs: 270976013
      frame_id: ''
    connected: True
    armed: True
    guided: True
    manual_input: False
    mode: "AUTO.LOITER"
    system_status: 4
    ---
    header: 
      seq: 79
      stamp: 
        secs: 1648394276
        nsecs: 674984034
[... 442 lines cut ...]
    INFO  [commander] Takeoff detected	
    WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported	
    WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported	
    WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported	
    WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported	
    WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported	
    WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported	
    WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported	
    WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported	
    WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported	
    WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported	
    WARN  [commander] Failsafe enabled: No manual control stick input	
    INFO  [commander] Failsafe mode activated	
    INFO  [tone_alarm] battery warning (fast)
    INFO  [navigator] RTL HOME activated	
    INFO  [navigator] RTL: landing at home position.	
    INFO  [navigator] RTL: climb to 104 m (4 m above destination)	
    INFO  [commander] Failsafe mode deactivated	
    INFO  [navigator] RTL: return at 104 m (4 m above destination)	
    INFO  [navigator] RTL: land at destination	
    INFO  [commander] Landing detected	
    INFO  [commander] Disarmed by landing	
    INFO  [logger] closed logfile, bytes written: 17511320
    ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
```

magic5al, 2022-06-29:
I meet the same question.

DARKMOONlite, 2024-12-09:
okay I came across the same issue and it seems to be because i didn't correctly set the `coordinate_frame` variable in the `PositionTarget` [message](http://docs.ros.org/en/noetic/api/mavros_msgs/html/msg/PositionTarget.html)

`WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported`
This error seems to state that the `coordinate_frame` variable is 0 and thus not an acceptable value, below are some values you can set it too that should resolve this issue. 
```
uint8 FRAME_LOCAL_NED = 1
uint8 FRAME_LOCAL_OFFSET_NED = 7
uint8 FRAME_BODY_NED = 8
uint8 FRAME_BODY_OFFSET_NED = 9
```

closed 2026-10-26 by github-actions[bot]