mavlink/mavros #1736: mavftp download : unable to download

dookei, 2022-04-26:
Hi there,
I am currently unable to download logs though mavros mavftp download. This worked fine on previous versions from px4. I recently updated PX4 to 1.13.
This is what I am currently get:
`
rosrun mavros mavftp cat /fs/microsd/log/2022-04-11/14_38_25.ulg
Verification failed: 0x00000000 != 0x4b2ffc46
`
or
` 
rosrun mavros mavftp download /fs/microsd/log/2022-04-11/14_38_25.ulg
Downloading: 100% |#######################################################################################| Time: 0:00:00  43.42 MB/s
Verification failed: 0x00000000 != 0x4b2ffc46
`
Of course the file is empty after download. 

PX4: v1.13.0 master
mavros: 1.9.0
ROS1: melodic
Any ideas why this is happening and where the issue might come from ?

vooon, 2022-04-26:
I think here need to check in packet level. My suggestion that server protocol might be changed. Can you try to test older firmware?

dookei, 2022-04-28:
I will give it a try now...

dookei, 2022-04-28:
I tested with Px4 v1.12.3 and this is working fine.

vooon, 2022-04-28:
Ok, then next thing to analyze - diff 1.12.3 and 1.13.0 (in mavlink app).

dookei, 2022-05-04:
Thanks @vooon ....there are simply to many changes. Where should I even start ? ;-)

````
Update submodule mavlink to latest Sat Feb 12 12:38:48 UTC 2022
41e48435c9 by PX4 BuildBot <bot@px4.io>, Sat Feb 12 2022 (3 months ago)
1 insertion(+), 1 deletion(-)

Update submodule mavlink to latest Wed Feb  9 00:39:13 UTC 2022
33ce1b9b64 by PX4 BuildBot <bot@px4.io>, Wed Feb 09 2022 (3 months ago)
1 insertion(+), 1 deletion(-)

vmount: refactor for v2 auto input, test command
39f0e97245 by Julian Oes <julian@oes.ch>, Tue Feb 08 2022 (3 months ago)
2 files changed, 25 insertions(+)

mavlink: don't send gimbal_device_attitude_status
f2216dff55 by Julian Oes <julian@oes.ch>, Tue Feb 08 2022 (3 months ago)
2 files changed, 10 insertions(+)

mavlink: handle for GIMBAL_DEVICE_ATTITUDE_STATUS
7e7a99b542 by Julian Oes <julian@oes.ch>, Tue Feb 08 2022 (3 months ago)
2 files changed, 31 insertions(+)

mavlink: don't send_mission_current if mission invalid
cf3db0d313 by Daniel Agar <daniel@agar.ca>, Fri Feb 04 2022 (3 months ago)
2 files changed, 31 insertions(+), 30 deletions(-)

mavlink: streams/COMMAND_LONG don't send internal vehicle_commands
e387f302f9 by Daniel Agar <daniel@agar.ca>, Tue Feb 01 2022 (3 months ago)
10 insertions(+), 1 deletion(-)

add odometry reset counter (ekf2, mavlink, etc)
90358f078f by Daniel Agar <daniel@agar.ca>, Mon Jan 31 2022 (3 months ago)
2 files changed, 6 insertions(+)

mavlink: fix linking
4c3234af38 by Jukka Laitinen <jukkax@ssrc.tii.ae>, Thu Jan 27 2022 (3 months ago)
4 insertions(+)

Enable logging of mavlink tunnel messages
c8349811d1 by Roman Dvořák <romandvorak@mlab.cz>, Wed Jan 26 2022 (3 months ago)
2 files changed, 27 insertions(+)

Update submodule mavlink to latest Sat Jan 22 00:39:17 UTC 2022
6203ad25c7 by PX4 BuildBot <bot@px4.io>, Sat Jan 22 2022 (3 months ago)
1 insertion(+), 1 deletion(-)

Update submodule mavlink to latest Tue Jan 18 00:39:07 UTC 2022
0bb9cce1ce by PX4 BuildBot <bot@px4.io>, Tue Jan 18 2022 (4 months ago)
1 insertion(+), 1 deletion(-)

extend support for Battery status message
d122513197 by Claudio Micheli <claudio@auterion.com>, Thu Jan 13 2022 (4 months ago)
24 insertions(+)

mavlink: make range params optional
22180a2639 by Julian Oes <julian@oes.ch>, Tue Jan 11 2022 (4 months ago)
2 files changed, 17 insertions(+), 5 deletions(-)

Mantis: add RC hacks
e4763f15f6 by Julian Oes <julian@oes.ch>, Tue Jan 11 2022 (4 months ago)
22 insertions(+)

mavlink: set signal_quality (and sane variance)
dd00e43ca3 by Julian Oes <julian@oes.ch>, Tue Jan 11 2022 (4 months ago)
4 insertions(+), 2 deletions(-)

mavlink: use optical flow limits from params
aefbd80b53 by Julian Oes <julian@oes.ch>, Tue Jan 11 2022 (4 months ago)
2 files changed, 5 insertions(+), 3 deletions(-)
[... 364 lines cut ...]

mavlink_main: add ESC_INFO & ESC_STATUS to onboard_lowbandwith configuration
a6d88cad18 by Claudio Micheli <claudio@auterion.com>, Wed Aug 11 2021 (9 months ago)
2 insertions(+)

mavlink: receiver fix geo include
a93f8dade4 by Oleg Evseev <ev.mipt@gmail.com>, Tue Aug 10 2021 (9 months ago)
2 files changed, 1 insertion(+), 1 deletion(-)

mavlink: receiver handle RC_CHANNELS if from MAV_COMP_ID_SYSTEM_CONTROL
fb3b7a4649 by Daniel Agar <daniel@agar.ca>, Wed Aug 04 2021 (9 months ago)
2 files changed, 72 insertions(+)

mavlink: receiver fix HIL_STATE_QUATERNION map projection init
d997a8d308 by Daniel Agar <daniel@agar.ca>, Mon Aug 02 2021 (9 months ago)
2 files changed, 32 insertions(+), 35 deletions(-)

mavlink: use dynamic camera comp id instead of hardcoded value
af2247bd94 by Igor Mišić <igy1000mb@gmail.com>, Sat Jul 31 2021 (9 months ago)
3 files changed, 19 insertions(+), 3 deletions(-)

Add internal combustion engine status uavcan bridge and mavlink EFI_STATUS stream
d08d0443bc by Dima Ponomarev <36133264+PonomarevDA@users.noreply.github.com>, Sat Jul 31 2021 (9 months ago)
3 files changed, 104 insertions(+), 1 deletion(-)
```

closed 2026-10-26 by github-actions[bot]