mavlink/mavros #1738: Telemetry swarm

pedro-fuoco, 2022-04-28:
### Issue details

Hello, I would like to connect multiple drones to mavros via a single usb telemetry, is this possible ? I know it can receive mavlink messages from all of them since QGroundControl can display them all and even allows for mission planning.
I've managed to connect different drones through different namespaces, but only using wifi and usb cable. When I try to run multiple mavros nodes in the same port (where the telemetry currently is)  the first one always works but the other is unable to use the same port simultaneously. Is there a known way to go around this?

### MAVROS version and platform

Mavros: 2.0.0
ROS: Foxy
Ubuntu: 20.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: v1.12

vooon, 2022-04-28:
Since you're referring to ROS2, then yes, you may use single port.

You should run one Router node and a set of UAS nodes.

In other words you need to run `mavros::router::Router` with params like that
```
mavros_router:
  ros__parameters:
    fcu_urls: [/dev/ttyUSB0:57600]
    gcs_urls: [udpb://@]
    uas_urls:
      - /uas1
      - /uas2
      - /uas3
```

Then you need to run instance of `mavros::uas::UAS` for each `/uas%d`.

squizz617, 2022-08-27:
Hi @vooon, then what about ROS 1 (e.g., with mavros 1.13.0)?

vooon, 2022-08-27:
@squizz617 it's not directly supported, as intended use one mavros connected to one fcu on obc.

But i think you may use intel's mavlink-router to split single connection to the radio to the set of udp streams for set of mavros'es.

squizz617, 2022-08-30:
I see. Thank you for the suggestion!

closed 2026-10-26 by github-actions[bot]