mavlink/mavros #1739: Feature: handle EKF_STATUS_REPORT msg

clydemcqueen, 2022-04-30:
### Issue details

ArduPilot sends EKF_STATUS_REPORT rather than ESTIMATOR_STATUS. I would like to get this message in ROS.

I suppose the flags can be sent in a mavros_msgs/msg/EstimatorStatus.msg, but I would like to convert the entire EKF_STATUS_REPORT message to a ROS message.

Thoughts on this? Does this sort of thing go in mavros_extras?

Thanks.

### MAVROS version and platform

Mavros: HEAD
ROS: Galactic
Ubuntu: 20.04

### Autopilot type and version

[x] ArduPilot
[  ] PX4

vooon, 2022-05-03:
If it's possible to convert to EstimatorStatus.msg, then i believe it would be better to just extend sys_status plugin.

vooon, 2022-05-03:
If not, then still probably better to put to sys_status.

Or maybe do a refactoring to extract estimator related things to a separate plugin.

clydemcqueen, 2022-05-03:
Thanks for the suggestions. I will do some investigations and make a concrete proposal.

closed 2026-10-26 by github-actions[bot]