mavlink/mavros #1749: Mission waypoint list does not update when a new mission file is uploaded to PX4

AhmedElsafy, 2022-05-25:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

MAVROS topic /mavros/mission/waypoints does not update when new mission file is uploaded from the Ground station software 

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

PX4

Version: 1.13-beta

### Diagnostics

Subscribing to the topic and uploading a new mission file 

```
rostopic echo -c /mavros/mission/waypoints
```
When the service /mavros/mission pull is called, the topic is updated

vooon, 2022-05-27:
Did you use different interface to upload mission than mavros uses?

Mission protocol does not broadcast changes. Plugin may catch mission messages directed to gcs and schedule a pull only if mavros used as a proxy.
Anyway it's not guaranteed to update. You must pull waypoints.

vooon, 2022-05-27:
Mavros works with MAVLink, not uORB. If the message traffic between GCS and FCU comes trough mavros, it's possible to detect mission update. Otherwise you cannot get any signal that the list have been updated.

Anyway, why do you need to renew the list so often?

vooon, 2022-05-27:
How do you connect QGC?

AhmedElsafy, 2022-05-27:
QGC is connected throw mav radio (SIK) on the first telemtry port, ros running on the OBC connected to a second telemetry port with a different mavlink forwarding

vooon, 2022-05-27:
If you connect SiK to the OBC and let gcs_url to proxy your telemetry, you'll be able to catch list update.
But at the same time, that setup less fault tolerant.

AhmedElsafy, 2022-05-30:
for now as a work around I will add a subscriber to the flight mode topic with a call back function that calls the mission list pull service when the flight mode is switched to Mission mode

closed 2026-10-25 by github-actions[bot]