mavlink/mavros #1757: Not working with ROS2 humble

Macavitycode, 2022-06-23:
I recently upgraded my setup to using Ubuntu 22.04 and ROS2 humble (from 20.04 and ROS2 foxy) and was looking to migrate and update few projects which use mavros.

- Mavros does not seem to be available in the ros humble repositories (only ros-humble-mavros-messages and ros-humble-mavros-messages-dbgsym are available)
- Mavros doesn't compile from source correctly and throws an error while following the installation instructions [here](https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation) and running colcon build instead of catkin build:

Questions:

- When will mavros be available as a binary package in ros-humble?
- What are potential fixes to the compile error below?

```bash
# $ colcon build

Starting >>> mavlink 
Starting >>> mavros_msgs
Finished <<< mavlink [0.18s]         
Starting >>> libmavconn
Finished <<< mavros_msgs [2.18s]     
--- stderr: libmavconn   
In file included from /usr/include/c++/11/cstdio:42,
                 from /usr/include/c++/11/ext/string_conversions.h:43,
                 from /usr/include/c++/11/bits/basic_string.h:6608,
                 from /usr/include/c++/11/string:55,
                 from /home/shashankrajivmoghe/ros/build/libmavconn/generated/src/mavlink_helpers.cpp:20:
/home/shashankrajivmoghe/ros/install/mavlink/include/mavlink/v2.0/common/common.hpp:2328:5: error: expected identifier before ‘(’ token
 2328 |     EOF=6, /* EOF: Offset past end of file for ListDirectory and ReadFile commands | */
      |     ^~~
/home/shashankrajivmoghe/ros/install/mavlink/include/mavlink/v2.0/common/common.hpp:2328:5: error: expected ‘}’ before ‘(’ token
In file included from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/include/mavconn/mavlink_dialect.hpp:44,
                 from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/include/mavconn/interface.hpp:26,
                 from /home/shashankrajivmoghe/ros/build/libmavconn/generated/src/mavlink_helpers.cpp:24:
/home/shashankrajivmoghe/ros/install/mavlink/include/mavlink/v2.0/common/common.hpp:2320:1: note: to match this ‘{’
 2320 | {
      | ^
In file included from /usr/include/c++/11/cstdio:42,
                 from /usr/include/c++/11/ext/string_conversions.h:43,
                 from /usr/include/c++/11/bits/basic_string.h:6608,
                 from /usr/include/c++/11/string:55,
                 from /home/shashankrajivmoghe/ros/build/libmavconn/generated/src/mavlink_helpers.cpp:20:
/home/shashankrajivmoghe/ros/install/mavlink/include/mavlink/v2.0/common/common.hpp:2328:5: error: expected unqualified-id before ‘-’ token
 2328 |     EOF=6, /* EOF: Offset past end of file for ListDirectory and ReadFile commands | */
      |     ^~~
/home/shashankrajivmoghe/ros/install/mavlink/include/mavlink/v2.0/common/common.hpp:2328:5: error: expected ‘)’ before ‘-’ token
 2328 |     EOF=6, /* EOF: Offset past end of file for ListDirectory and ReadFile commands | */
      |     ^~~
In file included from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/include/mavconn/mavlink_dialect.hpp:44,
                 from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/include/mavconn/interface.hpp:26,
                 from /home/shashankrajivmoghe/ros/build/libmavconn/generated/src/mavlink_helpers.cpp:24:
/home/shashankrajivmoghe/ros/install/mavlink/include/mavlink/v2.0/common/common.hpp:2366:1: error: expected declaration before ‘}’ token
 2366 | } // namespace mavlink
      | ^
In file included from /usr/include/c++/11/cstdio:42,
                 from /usr/include/c++/11/ext/string_conversions.h:43,
                 from /usr/include/c++/11/bits/basic_string.h:6608,
                 from /usr/include/c++/11/string:55,
                 from /usr/include/console_bridge/console.h:40,
                 from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/include/mavconn/console_bridge_compat.hpp:22,
                 from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/src/tcp.cpp:18:
/home/shashankrajivmoghe/ros/install/mavlink/include/mavlink/v2.0/common/common.hpp:2328:5: error: expected identifier before ‘(’ token
 2328 |     EOF=6, /* EOF: Offset past end of file for ListDirectory and ReadFile commands | */
      |     ^~~
/home/shashankrajivmoghe/ros/install/mavlink/include/mavlink/v2.0/common/common.hpp:2328:5: error: expected ‘}’ before ‘(’ token
In file included from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/include/mavconn/mavlink_dialect.hpp:44,
                 from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/include/mavconn/interface.hpp:26,
                 from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/include/mavconn/tcp.hpp:22,
                 from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/src/tcp.cpp:19:
/home/shashankrajivmoghe/ros/install/mavlink/include/mavlink/v2.0/common/common.hpp:2320:1: note: to match this ‘{’
 2320 | {
      | ^
In file included from /usr/include/c++/11/cstdio:42,
[... 122 lines cut ...]
 2328 |     EOF=6, /* EOF: Offset past end of file for ListDirectory and ReadFile commands | */
      |     ^~~
In file included from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/include/mavconn/mavlink_dialect.hpp:44,
                 from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/include/mavconn/interface.hpp:26,
                 from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/include/mavconn/udp.hpp:22,
                 from /home/shashankrajivmoghe/ros/src/mavros/libmavconn/src/udp.cpp:20:
/home/shashankrajivmoghe/ros/install/mavlink/include/mavlink/v2.0/common/common.hpp:2366:1: error: expected declaration before ‘}’ token
 2366 | } // namespace mavlink
      | ^
gmake[2]: *** [CMakeFiles/mavconn.dir/build.make:76: CMakeFiles/mavconn.dir/generated/src/mavlink_helpers.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/mavconn.dir/build.make:90: CMakeFiles/mavconn.dir/src/interface.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/mavconn.dir/build.make:104: CMakeFiles/mavconn.dir/src/serial.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/mavconn.dir/build.make:132: CMakeFiles/mavconn.dir/src/udp.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/mavconn.dir/build.make:118: CMakeFiles/mavconn.dir/src/tcp.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:157: CMakeFiles/mavconn.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< libmavconn [2.67s, exited with code 2]

Summary: 2 packages finished [2.99s]
  1 package failed: libmavconn
  1 package had stderr output: libmavconn
  2 packages not processed
```

vooon, 2022-06-23:
You should try to use ros2 branch, because master is for ros1.

vooon, 2022-06-23:
@clydemcqueen that require really simple fix to mavgen to enum name generation. But i do not have any interest in committing to ignorant people anymore, sorry.

Macavitycode, 2022-06-24:
@vooon I'm really sorry you feel that way. I hadn't realised the same bug was reported earlier, but as of now that thread too has no fix per say.
It'd be very helpful to everyone who's shifting to humble and facing the same issue if you could please take a minute and explain what is the fix to mavgen to enum name generation is.

vooon, 2022-06-24:
@Macavitycode fix it's really simple: just need to add a list of reserved words, like 'NULL' or 'EOF' (please check C spec).
Then you may change prefix removal to check that list and return part of prefix in case of that entries: in other words: MAV_FTP_ERR_EOF -> ERR_EOF. (That solution i believe more common).

Other possibility is just extend MACROS dict in mavgen_cpp11.py to have entries `'NULL': 'NULL_', 'EOF': 'EOF_'`, etc.
Also you may try to use older mavlink releases, without MAV_FTP_ERR enum.

vooon, 2022-06-24:
@Macavitycode cannot say anything yet. But have you tried to remove ws/build/ and ws/install/ dirs (to be sure that mavlink is rebuilt)?

Macavitycode, 2022-06-27:
Thank you so much for your help so far. I removed the build and install directories and sourced a fresh environment. I then compiled each package individually (mavlink first then libmavconn, then mavros_msgs) and those compiled fine. I then sourced the install/setup.sh and tried to compile again and I got an error saying mavlink_convert.hpp not found. I checked the install directory and found the mavlink_convert.hpp file in mavros_msgs directory under include. I then checked the CMakeList.txt of mavros I found that mavros_msgs is included as expected (its present in the find_packages command, the ament_target_dependancies command and also the ament_target_dependancies commands).

Please advice on what to do, thank you for your help so far.

```bash
shashankrajivmoghe@pop-os:~/ros$ find . -name mavlink_convert.hpp
./install/mavros_msgs/include/mavros_msgs/mavlink_convert.hpp
./src/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.hpp
```

```bash
shashankrajivmoghe@pop-os:~/ros$ colcon build --packages-select mavros
Starting >>> mavros  
--- stderr: mavros                             
In file included from /home/shashankrajivmoghe/ros/src/mavros/mavros/src/lib/enum_sensor_orientation.cpp:21:
/home/shashankrajivmoghe/ros/src/mavros/mavros/include/mavros/utils.hpp:27:10: fatal error: mavros_msgs/mavlink_convert.hpp: No such file or directory
   27 | #include "mavros_msgs/mavlink_convert.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/shashankrajivmoghe/ros/src/mavros/mavros/src/lib/enum_to_string.cpp:21:
/home/shashankrajivmoghe/ros/src/mavros/mavros/include/mavros/utils.hpp:27:10: fatal error: mavros_msgs/mavlink_convert.hpp: No such file or directory
   27 | #include "mavros_msgs/mavlink_convert.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/mavros.dir/build.make:90: CMakeFiles/mavros.dir/src/lib/enum_to_string.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/mavros.dir/build.make:76: CMakeFiles/mavros.dir/src/lib/enum_sensor_orientation.cpp.o] Error 1
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'author-email' will not be supported in future versions. Please use the underscore name 'author_email' instead
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' instead
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
  warnings.warn(
In file included from /home/shashankrajivmoghe/ros/src/mavros/mavros/include/mavros/mavros_router.hpp:35,
                 from /home/shashankrajivmoghe/ros/src/mavros/mavros/src/lib/mavros_router.cpp:20:
/home/shashankrajivmoghe/ros/src/mavros/mavros/include/mavros/utils.hpp:27:10: fatal error: mavros_msgs/mavlink_convert.hpp: No such file or directory
   27 | #include "mavros_msgs/mavlink_convert.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/shashankrajivmoghe/ros/src/mavros/mavros/include/mavros/uas_executor.hpp:28,
                 from /home/shashankrajivmoghe/ros/src/mavros/mavros/src/lib/uas_executor.cpp:16:
/home/shashankrajivmoghe/ros/src/mavros/mavros/include/mavros/utils.hpp:27:10: fatal error: mavros_msgs/mavlink_convert.hpp: No such file or directory
   27 | #include "mavros_msgs/mavlink_convert.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/mavros.dir/build.make:132: CMakeFiles/mavros.dir/src/lib/mavros_router.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/mavros.dir/build.make:202: CMakeFiles/mavros.dir/src/lib/uas_executor.cpp.o] Error 1
In file included from /home/shashankrajivmoghe/ros/src/mavros/mavros/include/mavros/mavros_uas.hpp:44,
                 from /home/shashankrajivmoghe/ros/src/mavros/mavros/src/lib/plugin.cpp:16:
/home/shashankrajivmoghe/ros/src/mavros/mavros/include/mavros/utils.hpp:27:10: fatal error: mavros_msgs/mavlink_convert.hpp: No such file or directory
   27 | #include "mavros_msgs/mavlink_convert.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/mavros.dir/build.make:160: CMakeFiles/mavros.dir/src/lib/plugin.cpp.o] Error 1
In file included from /home/shashankrajivmoghe/ros/src/mavros/mavros/include/mavros/mavros_uas.hpp:44,
                 from /home/shashankrajivmoghe/ros/src/mavros/mavros/src/lib/uas_data.cpp:19:
/home/shashankrajivmoghe/ros/src/mavros/mavros/include/mavros/utils.hpp:27:10: fatal error: mavros_msgs/mavlink_convert.hpp: No such file or directory
   27 | #include "mavros_msgs/mavlink_convert.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/mavros.dir/build.make:188: CMakeFiles/mavros.dir/src/lib/uas_data.cpp.o] Error 1
In file included from /home/shashankrajivmoghe/ros/src/mavros/mavros/include/mavros/mavros_uas.hpp:44,
                 from /home/shashankrajivmoghe/ros/src/mavros/mavros/src/lib/uas_ap.cpp:18:
/home/shashankrajivmoghe/ros/src/mavros/mavros/include/mavros/utils.hpp:27:10: fatal error: mavros_msgs/mavlink_convert.hpp: No such file or directory
[... cut ...]
   27 | #include "mavros_msgs/mavlink_convert.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/mavros.dir/build.make:146: CMakeFiles/mavros.dir/src/lib/mavros_uas.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:192: CMakeFiles/mavros.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< mavros [7.20s, exited with code 2]

Summary: 0 packages finished [7.34s]
  1 package failed: mavros
  1 package had stderr output: mavros
```

vooon, 2022-06-27:
Hmm, yes, can't main can't find include.
```
In file included from /home/ubuntu/ros2_ws/src/mavros/mavros/src/lib/enum_sensor_orientation.cpp:21:
/home/ubuntu/ros2_ws/src/mavros/mavros/include/mavros/utils.hpp:27:10: fatal error: mavros_msgs/mavlink_convert.hpp: No such file or directory
   27 | #include "mavros_msgs/mavlink_convert.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/mavros.dir/build.make:76: CMakeFiles/mavros.dir/src/lib/enum_sensor_orientation.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /home/ubuntu/ros2_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:21:
/home/ubuntu/ros2_ws/src/mavros/mavros/include/mavros/utils.hpp:27:10: fatal error: mavros_msgs/mavlink_convert.hpp: No such file or directory
   27 | #include "mavros_msgs/mavlink_convert.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/mavros.dir/build.make:90: CMakeFiles/mavros.dir/src/lib/enum_to_string.cpp.o] Error 1
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'author-email' will not be supported in future versions. Please use the underscore name 'author_email' instead
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' instead
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
  warnings.warn(
gmake[1]: *** [CMakeFiles/Makefile2:192: CMakeFiles/mavros.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< mavros [29.7s, exited with code 2]

Summary: 3 packages finished [5min 44s]
  1 package failed: mavros
  2 packages had stderr output: mavlink mavros
  1 package not processed
```

vooon, 2022-06-27:
@Macavitycode i've released fixed mavlink 2022.6.27 for humble. Trying to build mavros now. Looks like something breaks in rclcpp when C++20 is enabled.

https://github.com/mavlink/mavlink-gbp-release/blob/master/patch/pymavlink/generator/mavgen_cpp11.py#L44-L47

Macavitycode, 2022-06-28:
Thank you so much for your help, new release fixed it. I have submitted a PR for mavgen on pymavlink [here](https://github.com/ArduPilot/pymavlink/pull/698) I will continue on that thread to try getting it accepted. :)

closed 2022-06-28 by Macavitycode