mavlink/mavros #1762: How to change EKF origin using Mavros

maxBHarris, 2022-06-30:
I have been trying to find a way to change the EKF origin using mavros. I think there is a MavLink message SET_GPS_GLOBAL_ORIGIN that in theory changes the EKF origin of the copter. However, I don't know how to access it using mavros. I have been able to get Mavlink services working well, but have yet to be able to get a Mavlink message to work. Is it still considered a service in mavros? What topic does it fall under? Any help would be much appreciated. Thank you.

vooon, 2022-06-30:
https://github.com/mavlink/mavros/blob/686bd833e7d6ea5542977178872762dfbec5ed89/mavros/src/plugins/global_position.cpp#L130-L132

maxBHarris, 2022-06-30:
ok thank you. I will test it and let you know.

muakcal, 2023-11-15:
@maxBHarris did you figure this out?

closed 2026-10-25 by github-actions[bot]