mavlink/mavros #1763: MAVROS with SITL (ArduCopter) Broadcast timeout

JD-Florez, 2022-07-18:
### MAVROS version and platform

Mavros: 2.2.0
ROS: Foxy
Ubuntu: 20.04

### Autopilot type and version

[ X ] ArduPilot ver. 4.1.5
[  ] PX4

### Issue details

To preface, I am new to MAVROS and ROS2 in general. I made a basic drone controller that interfaces with ArduCopter SITL, and I have been trying to switch from Dronekit to MAVROS so that I can eventually run simulations in Gazebo. However, I am having trouble connecting to the drone. I am trying to, at a basic level, query (echo) the vehicle states (e.g. /mavros/altitude).

I saw another issue posted about launch files not working properly. Could someone please direct me with the basic commands or setup I would need to be able to connect properly to SITL? Currently I am trying the following:

`ros2 run mavros mavros_node --ros-args --param fcu_url:=udp://127.0.0.1:14550@`

and my output is the following:

### Node logs

```
[INFO] [1658161349.872866019] [mavros_node]: Starting mavros_node container
[INFO] [1658161349.876081218] [mavros_node]: FCU URL: udp://127.0.0.1:14550@
[INFO] [1658161349.876840414] [mavros_node]: GCS URL: 
[INFO] [1658161349.876900666] [mavros_node]: UAS Prefix: /uas1
[INFO] [1658161349.877401500] [mavros_node]: Starting mavros router node
[INFO] [1658161349.916197640] [mavros_router]: Built-in SIMD instructions: SSE, SSE2
[INFO] [1658161349.920561486] [mavros_router]: Built-in MAVLink package version: 2022.3.3
[INFO] [1658161349.920635114] [mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[INFO] [1658161349.920692962] [mavros_router]: MAVROS Router started
[INFO] [1658161349.920878552] [mavros_router]: Requested to add endpoint: type: 0, url: udp://127.0.0.1:14550@
[INFO] [1658161349.921270239] [mavros_router]: Endpoint link[1000] created
[INFO] [1658161349.936315559] [mavros_router]: link[1000] opened successfully
[INFO] [1658161349.939789688] [mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[INFO] [1658161349.943226709] [mavros_router]: Endpoint link[1001] created
[INFO] [1658161349.948734921] [mavros_router]: link[1001] opened successfully
[INFO] [1658161349.949163758] [mavros_node]: Starting mavros uas node
[INFO] [1658161350.039183581] [mavros_router]: link[1000] detected remote address 1.1
[INFO] [1658161350.128212287] [mavros]: UAS Executor started, threads: 4
[INFO] [1658161350.183826966] [mavros]: Plugin actuator_control created
[INFO] [1658161350.184302115] [mavros]: Plugin actuator_control initialized
[INFO] [1658161350.261260721] [mavros]: Plugin adsb created
[INFO] [1658161350.261702568] [mavros]: Plugin adsb initialized
[INFO] [1658161350.291555657] [mavros]: Plugin altitude created
[INFO] [1658161350.295747126] [mavros]: Plugin altitude initialized
[INFO] [1658161350.322251372] [mavros]: Plugin cam_imu_sync created
[INFO] [1658161350.327278274] [mavros]: Plugin cam_imu_sync initialized
[INFO] [1658161350.361500362] [mavros]: Plugin camera created
[INFO] [1658161350.364219517] [mavros]: Plugin camera initialized
[INFO] [1658161350.402424881] [mavros]: Plugin cellular_status created
[INFO] [1658161350.402611811] [mavros]: Plugin cellular_status initialized
[INFO] [1658161350.479756978] [mavros]: Plugin command created
[INFO] [1658161350.485651665] [mavros]: Plugin command initialized
[INFO] [1658161350.508787188] [mavros]: Plugin companion_process_status created
[INFO] [1658161350.508970881] [mavros]: Plugin companion_process_status initialized
[INFO] [1658161350.564256630] [mavros]: Plugin debug_value created
[INFO] [1658161350.567371727] [mavros]: Plugin debug_value initialized
[INFO] [1658161350.616485128] [mavros.distance_sensor]: DS: Plugin not configured!
[INFO] [1658161350.619299220] [mavros]: Plugin distance_sensor created
[INFO] [1658161350.622543776] [mavros]: Plugin distance_sensor initialized
[INFO] [1658161350.651783909] [mavros]: Plugin esc_status created
[INFO] [1658161350.652185073] [mavros]: Plugin esc_status initialized
[INFO] [1658161350.698958104] [mavros]: Plugin esc_telemetry created
[INFO] [1658161350.699491281] [mavros]: Plugin esc_telemetry initialized
[INFO] [1658161350.793811336] [mavros]: Plugin fake_gps created
[INFO] [1658161350.798635865] [mavros]: Plugin fake_gps initialized
[INFO] [1658161351.196220393] [mavros]: Plugin ftp created
[... 101 lines cut ...]
[WARN] [1658161360.821030116] [mavros.imu]: IMU: ~imu/data_raw stores unscaled raw acceleration report.
[INFO] [1658161360.922237319] [mavros]: CON: Got HEARTBEAT, connected. FCU: ArduPilot
[INFO] [1658161360.924240088] [mavros.mission]: WP: detected enable_partial_push: 1
[INFO] [1658161361.039443074] [mavros.rc]: RC_CHANNELS message detected!
[INFO] [1658161361.043325579] [mavros.imu]: IMU: Raw IMU message used.
[INFO] [1658161361.586315961] [mavros_router]: link[1001] detected remote address 1.191
[WARN] [1658161361.935703352] [mavros.cmd]: CMD: Unexpected command 520, result 0
[INFO] [1658161361.936036280] [mavros.geofence]: GF: Using MISSION_ITEM_INT
[INFO] [1658161361.936127268] [mavros.rallypoint]: RP: Using MISSION_ITEM_INT
[INFO] [1658161361.936199593] [mavros.mission]: WP: Using MISSION_ITEM_INT
[INFO] [1658161361.936324865] [mavros.sys]: VER: 1.1: Capabilities         0x000000000000fbef
[INFO] [1658161361.936417652] [mavros.sys]: VER: 1.1: Flight software:     040105ff (1002fd6e)
[INFO] [1658161361.936480076] [mavros.sys]: VER: 1.1: Middleware software: 00000000 (        )
[INFO] [1658161361.936538736] [mavros.sys]: VER: 1.1: OS software:         00000000 (        )
[INFO] [1658161361.936598722] [mavros.sys]: VER: 1.1: Board hardware:      00000000
[INFO] [1658161361.936659986] [mavros.sys]: VER: 1.1: VID/PID:             0000:0000
[INFO] [1658161361.936719095] [mavros.sys]: VER: 1.1: UID:                 0000000000000000
[INFO] [1658161370.921576865] [mavros.home_position]: HP: requesting home position
[INFO] [1658161370.925442222] [mavros.sys]: FCU: ArduCopter V4.1.5 (1002fd6e)
[INFO] [1658161370.929271284] [mavros.sys]: FCU: 48e7fc3aa4954f5db8be2ebdc643733d
[INFO] [1658161370.929449297] [mavros.sys]: FCU: Frame: QUAD/PLUS
[INFO] [1658161374.516339703] [mavros.param]: PR: parameters list received
[INFO] [1658161376.002710337] [mavros.mission]: WP: mission received
[INFO] [1658161380.927544521] [mavros.rallypoint]: RP: mission received
[INFO] [1658161385.921111772] [mavros.geofence]: GF: mission received```

vooon, 2022-07-18:
Log looks ok. What is the problem? What `ros2 topic list` shows? What happen if you subscribe to the status?
Is `ros2 run mavros mav checkid` shows?

JD-Florez, 2022-07-19:
This is my output following the code you posted:

Python 3.8.10 (default, Jun 22 2022, 20:18:18) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.

>>> import rclpy
>>> from rclpy.node import Node
>>> rclpy.init()
>>> import mavros
>>> c = mavros.Client()
>>> c.start_spinner()
<Thread(mavros_py_spin_mavpy_a6tz, started daemon 140455739983616)>
>>> c.system.subscribe_state(lambda m: print(m)
... )
<rclpy.subscription.Subscription object at 0x7fbe668d1fa0>

Sorry I am a bit new to this!

JD-Florez, 2022-07-19:
I ran the following: `ros2 run mavros mavros_node --ros-args --params-file mavros_param.yaml`

With the following yaml file:
~~~~
mavros:
  ros__parameters: {}

mavros_router:
  ros__parameters: {}

mavros_node:
  ros__parameters:
    fcu_url: udp://127.0.0.1:14550@
    gcs_url: tcp-l://:5670
~~~~

and got the following output when running 'ros2 topic list'

~~~~
/diagnostics
/mavros/actuator_control
/mavros/altitude
/mavros/battery
/mavros/estimator_status
/mavros/extended_state
/mavros/geofence/fences
/mavros/global_position/compass_hdg
/mavros/global_position/global
/mavros/global_position/gp_lp_offset
/mavros/global_position/gp_origin
/mavros/global_position/local
/mavros/global_position/raw/fix
/mavros/global_position/raw/gps_vel
/mavros/global_position/raw/satellites
/mavros/global_position/rel_alt
/mavros/global_position/set_gp_origin
/mavros/home_position/home
/mavros/home_position/set
/mavros/imu/data
/mavros/imu/data_raw
/mavros/imu/diff_pressure
/mavros/imu/mag
/mavros/imu/static_pressure
/mavros/imu/temperature_baro
/mavros/imu/temperature_imu
/mavros/local_position/accel
/mavros/local_position/odom
/mavros/local_position/pose
/mavros/local_position/pose_cov
/mavros/local_position/velocity_body
/mavros/local_position/velocity_body_cov
/mavros/local_position/velocity_local
/mavros/manual_control/control
/mavros/manual_control/send
/mavros/mission/reached
/mavros/mission/waypoints
/mavros/nav_controller_output/output
/mavros/param/event
/mavros/rallypoint/rallypoints
/mavros/rc/in
/mavros/rc/out
/mavros/rc/override
/mavros/setpoint_accel/accel
/mavros/setpoint_attitude/cmd_vel
/mavros/setpoint_attitude/thrust
/mavros/setpoint_position/global
/mavros/setpoint_position/global_to_local
/mavros/setpoint_position/local
/mavros/setpoint_raw/attitude
/mavros/setpoint_raw/global
/mavros/setpoint_raw/local
/mavros/setpoint_raw/target_attitude
/mavros/setpoint_raw/target_global
/mavros/setpoint_raw/target_local
/mavros/setpoint_trajectory/desired
/mavros/setpoint_trajectory/local
/mavros/setpoint_velocity/cmd_vel
/mavros/setpoint_velocity/cmd_vel_unstamped
/mavros/state
/mavros/statustext/recv
/mavros/statustext/send
/mavros/target_actuator_control
/mavros/time_reference
/mavros/timesync_status
/mavros/wind_estimation
/parameter_events
/rosout
/tf
/tf_static
/uas1/mavlink_sink
/uas1/mavlink_source
~~~~

vooon, 2022-07-20:
`ros2 topic echo /mavros/state`?

But looks like not all topics are advertised.

JD-Florez, 2022-07-20:
My output from echoring '/mavros/state' was:
~~~~
---
header:
  stamp:
    sec: 1658340088
    nanosec: 365302938
  frame_id: ''
connected: true
armed: false
guided: false
manual_input: true
mode: STABILIZE
system_status: 3
---
~~~~

My impression is that the topics are being instantiated but are just empty/no being populated. Do you have any example sampling nodes or something I might try to run? I can currently read some topics but not all. For example, I can now read 'mavros/global_position/local' as:

~~~~
header:
  stamp:
    sec: 1658340359
    nanosec: 38116195
  frame_id: map
child_frame_id: base_link
pose:
  pose:
    position:
      x: 0.03635660492855464
      y: -0.03328734375240672
      z: 2.001
    orientation:
      x: 0.0027573371182975466
      y: -0.00022430042808749436
      z: -0.7526145528990368
      w: -0.6584555273865631
  covariance:
  - 0.09
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.09
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.09
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 99999.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 99999.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 99999.0
twist:
  twist:
    linear:
      x: -0.01
      y: 0.01
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: 0.0
  covariance:
  - -1.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
---
~~~~

But I can't get a readout for most other topics.

vooon, 2022-07-22:
I suggest first to limit the footprint by allowing only small subset of plugins to load. I.e.
```
mavros:
  ros_parameters:
    plugin_denylist: ['*']
    plugin_allowlist:
      - sys_*
      - some_plugin

lidorshimoni, 2022-07-23:
Hi, Im having a similiar problem. All of the topics are shown in ros2 topic list  and rqt but only 2-4 of them are published.
things I tried to do in order to solve it:
- running with QGroundControl open in the background
- calling /mavros/set_stream_rate with relevant parameters to make sure mavros will publis (this had effect on only 2 topics)
- porting the launch file to python and making new param files

I really have no clue on why the topics are not publishing anything.

lidorshimoni, 2022-07-23:
My published topics:
```
- /diagnostics
- /mavros/gpsstatus/gps1/raw
- /mavros/home_position/home
- /mavros/nav_controller_output/output
- /mavros/rc/in
- /mavros/rc/out
- /mavros/state
- /mavros/terrain/report
- /mavros/tunnel/out
- /mavros/vfr_hud
```

JD-Florez, 2022-07-27:
I was able to get my outputs after careful reinstall and reconfigure!

closed 2022-07-27 by JD-Florez

lidorshimoni, 2022-07-27:
@JD-Florez should I reinstall mavros? ros? what exactly was causing the problem?

lidorshimoni, 2022-07-27:
I just reinstalled ROS and all dependencies. also cloned the most recent mavros repo from ros2 branch. the results were the same.
`/mavros/state `and `/mavros/battery` are published while `/mavros/altitude` and everything else wont be published.
maybe it's related to ardupilot behavior?

lidorshimoni, 2022-07-30:
Well, that's weird. 
`ros2 topic echo "/mavros/local_position/odom" ` works and show the published data, while the same topic shows no data on rqt. It probably happens in more topics.

lidorshimoni, 2022-08-07:
My problem was in the DDS QoS profile. I did not see any traffic in RQT but in `ros2 topic echo` it worked.