mavlink/mavros #1764: no data on /mavros/local_position/odom topic

Captain299792458, 2022-07-22:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

vooon, 2022-07-22:
Maybe you need  to setup the port? What `checkid` shows?

Captain299792458, 2022-07-22:
> Maybe you need to setup the port? What `checkid` shows?

I have run it. target system id is not set used default id value 1

Also, 
I have performed set_stream_rate and /mavros/imu/data is streaming but /mavros/local_position/pose is not streaming any data. I have autopilot indoors and GPS fix is 0. Should i wait for GPS fix to be 3D to get data from topic mavros/local_position/pose

vooon, 2022-07-22:
Yes, i suppose fw will send position only when it'll get fix. That could be changed by fw setup + vision pose source.

Captain299792458, 2022-07-23:
setting stream rate solves the problem. thanks for your replies.

closed 2022-07-23 by Captain299792458