mavlink/mavros #1766: Mavros Offboard Control Velocity Setpoint Doesn't Hold.

DEADLYBBS, 2022-07-28:
### Issue details
Velocity setpoints sent via python mavros are 'jumpy'. I send the command but then it jumps to a different value.
https://review.px4.io/plot_app?log=5a0d4421-9e1f-43b5-aee3-bc40eeff4097
This can be seen in the above flight log. 
In the python files, the velocity setpoints are sent with the function vel_raw. These are posted to the /mavors/setpoint_raw/local topic.
My typemask should be set up to allow only velocity and yaw rate setpoints.

### MAVROS version and platform
Mavros: 2.0.0, possibly 2.1.0 (I installed it May 31st)
ROS: Noetic
Ubuntu: 18.04

### Autopilot type and version
 PX4 v1.13.0

### Node logs
[roslaunch_info.txt](https://github.com/mavlink/mavros/files/9211831/roslaunch_info.txt)

### Diagnostics
[ros diagnostics.txt](https://github.com/mavlink/mavros/files/9211812/ros.diagnostics.txt)

### Check ID
I could not get this to work. Fairly new to ros and linux. It wanted me to install rosbash but that would not install and I figured I could troubleshoot it for the day or finish this post.
```
rosrun mavros checkid
```

### mavros python files
[zippy.zip](https://github.com/mavlink/mavros/files/9211841/zippy.zip)

### Hardware
-pixracer v1
-esp8266 wifi chip (station mode)
-optitrack mocap using vrpn_client_ros
-offboard laptop running python to send velocity setpoints
-only wireless connection is drone to the router

Jaeyoung-Lim, 2022-07-31:
[image]

@DEADLYBBS It seems like you are actually sending discontinuous velocity setpoints to the vehicle. Have you checked what setpoints you are sending?

DEADLYBBS, 2022-08-01:
I realized that with the code provided, I was accidently sending multiple setpoints while not actually meaning to send multiple setpoints just by creating separate threads for them.