mavlink/mavros #1767: mavlink missing develop dialect

Jaeyoung-Lim, 2022-07-31:
### Issue details
I wasn't sure if this is the right repository, but posting it here for @vooon 's attention

There has not been a `mavlink-gbp-release` for Ubuntu focal for a while now, which means that the `develop` dialect is missing for the mavlink package released over ROS

This is a problem for repositories such as https://github.com/PX4/PX4-SITL_gazebo/pull/840 that uses the `develop` dialect.
```
Errors << mavlink_sitl_gazebo:make /github/home/catkin_ws/logs/mavlink_sitl_gazebo/build.make.000.log
In file included from /github/home/catkin_ws/src/PX4-SITL_gazebo/src/gazebo_camera_manager_plugin.cpp:18:
/github/home/catkin_ws/src/PX4-SITL_gazebo/include/gazebo_camera_manager_plugin.h:21:10: fatal error: development/mavlink.h: No such file or directory
   21 | #include <development/mavlink.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/gazebo_camera_manager_plugin.dir/build.make:63: CMakeFiles/gazebo_camera_manager_plugin.dir/src/gazebo_camera_manager_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:776: CMakeFiles/gazebo_camera_manager_plugin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:152: all] Error 2
cd /github/home/catkin_ws/build/mavlink_sitl_gazebo; catkin build --get-env mavlink_sitl_gazebo | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
```
@vooon Would it be possible for a new release on the `mavlink-gbp-release` so that it contains the `develop` dialect?

### MAVROS version and platform

Mavros: NA
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[  ] ArduPilot
[ X] PX4

### Problem Description

vooon, 2022-08-08:
https://github.com/ros/rosdistro/pull/34088
https://github.com/ros/rosdistro/pull/34087
https://github.com/ros/rosdistro/pull/34086

Jaeyoung-Lim, 2022-08-08:
@vooon Thanks!

closed 2022-08-22 by Jaeyoung-Lim