mavlink/mavros #1768: Detect heartbeat from GCS

esharet, 2022-07-31:
### Issue details

I'm trying to detect if I get TYPE_GCS heartbeat. How can I do it with mavros? I wonder if sys_status plugin may help but I didn't understand how to use it. 
Thank you,

### MAVROS version and platform

Mavros: 2.2.0
ROS: Foxy
Ubuntu: 20.04

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: 4.2.2 

### Node logs

```
[mavros_node-1] [INFO] [1659267737.277853467] [mavros_node]: Starting mavros_node container
[mavros_node-1] [INFO] [1659267737.277910084] [mavros_node]: FCU URL: udp://:14551@14555
[mavros_node-1] [INFO] [1659267737.277916927] [mavros_node]: GCS URL: udp://@
[mavros_node-1] [INFO] [1659267737.277921854] [mavros_node]: UAS Prefix: /uas1
[mavros_node-1] [INFO] [1659267737.277925755] [mavros_node]: Starting mavros router node
[mavros_node-1] [INFO] [1659267737.280637353] [mavros_router]: Built-in SIMD instructions: SSE, SSE2
[mavros_node-1] [INFO] [1659267737.280725646] [mavros_router]: Built-in MAVLink package version: 2022.3.3
[mavros_node-1] [INFO] [1659267737.280731950] [mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[mavros_node-1] [INFO] [1659267737.280736951] [mavros_router]: MAVROS Router started
[mavros_node-1] [INFO] [1659267737.280815828] [mavros_router]: Requested to add endpoint: type: 0, url: udp://:14551@14555
[mavros_node-1] [INFO] [1659267737.280990750] [mavros_router]: Endpoint link[1000] created
[mavros_node-1] [INFO] [1659267737.281970250] [mavros_router]: link[1000] opened successfully
[mavros_node-1] [INFO] [1659267737.282044631] [mavros_router]: Requested to add endpoint: type: 1, url: udp://@
[mavros_node-1] [INFO] [1659267737.282091129] [mavros_router]: Endpoint link[1001] created
[mavros_node-1] [INFO] [1659267737.282195402] [mavros_router]: link[1001] opened successfully
[mavros_node-1] [INFO] [1659267737.282230315] [mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[mavros_node-1] [INFO] [1659267737.282284507] [mavros_router]: Endpoint link[1002] created
[mavros_node-1] [INFO] [1659267737.283167694] [mavros_router]: link[1002] opened successfully
[mavros_node-1] [INFO] [1659267737.283207972] [mavros_node]: Starting mavros uas node
[mavros_node-1] [INFO] [1659267737.314103229] [mavros]: UAS Executor started, threads: 12
[mavros_node-1] [INFO] [1659267737.327913996] [mavros]: Plugin actuator_control created
[mavros_node-1] [INFO] [1659267737.328178340] [mavros]: Plugin actuator_control initialized
[mavros_node-1] [INFO] [1659267737.340732781] [mavros]: Plugin adsb created
[mavros_node-1] [INFO] [1659267737.340912445] [mavros]: Plugin adsb initialized
[mavros_node-1] [INFO] [1659267737.343598249] [mavros]: Plugin altitude created
[mavros_node-1] [INFO] [1659267737.343722644] [mavros]: Plugin altitude initialized
[mavros_node-1] [INFO] [1659267737.346358893] [mavros]: Plugin cam_imu_sync created
[mavros_node-1] [INFO] [1659267737.346482003] [mavros]: Plugin cam_imu_sync initialized
[mavros_node-1] [INFO] [1659267737.349119995] [mavros]: Plugin camera created
[mavros_node-1] [INFO] [1659267737.349256402] [mavros]: Plugin camera initialized
[mavros_node-1] [INFO] [1659267737.355345225] [mavros]: Plugin command created
[mavros_node-1] [INFO] [1659267737.355483320] [mavros]: Plugin command initialized
[mavros_node-1] [INFO] [1659267737.358410431] [mavros]: Plugin companion_process_status created
[mavros_node-1] [INFO] [1659267737.358443127] [mavros]: Plugin companion_process_status initialized
[mavros_node-1] [INFO] [1659267737.362086590] [mavros]: Plugin debug_value created
[mavros_node-1] [INFO] [1659267737.362361229] [mavros]: Plugin debug_value initialized
[mavros_node-1] [INFO] [1659267737.364903317] [mavros_router]: link[1000] detected remote address 1.1
[mavros_node-1] [INFO] [1659267737.365181086] [mavros.distance_sensor]: DS: Plugin not configured!
[mavros_node-1] [INFO] [1659267737.365289551] [mavros]: Plugin distance_sensor created
[mavros_node-1] [INFO] [1659267737.365422924] [mavros]: Plugin distance_sensor initialized
[mavros_node-1] [INFO] [1659267737.369034730] [mavros]: Plugin esc_status created
[mavros_node-1] [INFO] [1659267737.369204456] [mavros]: Plugin esc_status initialized
[mavros_node-1] [INFO] [1659267737.372875069] [mavros]: Plugin esc_telemetry created
[mavros_node-1] [INFO] [1659267737.373120040] [mavros]: Plugin esc_telemetry initialized
[mavros_node-1] [INFO] [1659267737.379072694] [mavros]: Plugin fake_gps created
[mavros_node-1] [INFO] [1659267737.379116971] [mavros]: Plugin fake_gps initialized
[mavros_node-1] [INFO] [1659267737.388748764] [mavros]: Plugin ftp created
[mavros_node-1] [INFO] [1659267737.388883196] [mavros]: Plugin ftp initialized
[mavros_node-1] [INFO] [1659267737.394104245] [mavros]: Plugin geofence created
[... 188 lines cut ...]
  - key: Remaining
    value: '89.0'
- level: "\0"
  name: 'mavros: Heartbeat'
  message: Normal
  hardware_id: uas:///uas1
  values:
  - key: Heartbeats since startup
    value: '239'
  - key: Frequency (Hz)
    value: '1.000000'
  - key: Vehicle type
    value: Ground rover
  - key: Autopilot type
    value: ArduPilot
  - key: Mode
    value: MANUAL
  - key: System status
    value: ACTIVE
```

### Check ID

```
Router topic: /uas1/mavlink_source, target: 1.1```

vooon, 2022-08-01:
@esharet mavros doesn't have plugin that checks for heartbeats from the ground station.

Plugin `sys_status` has a SystemAndOk filter on HEARTBEAT handler, so it'll filter any other sources.
You may use that plugin as a reference to write your custom one.

Please note that i do not see GCS system in checkid. Did you connect with GCS during test?

esharet, 2022-08-02:
I don't remember if I checked with the GCS during the test. I did it again with QGC connected and got:
```
Router topic: /uas1/mavlink_source, target: 1.1
OK. I got messages from 1.1.
---
Received 1531, from 1 addresses
address   list of messages
    1.1   0 (HEARTBEAT), 1 (SYS_STATUS), 2 (SYSTEM_TIME), 129 (SCALED_IMU3), 137 (SCALED_PRESSURE2), 147 (BATTERY_STATUS), 22 (PARAM_VALUE), 152, 24 (GPS_RAW_INT), 27 (RAW_IMU), 29 (SCALED_PRESSURE), 30 (ATTITUDE), 32 (LOCAL_POSITION_NED), 33 (GLOBAL_POSITION_INT), 34 (RC_CHANNELS_SCALED), 163, 164, 165, 36 (SERVO_OUTPUT_RAW), 42 (MISSION_CURRENT), 178, 193, 65 (RC_CHANNELS), 74 (VFR_HUD), 111 (TIMESYNC), 241 (VIBRATION), 242 (HOME_POSITION), 116 (SCALED_IMU2), 125 (POWER_STATUS)
```

I don't know yet how to work with a mavros plugin but I will check it now. Thank you!

vooon, 2022-08-02:
Still doesn't look like it have GCS HEARTBEATs. Please look at router logs, there should be lines like that:
```
[mavros_router]: link[1002] detected remote address 1.191
```

Also try to catch more `/diagnostics` messages, router sends active endpoints there (and addresses, SYS.COMP).

Look at `dummy.cpp`, that is an example how to write a plugin.

esharet, 2022-09-07:
I used the example of dummy.cpp plugin and it worked for me. 
Thank you!

closed 2022-09-07 by vooon