mavlink/mavros #1769: "Unexpected command 176" when setting OFFBOARD-mode with PX4 in Ignition Gazebo via service /mavros/set_mode

snimu, 2022-08-02:
### Issue details

Setup: Try to set mode to OFFBOARD via service `/mavros/set_mode`.

Issue: Get warning `[WARN] [1659439086.350950582] [mavros.cmd]: CMD: Unexpected command 176, result 1`

Might be caused by [issue #19999](https://github.com/PX4/PX4-Autopilot/issues/19999), but could also be additional error on top.

### MAVROS version and platform

Mavros: Package: ros-foxy-mavros, Version: 2.1.1-1focal.20220303.113808
ROS: Foxy
Ubuntu: 20.04.4 LTS

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: v1.13.0

### Node logs

Full output on **PX4**- and **mavros**-terminal:

```bash
faps@faps:~/ros_ws$ ros2 launch intrafly_essentials intrafly_essentials.launch.xml 
[INFO] [launch]: All log files can be found below /home/faps/.ros/log/2022-08-02-13-17-29-336203-faps-2055132
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [2055134]
[INFO] [create-2]: process started with pid [2055137]
[INFO] [export-3]: process started with pid [2055139]
[INFO] [export-3]: process has finished cleanly [pid 2055139]
[INFO] [bash-4]: process started with pid [2055141]
[INFO] [mavros-5]: process started with pid [2055143]
[bash-4] SITL ARGS
[bash-4] sitl_bin: /home/faps/PX4-Autopilot/build/px4_sitl_default/bin/px4
[bash-4] debugger: none
[bash-4] program: ignition
[bash-4] model: iris
[bash-4] world: base_world.sdf
[bash-4] src_path: /home/faps/PX4-Autopilot
[bash-4] build_path: /home/faps/PX4-Autopilot/build/px4_sitl_default
[bash-4] Ignition Gazebo
[bash-4] LD_LIBRARY_PATH /home/faps/ros_ws/install/vision_msgs/lib:/home/faps/ros_ws/install/intrafly_essentials_interfaces/lib:/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib:/home/faps/PX4-Autopilot//build/px4_sitl_default/build_ign_gazebo:/home/faps/PX4-Autopilot/build/px4_sitl_default/build_ign_gazebo
[bash-4] SITL COMMAND: "/home/faps/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/faps/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/faps/PX4-Autopilot"/test_data
[bash-4] Creating symlink /home/faps/PX4-Autopilot/build/px4_sitl_default/etc -> /home/faps/PX4-Autopilot/build/px4_sitl_default/tmp/ro
[bash-4] 
[bash-4] ______  __   __    ___ 
[bash-4] | ___ \ \ \ / /   /   |
[bash-4] | |_/ /  \ V /   / /| |
[bash-4] |  __/   /   \  / /_| |
[bash-4] | |     / /^\ \ \___  |
[bash-4] \_|     \/   \/     |_/
[bash-4] 
[bash-4] px4 starting.
[bash-4] 
[bash-4] INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
[bash-4] INFO  [init] found model autostart file as SYS_AUTOSTART=10016
[bash-4] INFO  [param] selected parameter default file eeprom/parameters_10016
[bash-4] INFO  [parameters] BSON document size 312 bytes, decoded 312 bytes (INT32:13, FLOAT:3)
[bash-4] [param] Loaded: eeprom/parameters_10016
[bash-4] INFO  [dataman] data manager file './dataman' size is 7866640 bytes
[bash-4] PX4 SIM HOST: localhost
[bash-4] INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[mavros-5] [INFO] [1659439050.085920639] [mavros_node]: Starting mavros_node container
[mavros-5] [INFO] [1659439050.086003626] [mavros_node]: FCU URL: udp://:14540@127.0.0.1:14580
[mavros-5] [INFO] [1659439050.086011602] [mavros_node]: GCS URL: 
[mavros-5] [INFO] [1659439050.086016321] [mavros_node]: UAS Prefix: /uas1
[mavros-5] [INFO] [1659439050.086020256] [mavros_node]: Starting mavros router node
[... 281 lines cut ...]
  name: 'mavros: Heartbeat'
  message: Normal
  hardware_id: uas:///uas1
  values:
  - key: Heartbeats since startup
    value: '31'
  - key: Frequency (Hz)
    value: '0.599975'
  - key: Vehicle type
    value: Quadrotor
  - key: Autopilot type
    value: PX4 Autopilot
  - key: Mode
    value: AUTO.LOITER
  - key: System status
    value: STANDBY
---

```

### Check ID

On **ROS2**, *mavros checkid* is not an executable (if I simply missed it, I would be happy to provide the output).

Instead, here is [the flight log](https://logs.px4.io/plot_app?log=6d847a4c-2b4b-4116-81d9-b6c9c7542799) from **PX4**.

O2KAN, 2022-09-07:
Have you found a solution?

snimu, 2022-09-15:
Unfortunately, no. I've come upon this issue during my student job and haven't worked on the project for a while now. Sorry.

zinuok, 2022-10-17:
@snimu what is the package named 'intrafly_essentials' you used? 
It seems that you can run the SITL on ROS2, 
but I can't find the 'intrafly_essentials' package

snimu, 2022-10-17:
@zinuok 'intrafly_essentials' is an internal package. I'm pretty sure that the fault wasn't in that package, but it might have been some mistake in the QGroundControl-settings or something along those lines. I believe it would be appropriate to close this issue. Sorry for the inconvenience.