mavlink/mavros #1773: Odometry plugin

pedro-fuoco, 2022-08-20:
### Issue details
Hello, I would like to know when the Odometry plugin will be ported to ROS2, me and my team will be participating a drone competition and would like to use [our version](https://github.com/SkyRats/realsense_mav_bridge) of Autherion's VIO plugin in ROS2 .
### MAVROS version and platform

Mavros: ?2.2.0?
ROS: ?fFoxy?
Ubuntu: ?20.04?

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: ?1.13.0?

### Node logs

[mavros_node-1] [ERROR] [1660956039.650630227] [mavros.odometry]: ODOM: Ex: Could not find a connection between 'map' and 'base_link_frd' because they are not part of the same tree.Tf has two or more unconnected trees.

pedro-fuoco, 2022-08-23:
Any ideia if its gonna take weeks or months ? The team needs to plan ahead.

vooon, 2022-08-23:
Have no idea, i don't have any test bed nor free time.

pedro-fuoco, 2022-08-24:
Thanks for the quick reply! We will be using ros noetic this year for the indoor competition then.

ripdk12, 2022-11-07:
Any updates on this? Or should I Use the ROS1 bridge for this feature?

kribe48, 2023-05-09:
Any updates on this? I only get the error if I change the namespace parameter from mavros to something else.

pedro-fuoco, 2023-05-09:
I don't have any updates regarding mavros but I'm currently developing external odometry support for Ardupilot's native DDS support Ardupilot/ardupilot#23387. Should be done this month

closed 2026-10-24 by github-actions[bot]