mavlink/mavros #1774: Could not find GeographicLib

acxz, 2022-08-21:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Cannot build
GeographicLib not found

```
CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find GeographicLib (missing: GeographicLib_LIBRARIES)
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /opt/ros/noetic/share/libmavconn/cmake/Modules/FindGeographicLib.cmake:14 (find_package_handle_standard_args)
  CMakeLists.txt:42 (find_package)

-- Configuring incomplete, errors occurred!
```

GeographicLib version: 2.1.1

### MAVROS version and platform

Mavros: 1.13.0-1
ROS: Noetic
OS: Archlinux

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

N/A

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```
N/A

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```
N/A
### Check ID

```
rosrun mavros checkid
```
N/A

vooon, 2022-08-23:
Fixed by #1775.

closed 2022-08-23 by vooon

acxz, 2022-08-23:
I this issue also affects the ros2 branch and should stay open until those commits find their way there.