mavlink/mavros #1781: mavftpfuse unable to run when called with rosrun

hanzale, 2022-09-07:
```rosrun mavros_extras mavftpfuse``` raises errors listed below 

----

### Issue details
While roscore & mavros are running with an active connection to an pixhawk I've called mav ftp fuse via rosrun. Posting errors and my fixes below.
My setup is: Ubuntu Focal, Python 3.9, ROS Noetic running on a conda enviroment;
mavros & mavros_extras installed via robostack conda channel's ```ros-noetic-mavros``` & ```ros-noetic-mavros-extras``` conda packages.

### MAVROS version and platform

Mavros: 1.13.0
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: ?

### Node logs

> #1
> /ros/lib/mavros_extras/mavftpfuse", line 41
>     except OSError, e:
> SyntaxError: invalid syntax
> 
>  -> changed 41 line to: except OSError as e:
> 
> #2
> /ros/lib/mavros_extras/mavftpfuse", line 80
>     dir_mode, file_mode = S_IFDIR | 0555, S_IFREG | 0444
>                                        ^
> SyntaxError: leading zeros in decimal integer literals are not permitted; use an 0o prefix for octal integers
> 
> -> changed leading '0's to '0o' in lines 80, 82, 109  
> 
> #3
> 
> ros/lib/mavros_extras/mavftpfuse", line 23, in <module>
>     from fuse import FUSE, FuseOSError, Operations, \
> ImportError: cannot import name 'ENOENT' from 'fuse' (/ros/lib/python3.9/site-packages/fuse.py)
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/ros/lib/mavros_extras/mavftpfuse", line 26, in <module>
>     print("Fuse wrapper module not found. Please install fusepy: pip install fusepy", file=os.stderr)
> 
> -> changed line 26, "file=os.stderr" argument into: "file=sys.stderr"
> 
> #4
> 
> ros/lib/mavros_extras/mavftpfuse", line 27, in <module>
>     os.exit(1)
> AttributeError: module 'os' has no attribute 'exit'
> 
> -> changed line 27 "os.exit(1)" into: "sys.exit(1)"
> 
> #5
> 
> Fuse wrapper module not found. Please install fusepy: pip install fusepy
> 
> -> deleted "ENOENT" from line 24
> -> added "from errno import ENOENT" as line 23
> 

### Diagnostics

```
header: 
  seq: 2645
  stamp: 
    secs: 1662566219
    nsecs: 774468006
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Received packets:"
        value: "28544"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "130"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "8849018"
      - 
        key: "Tx total bytes:"
        value: "1183605"
      - 
        key: "Rx speed:"
        value: "3683.000000"
      - 
        key: "Tx speed:"
        value: "460.000000"
  - 
    level: 2
    name: "mavros: GPS"
    message: "No satellites"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "Unknown"
      - 
        key: "EPV (m)"
        value: "Unknown"
  - 
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "3628"
      - 
        key: "Frequency (Hz)"
        value: "1.037053"
      - 
        key: "Vehicle type"
        value: "Ground rover"
      - 
        key: "Autopilot type"
        value: "ArduPilot"
      - 
        key: "Mode"
        value: "MANUAL"
      - 
        key: "System status"
        value: "Active"
  - 
    level: 2
    name: "mavros: System"
    message: "Sensor health"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Sensor present"
        value: "0x1330DC0F"
      - 
        key: "Sensor enabled"
        value: "0x0020800F"
      - 
        key: "Sensor health"
        value: "0x0310810F"
      - 
        key: "3D gyro"
        value: "Ok"
      - 
        key: "3D accelerometer"
        value: "Ok"
      - 
        key: "3D magnetometer"
        value: "Ok"
      - 
        key: "absolute pressure"
        value: "Ok"
      - 
        key: "motor outputs / control"
        value: "Ok"
      - 
        key: "AHRS subsystem health"
        value: "Fail"
      - 
        key: "CPU Load (%)"
        value: "10.2"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 1
    name: "mavros: Battery"
    message: "Low voltage"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Voltage"
        value: "0.00"
      - 
        key: "Current"
        value: "-0.0"
      - 
        key: "Remaining"
        value: "-1.0"
  - 
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "36195"
      - 
        key: "Frequency (Hz)"
        value: "10.000256"
      - 
        key: "Last RTT (ms)"
        value: "3.968089"
      - 
        key: "Mean RTT (ms)"
        value: "13.760457"
      - 
        key: "Last remote time (s)"
        value: "4155.733724000"
      - 
        key: "Estimated time offset (s)"
        value: "1662562064.023283482"
---

```

### Check ID

```
OK. I got messages from 1:1.

Received 1203 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 152, 24, 27, 29, 30, 33, 34, 163, 36, 165, 42, 178, 193, 65, 74, 77, 111, 241, 116, 125
```

hanzale, 2022-09-07:
Edited file as it works in my system.
[mavftpfuse.zip](https://github.com/mavlink/mavros/files/9507870/mavftpfuse.zip)

vooon, 2022-09-07:
Could you please open Merge Request?

closed 2026-10-24 by github-actions[bot]