mavlink/mavros #1782: PX4-SITL offboard mode does not set

O2KAN, 2022-09-07:
I can not use iris drone in PX4 SITL on offboard mode.
I have a code written with python using mavros to control iris drone in offboard mode. But when I set, command the mode to offboard drone does not takeoff or do what I want it to do. I also run exact same code with mavsdk, It works.

MAVROS version and platform
Mavros: I don't know how to look
ROS: ?Noetic?
Ubuntu: ?20.04?

Autopilot type and version
[ ] ArduPilot
[ * ] PX4

Version:

Node logs
roslaunch uav_mavros_sitl.launch
... logging to /home/ozkan/.ros/log/0e6952ee-2ed9-11ed-9296-45e5a2a5d154/roslaunch-ozkan-648552.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://ozkan:36277/

SUMMARY
CLEAR PARAMETERS

/uav0/mavros/
PARAMETERS

/gazebo/enable_ros_network: True
/rosdistro: noetic
/rosversion: 1.15.14
/uav0/mavros/cmd/use_comp_id_system_control: False
/uav0/mavros/conn/heartbeat_rate: 1.0
/uav0/mavros/conn/system_time_rate: 1.0
/uav0/mavros/conn/timeout: 10.0
/uav0/mavros/conn/timesync_rate: 10.0
/uav0/mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
/uav0/mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
/uav0/mavros/distance_sensor/hrlv_ez4_pub/id: 0
/uav0/mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
/uav0/mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
/uav0/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
/uav0/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
/uav0/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
/uav0/mavros/distance_sensor/laser_1_sub/id: 3
/uav0/mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
/uav0/mavros/distance_sensor/laser_1_sub/subscriber: True
/uav0/mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
/uav0/mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
/uav0/mavros/distance_sensor/lidarlite_pub/id: 1
/uav0/mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
/uav0/mavros/distance_sensor/lidarlite_pub/send_tf: True
/uav0/mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
/uav0/mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
/uav0/mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
/uav0/mavros/distance_sensor/sonar_1_sub/id: 2
/uav0/mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
/uav0/mavros/distance_sensor/sonar_1_sub/subscriber: True
/uav0/mavros/fake_gps/eph: 2.0
/uav0/mavros/fake_gps/epv: 2.0
/uav0/mavros/fake_gps/fix_type: 3
/uav0/mavros/fake_gps/geo_origin/alt: 408.0
/uav0/mavros/fake_gps/geo_origin/lat: 47.3667
/uav0/mavros/fake_gps/geo_origin/lon: 8.55
/uav0/mavros/fake_gps/gps_rate: 5.0
/uav0/mavros/fake_gps/mocap_transform: True
/uav0/mavros/fake_gps/satellites_visible: 5
/uav0/mavros/fake_gps/tf/child_frame_id: fix
[... 248 lines cut ...]
[ INFO] [1662574549.421536669, 2.156000000]: VER: 1.1: OS software: 050f00ff (ec20f2e6c5cc35b2)
[ INFO] [1662574549.421557141, 2.156000000]: VER: 1.1: Board hardware: 00000001
[ INFO] [1662574549.421577720, 2.156000000]: VER: 1.1: VID/PID: 0000:0000
[ INFO] [1662574549.421600529, 2.156000000]: VER: 1.1: UID: 4954414c44494e4f
INFO [ecl/EKF] 3116000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO [ecl/EKF] 3116000: reset position to last known position
INFO [ecl/EKF] 3116000: reset velocity to zero
[ INFO] [1662574550.564955049, 3.296000000]: IMU: Attitude quaternion IMU detected!
INFO [ecl/EKF] 5484000: reset position to GPS
INFO [ecl/EKF] 5484000: reset velocity to GPS
INFO [ecl/EKF] 5484000: starting GPS fusion

***************************after running the code ************************************

[ WARN] [1662575338.653643636, 790.320000000]: CMD: Unexpected command 176, result 1
INFO [commander] Armed by external command
WARN [commander] Failsafe enabled: No manual control stick input
INFO [commander] Failsafe mode activated
INFO [navigator] RTL HOME activated
INFO [navigator] RTL: landing at home position.
INFO [commander] Failsafe mode deactivated
INFO [commander] Disarmed by auto preflight disarming
INFO [logger] closed logfile, bytes written: 40289119

I need to fix this issue as soon as possible. Could you help?

leommiranda, 2022-09-07:
I ran into the same issue. Check the following parameters.

```
param show NAV_RCL_ACT
param show NAV_DLL_ACT
```

If you plan to fly with no remote control, just using DroneShell commands for example, then you will need to set the following parameters to stop the PX4 from triggering “failsafe mode on” every time a move command is finished.

```
param set NAV_RCL_ACT 0
param set NAV_DLL_ACT 0
```

https://airsim-fork.readthedocs.io/en/latest/px4_sitl.html#no-remote-control

O2KAN, 2022-09-07:
Thank you for turning back
I tried to set them using GNC but I am still getting the same errors. Is there any other way to set them?

leommiranda, 2022-09-08:
What about COM_RCL_EXCEPT? Is it set to 4? It is also possible to set the parameters using QGroundControl.

O2KAN, 2022-09-11:
I don't have a parameter like that. Probably because of the version of px4 that I use. I am using 1.11.2. But I think I have found a solution. My code did not publish set points correctly before setting offboard mode. Since I changed the code It is setting offboard mode and working as I expected. Thank you @leomichalski

closed 2022-09-11 by O2KAN