mavlink/mavros #1787: FCU Protocol Version is not being set to m_mavlink_status

rossizero, 2022-09-21:
### Issue details
When setting fcu_protocol to v1.0 with mavros 0.33 or later I faced problems communicating with my drone that still runs autoquad.
This is since [this commit](https://github.com/mavlink/mavros/commit/56d782d4bc55209550cf3d2628bb1a6ae94a384c) that was added a bit after mavros version 0.32.3.
Prior the version v1.0 flag `MAVLINK_STATUS_FLAG_OUT_MAVLINK1` was being set to `m_status` correctly. But in this commit the old `m_status` was split into `m_mavlink_status` and `m_parse_status` and the flag now is only being applied to `m_mavlink_status`.
When sending commands (at least via udp) the function `get_status_p()` is called in [line 237](https://github.com/mavlink/mavros/blob/master/libmavconn/src/udp.cpp#L237) which returns the `m_parse_status` and not the `m_mavlink_status` where the flag for the protocol version was set. After adding the flag to `m_mavlink_status` I was able to communicate with the drone again.
I also looked at the messages with [this](https://mavlink.io/en/guide/wireshark.html) wireshark plugin to double check. With mavros version >= 0.33 I was actually still sending mavlink v2.0 messages even after successfully setting the fcu_version=v1.0. After adding the flag to the `m_mavlink_status` I was sending v1.0 message again. This also works for the latest version for mavros and noetic.
While searching for possible causes for my problem I stumbled across two issues that also struggled with setting the flag to v1.0 where people suggested to just use version 0.32: [this](https://github.com/mavlink/mavros/issues/1423) and [that](https://github.com/mavlink/mavros/issues/1687). Maybe this will solve those problems as well.
I am not quite sure what would be the best approach: also setting the flag in `m_parse_status`, returning `m_mavlink_status` in function `get_status_p()` or instead using `get_status()` in above mentioned line.

### MAVROS version and platform
Mavros: 0.32.3 & 0.33 & latest
ROS: tested for melodic & noetic 
Ubuntu: tested on 18.04 & 20.04

vooon, 2022-09-21:
Aha, yes, great finding!

Perhaps best option to add new `get_send_status_p()` and update `send_message()`.
I'll try doing that on the weekend.

vooon, 2022-09-24:
Should be fixed in 1.14. Please reopen if i'm wrong.

rossizero, 2022-10-10:
I just tried it out but for some reason the `m_mavlink_status.flags` is back to v2.0 when calling `get_status_p` even it is being set to v1.0 correctly. Does this mean it is overridden or being set to v2.0 somewhere else?

rossizero, 2022-10-10:
If I change line 105 in interface.cpp to use `m_mavlink_status` too, it works:
 `auto msg_received = static_cast<Framing>(mavlink::mavlink_frame_char_buffer(&m_buffer, &m_mavlink_status, c, &message, &m_mavlink_status));`

This would  make `m_parse_status` obsolete though. Maybe it would be better to just set the correct flag in `m_parse_status` and make `get_status_p` return `m_parse_status` again?

rossizero, 2022-10-25:
like the above

commit 48f93661 (2022-09-24): libmavconn: fix MAVLink v1.0 output selection

Fix #1787
PR #1794 (2022-10-25): libmavconn: fix MAVLink v1.0 output selection
Fix #1787
commit 6b4949bd (2022-12-30): libmavconn: fix MAVLink v1.0 output selection

Fix #1787
closed 2022-12-30 by vooon