mavlink/mavros #1790: Mavros problem with WSL2

paoloelle, 2022-10-07:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I have a problem when I run the command `roslaunch mavros px4.launch fcu_url:=udp://:14555@192.168.4.1:14550` in WSL2 with Ubuntu 20.04 because all remains stacked at this line `[ INFO] [1665158081.290587968]: MAVROS started. MY ID 1.240, TARGET ID 1.1`. If I run the same command in Ubunt 20.04 all seems to work. How can I fix this problem?

### Autopilot type and version

[  ] ArduPilot
[ x] PX4

This is the output after launching the commmand:
`roslaunch mavros px4.launch fcu_url:=udp://:14555@192.168.4.1:14550
... logging to /home/paolo/.ros/log/5a55d510-4658-11ed-a318-00155df3a8d8/roslaunch-Asus_Paolo-885.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Asus_Paolo:39293/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/camera/frame_id: base_link
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
[... 232 lines cut ...]
[ INFO] [1665158081.264669158]: TM: Timesync mode: MAVLINK
[ INFO] [1665158081.265261758]: TM: Not publishing sim time
[ INFO] [1665158081.266412459]: Plugin sys_time initialized
[ INFO] [1665158081.266577459]: Plugin terrain loaded
[ INFO] [1665158081.267185859]: Plugin terrain initialized
[ INFO] [1665158081.267322859]: Plugin trajectory loaded
[ INFO] [1665158081.271304461]: Plugin trajectory initialized
[ INFO] [1665158081.271429961]: Plugin tunnel loaded
[ INFO] [1665158081.273742062]: Plugin tunnel initialized
[ INFO] [1665158081.273867462]: Plugin vfr_hud loaded
[ INFO] [1665158081.274436962]: Plugin vfr_hud initialized
[ INFO] [1665158081.274591762]: Plugin vibration blacklisted
[ INFO] [1665158081.274820762]: Plugin vision_pose_estimate loaded
[ INFO] [1665158081.281730765]: Plugin vision_pose_estimate initialized
[ INFO] [1665158081.281913465]: Plugin vision_speed_estimate loaded
[ INFO] [1665158081.285859366]: Plugin vision_speed_estimate initialized
[ INFO] [1665158081.285988666]: Plugin waypoint loaded
[ INFO] [1665158081.289485168]: Plugin waypoint initialized
[ INFO] [1665158081.289558568]: Plugin wheel_odometry blacklisted
[ INFO] [1665158081.289657768]: Plugin wind_estimation loaded
[ INFO] [1665158081.290194468]: Plugin wind_estimation initialized
[ INFO] [1665158081.290335068]: Built-in SIMD instructions: SSE, SSE2
[ INFO] [1665158081.290419368]: Built-in MAVLink package version: 2022.8.8
[ INFO] [1665158081.290506868]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[ INFO] [1665158081.290587968]: MAVROS started. MY ID 1.240, TARGET ID 1.1`

niepingyu, 2024-07-24:
Hello, I also had the same problem, did you solve it？

paoloelle, 2024-11-08:
@niepingyu sorry for the late reply. No I didn't I just switched to the native version of Ubuntu