mavlink/mavros #1791: can't connect mavros on raspi to FCU PX4 mini

huytd-pi, 2022-10-11:
----

### Issue details

I connect px4 to raspberry pi 4 via uart0, GPIO port 14 and 15.

### MAVROS version and platform

ROS: noetic
OS: Raspbian Debian version: 10 (buster)

### Autopilot type and version

[x] PX4

Version: stable release

### Node logs

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/camera/frame_id: base_link
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
[... 231 lines cut ...]
[ INFO] [1665484055.256424940]: Plugin waypoint initialized
[ INFO] [1665484055.256658658]: Plugin wheel_odometry blacklisted
[ INFO] [1665484055.257033780]: Plugin wind_estimation loaded
[ INFO] [1665484055.259197071]: Plugin wind_estimation initialized
[ INFO] [1665484055.259598118]: Autostarting mavlink via USB on PX4
[ INFO] [1665484055.259770263]: Built-in SIMD instructions: None
[ INFO] [1665484055.259875650]: Built-in MAVLink package version: 2022.3.3
[ INFO] [1665484055.259992147]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[ INFO] [1665484055.260106238]: MAVROS started. MY ID 1.240, TARGET ID 1.1

```

### Diagnostics

launch file
[image]

config px4 port uart&i2c
[image]
[image]

Test with command, but no output
```
rostopic echo /mavros/imu/data_raw
```

vooon, 2022-10-11:
What `rostopic echo -n1 /diagnostics` prints?
Looks like you have problem on port setup. Have you tried other mavlink software (e.g. GCS)?

huytd-pi, 2022-11-08:
có vẻ uart gpio raspi không nhận không đúng vì, tôi cũng đang sử dụng một cổng uart khác trên raspi. tôi sử dụng usb raspi với usb debug trên px4 một cách nhanh chóng.

closed 2022-11-08 by huytd-pi