mavlink/mavros #1795: I cannot read messages from pixhawk mini mavlink with ROS1

ZA0068, 2022-11-04:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

i was trying to read message from mavlink with ROS1 but i cannot read any data. My QGroundControl works just fine...

### MAVROS version and platform

Mavros: 0.18.4
ROS: Noetic
Ubuntu: 20.04.4

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: Latest

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
... logging to /home/zain/.ros/log/ce1a9e3e-5c85-11ed-9b9d-47657f642163/roslaunch-zain-ZenBook-Pro-Duo-UX581LV-UX581LV-33301.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zain-ZenBook-Pro-Duo-UX581LV-UX581LV:37385/

SUMMARY
========

PARAMETERS
 * /mavlink_lora/heartbeats: True
 * /mavlink_lora/serial_baudrate: 2000000
 * /mavlink_lora/serial_device: /dev/ttyACM0
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    mavlink_lora (mavlink_lora/mavlink_lora_node)

auto-starting new master
process[master]: started with pid [33310]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ce1a9e3e-5c85-11ed-9b9d-47657f642163
process[rosout-1]: started with pid [33320]
started core service [/rosout]
process[mavlink_lora-2]: started with pid [33323]
[ INFO] [1667596528.061265706]: Opening serial device: /dev/ttyACM0 baudrate 2000000
[ INFO] [1667596528.061961719]: Serial Port initialized

```

But when i run rostopic mavlink_rx on another terminal i get nothing but empty output.
and even running the python script show_pos.py whill show that no data is being read:

```
Last heard:         Never
Position:           Unknown
Altitude:           Unknown

Last heard:         Never
Position:           Unknown
Altitude:           Unknown

Last heard:         Never
Position:           Unknown
Altitude:           Unknown

```

ZA0068, 2022-11-04:
[IDT_materials_module6.zip](https://github.com/mavlink/mavros/files/9942003/IDT_materials_module6.zip)

ZA0068, 2022-11-04:
This is the file

vooon, 2022-11-05:
That code had nothing from mavros, so it completely unrelated.

closed 2022-11-05 by vooon

ZA0068, 2022-11-05:
Yes it is related to mavros
'

ZA0068, 2022-11-05:
> That code had nothing from mavros, so it completely unrelated.

Yes it is.

source ros noetic
create a catkin workspace and source folder
build it
then add the mavlink_lora folder in the source folder
build it again
source devel setup.bash
then run the node:
``roslaunch mavlink_lora mavlink_lora.launch serial_device:=/dev/ttyACM0 serial_baudrate:=2000000 heartbeats:=true``

open another terminal and source both ros and devel setup.bash
then run ```rostopic echo /mavlink_rx``` it outputs nothing.

vooon, 2022-11-06:
I mean, that this code you've attached is not mavros, do not use any of mavros libs nor use it via ROS topics.
So it only related to mavlink package, not mavros.

I'd suggest to drop this and use mavros itself.