mavlink/mavros #1797: Disarm or kill to stop the drone motors

huytd-pi, 2022-11-07:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I want when the drone lands on the ground, a distance from the ground (eg 5cm) the engine is stopped for landing, then message Disarming denied! Not landed
How can I stop the engine now with mavros ?

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?noetic?
Ubuntu: ?20.04?

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: ?v1.12.3?

### Node logs

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/camera/frame_id: base_link
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
[... 378 lines cut ...]
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
[ INFO] [1667836042.635474388, 16.389000000]: GF: mission received
[ INFO] [1667836042.635653901, 16.389000000]: RP: mission received
[ INFO] [1667836042.635747168, 16.389000000]: WP: mission received```

### Diagnostics

```
When landing 0.1cm close to the ground, the message Disarming denied! Not landed
How can I stop the engine now with mavros ?
```

### Check ID

```
rosrun mavros checkid
```

huytd-pi, 2022-11-08:
in MAVSDK have `kill()` and `terminate()` http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/action.html
so does mavros have similar function or something like that.

vooon, 2022-11-08:
https://mavlink.io/en/messages/common.html#MAV_CMD_COMPONENT_ARM_DISARM

You can do `mavcmd long 400 0 21196 0 0 0 0 0`.

huytd-pi, 2022-11-08:
> https://mavlink.io/en/messages/common.html#MAV_CMD_COMPONENT_ARM_DISARM
> 
> You can do `mavcmd long 400 0 21196 0 0 0 0 0`.

How can i write in python script? thanks!

vooon, 2022-11-08:
https://github.com/mavlink/mavros/blob/master/mavros/scripts/mavcmd#L32

closed 2022-11-18 by huytd-pi