mavlink/mavros #1799: Colcon build Error

OscarDeGar, 2022-11-10:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I am unable to build the mavros package. I was following the source install tutorial on this page https://github.com/mavlink/mavros/blob/ros2/mavros/README.md

I run into errors at the colcon build line. Where mavros_msgs fails to build due to a CMAKE error. I am not quite sure how to solve this issue.  Thank you!

### MAVROS version and platform

Mavros: 2.3.0
ROS: Foxy
Ubuntu: 20.04

### Autopilot type and version
*Cant get far enough to test

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Output of colcon build command

Starting >>> mavlink 
Starting >>> mavros_msgs
--- stderr: mavros_msgs                                         
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< mavros_msgs [0.09s, exited with code 1]
Aborted  <<< mavlink [0.10s]

Summary: 0 packages finished [0.22s]
  1 package failed: mavros_msgs
  1 package aborted: mavlink
  2 packages had stderr output: mavlink mavros_msgs
  3 packages not processed

vooon, 2022-11-10:
Have you done `source /opt/ros/foxy/setup.bash` as suggested by ros distro install?

OscarDeGar, 2022-11-11:
Yes, thanks. Just did this and now I have a new error. It would seem that the python module future cannot be found however it is definitely install and up to date. See error below:

Starting >>> mavlink 
Starting >>> mavros_msgs
--- stderr: mavlink                                             
Traceback (most recent call last):
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ImportError: No module named future
Traceback (most recent call last):
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ImportError: No module named future
make[2]: *** [CMakeFiles/storm32.xml-v1.0.dir/build.make:63: include/v1.0/storm32/storm32.h] Error 1
make[2]: *** [CMakeFiles/standard.xml-v1.0.dir/build.make:63: include/v1.0/standard/standard.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:153: CMakeFiles/storm32.xml-v1.0.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:261: CMakeFiles/standard.xml-v1.0.dir/all] Error 2
Traceback (most recent call last):
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ImportError: No module named future
Traceback (most recent call last):
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ImportError: No module named future
make[2]: *** [CMakeFiles/icarous.xml-v1.0.dir/build.make:63: include/v1.0/icarous/icarous.h] Error 1
make[2]: *** [CMakeFiles/paparazzi.xml-v1.0.dir/build.make:63: include/v1.0/paparazzi/paparazzi.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:207: CMakeFiles/icarous.xml-v1.0.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:234: CMakeFiles/paparazzi.xml-v1.0.dir/all] Error 2
Traceback (most recent call last):
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ImportError: No module named future
make[2]: *** [CMakeFiles/uAvionix.xml-v2.0.dir/build.make:63: uAvionix-v2.0-cxx-stamp] Error 1
make[1]: *** [CMakeFiles/Makefile2:315: CMakeFiles/uAvionix.xml-v2.0.dir/all] Error 2
Traceback (most recent call last):
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ImportError: No module named future
Traceback (most recent call last):
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ImportError: No module named future
make[2]: *** [CMakeFiles/uAvionix.xml-v1.0.dir/build.make:63: include/v1.0/uAvionix/uAvionix.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:180: CMakeFiles/uAvionix.xml-v1.0.dir/all] Error 2
make[2]: *** [CMakeFiles/ASLUAV.xml-v2.0.dir/build.make:63: ASLUAV-v2.0-cxx-stamp] Error 1
make[1]: *** [CMakeFiles/Makefile2:288: CMakeFiles/ASLUAV.xml-v2.0.dir/all] Error 2
Traceback (most recent call last):
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/skull/ros2_ws/src/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ImportError: No module named future
make[2]: *** [CMakeFiles/ualberta.xml-v1.0.dir/build.make:63: include/v1.0/ualberta/ualberta.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:126: CMakeFiles/ualberta.xml-v1.0.dir/all] Error 2
make: *** [Makefile:118: all] Error 2
---
Failed   <<< mavlink [0.16s, exited with code 2]
Aborted  <<< mavros_msgs [0.57s]                               

Summary: 0 packages finished [0.76s]
  1 package failed: mavlink
  1 package aborted: mavros_msgs
  1 package had stderr output: mavlink
  3 packages not processed

aa-calvo, 2022-11-13:
Try installing the "future" python package:

```
python -m pip install -U future
```

vooon, 2022-11-13:
@OscarDeGar have you done rosdep stage?

@aa-calvo i'd suggest not to use pip. Usually it's preferable to use package from distro. E.g. `apt install python3-future`.

jerWenger, 2022-11-14:
I am updating for @OscarDeGar I am working on this project with them. We managed a successful build of the environment by simply uninstalling all packages and repositories and starting over. At that point, there was a permissions error when running macros but this was solved by elevating the read write privileges of the usb/serial interface. 

Thank you for your help!

closed 2022-11-14 by OscarDeGar

vooon, 2022-11-15:
@jerWenger if you on Debian or Ubuntu simply add a user to `dialout` group. Look at `ls -la /dev/tty{USB,ACM}*`