mavlink/mavros #1803: /mavros/imu/data not working in Pixhawk 6C

EhrazImam, 2022-11-17:
### Issue details

Hi i am using Pixhawk 6C with px4(1.13.1)
When I’m changing the Pixhawk parameters in QGC **(SYS_MC_EST_GROUP = No position**, **SYS_HAS_BARO & SYS_HAS_MAG = 0**), it only displays /mavros/imu/data_raw not **/mavros/imu/data_raw**. With same parameter I have also tried on Pixhawk 4, Cube Orange and Cube Black with different firmware(1.11.0, 1.11.1, 1.12.0) and they all displaying /mavros/imu/data and /mavros/imu/data_raw as well.

### MAVROS version and platform

Mavros: 1.14.0
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: 1.13.1

OPyshkin, 2022-11-17:
Similar problem. Using pixhawk6c and cannot receive linear acceleration values from imu/data topic. imu/data_raw is not published at all

EhrazImam, 2022-11-18:
> Similar problem. Using pixhawk6c and cannot receive linear acceleration values from imu/data topic. imu/data_raw is not published at all

try it by increasing the IMU hz this may solve the issue

bai225324, 2022-12-04:
similar problem.Could you share how you solve it?

OPyshkin, 2022-12-04:
@bai225324 yes
Firstly, I selected the right airframe for my vehicle in qgc. Then in parameters->mavlink tab set the following params:
- MAV_x_FORWARD - enabled 
- MAV_x_MODE - ONBOARD
Where x is the number of used interface.
On the sd card of flight controller Edit/create file /etc/extras.txt and add line 
mavlink stream -d *your connection port on px4, for me it's /dev/ttyS3 for gps2* -s HIGHRES_IMU -r 200
 and then reboot the controller.

EhrazImam, 2022-12-05:
Hi @OPyshkin are you getting the /mavros/imu/data when you are switching SYS_MC_EST_GROUP = Q attitude  No position 
waiting for your reply

OPyshkin, 2022-12-05:
@EhrazImam no, I didn't switch that

EhrazImam, 2022-12-05:
Hi @OPyshkin can you check this on your hardware for me as i am having only one 6C

OPyshkin, 2022-12-05:
@EhrazImam ok, I'll try when I have time. Also could you briefly describe here what SYS_MC_EST_GROUP parameter stands for?

EhrazImam, 2022-12-05:
@OPyshkin By default this comes with EKF2 (preferred) option that comes with the flight control requires a magnetometer and a barameter for state estimation. These two are greatly interfered in small aircraft, resulting in unstable state estimation and control, low accuracy and frequently flight control restarts, but the Advantage is that you can have fixed-height and fixed point control. As we are you Visual odometery so we are giving the attitude with Vision only so we don't need it.

EhrazImam, 2022-12-08:
Hi @OPyshkin have you tried it.

EhrazImam, 2022-12-30:
@OPyshkin Didi you checked it.

closed 2026-10-23 by github-actions[bot]