mavlink/mavros #1810: Optical Flow sensor data not echoing

Sanjil96, 2022-12-08:
### Issue details

Hi, I am new to this topic. I am using a ThoneFlow 3901U optical flow sensor and a VLX53L1X ToF sensor with Arducopter firmware. In the apm_config.yaml file I removed the px4flow from the blacklist so that I can have the optical flow topics published. When I launch apm.launch file, I see a topic **/mavros/px4flow/raw/optical_flow_rad** being published but there is no data when I try to echo this topic. However, the other topics such as the range finder and imu topics are publishing the data.

How can I get the optical flow sensor data to get published? Any hint would be of great help

### MAVROS version and platform

Mavros: 1.14.0
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

ArduCopter
Version: 4.0.7

### Node logs

```
roslaunch mavros apm.launch fcu_url:=udp://:14550@
... logging to /home/tello22/.ros/log/48c23550-7623-11ed-88a9-0970a0aff240/roslaunch-tello22-HP-ProBook-650-G2-46150.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tello22-HP-ProBook-650-G2:34147/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.1
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
 * /mavros/fake_gps/satellites_visible: 6
 * /mavros/fake_gps/speed_accuracy: 0.0
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_hil_gps: True
[... 517 lines cut ...]
    name: "mavros: APM Hardware"
    message: "Low voltage"
    hardware_id: "udp://:14550@"
    values: 
      - 
        key: "Core voltage"
        value: "0.000000"
      - 
        key: "I2C errors"
        value: "0"

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 3721 messages, from 2 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 132, 147, 11030, 22, 24, 152, 27, 29, 30, 33, 163, 36, 165, 42, 44, 173, 178, 62, 65, 193, 74, 77, 100, 111, 241, 253, 125
  2:0     185

```

ZhongmouLi, 2023-12-16:
Did you solve it?

Sanjil96, 2024-07-24:
Unfortunately not, I stopped working on that project.

closed 2024-07-24 by Sanjil96