mavlink/mavros #1814: MAVROS does not support gimbal v2 protocol

boris-gu, 2022-12-24:
### Issue details
 
I need to launch a Typhoon H480 quadcopter in Gazebo and get the position of its gimbal through MAVROS.  
But I couldn't do it, no data is published in the topics `/mavros/mount_control/orientation` and `/mavros/mount_control/status`.  
A little more detail: https://discuss.px4.io/t/getting-gimbal-position-typhoon-h480/30162

### MAVROS version and platform

Mavros: `dpkg -s ros-noetic-mavros | grep Version` command gives me the following result: **1.14.0-1focal.20221209.205530**
ROS: Noetic
Ubuntu: 20.04.5

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: 1.13.1

Mark-Beaty, 2023-01-31:
@boris-gu I actually have a plugin for this feature almost completed and should have a PR/fork set up publicly hopefully this week, I will update here when that is available

Mark-Beaty, 2023-02-02:
@boris-gu I've created a [PR](https://github.com/mavlink/mavros/pull/1825) which adds a plugin to enable support for gimbal protocol v2, though I don't think it will be compatible with your ROS version immediately.  I've tested this with ROS2 Foxy and Humble so far

closed 2026-10-22 by github-actions[bot]