mavlink/mavros #1818: Mavros connection error

bilalkah, 2022-12-30:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I am having struggle to connect PX4 device with mavros. I use same launch file with same parameters and sometimes it works sometimes it does not. Devices is connecting to QGroundcontrol directly when I plug it to my computer via USB port and I can see it in /dev and /dev/serial/by-id/ folder. Also it does boot normally the tune sounds and the led colors are correct. So where is the problem? I tried several baudrates and nothing changed. I tried 57600, 115200 and 921600, response was same. Sometimes connecting and sometimes not. I tried it with two different computer and the results are same.  

Additionally, I have connected device before. Then I changed the autopilot to Ardupilot to see something for 3 days. Then I changed it back to PX4 and there is mavros connection problem.

### MAVROS version and platform

Mavros: ?1.13.0?
ROS: ?Melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: ?1.13.2?

### Node logs

```
... logging to /root/.ros/log/ba88ffa4-881e-11ed-b932-70f754eb32bb/roslaunch-Khadas-437.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.100.167:34349/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/camera/frame_id: base_link
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
[... 416 lines cut ...]
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "0.000000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
---

```

### Check ID

```
ERROR. I got 4 addresses, but not your target 1:1

---
Received 16 messages, from 4 addresses
sys:comp   list of messages
 190:8     27
 212:134   53
 53:205   179
 91:75    255

```

vooon, 2022-12-30:
Hi,

Baud rate doesn't really matter for USB port on STM32 based devices as they naively implements ACM device.
It only matters if you have real UART somewhere on the chain, e.g. FTDI chip -> AVR on old APM 2.6.
Or if you use TELEM port with the UART of the onboard computer.

I suspect that QGC claims the port, at least i saw that on linux...

Other possibility that you've installed newer PX4 which for whatever reason has incompatible HEARTBEAT.
But that realy-really-really strange to happen. Some signal noise more probable.

vooon, 2022-12-30:
Hmm, more over, you have some traffic on port (see Rx/Tx speed diag), but check-id looks like you catch some garbage (because you have several random addresses, it's unlikely to have different system id for the single FCU, except for 3D RADIO message).

vooon, 2022-12-30:
Have you restored params after reflash?

bilalkah, 2022-12-30:
I closed QGroundcontrol on my computer and did not work. Also The second computer I did not download it so it can not occupied the port I think.  After I reflash I just did basic configuration and calibration. I believe, I should able to connect fcu with mavros right? Also, I have second board of mine and its same flight controller cuav v5+. I just flashed last stable version from QGroundcontrol and able to connect with QGroundcontrol. However, it did not work with mavros again.

bilalkah, 2022-12-30:
As you said, mavros sees the board. Because when I eject the usb connection node kills itself. If I wait for a long time without cutting the connection, it throws 
```
serial0: write: No such device px4
```
error and kills itself. After that device start playing the error tune.

bilalkah, 2022-12-30:
I flashed previous version of PX4 which is 1.13.1 and same problem continues. I need help...

vooon, 2022-12-30:
Ok, i think it's not related to QGC. Could you try to connect with mavproxy (pymavlink)?

vooon, 2022-12-30:
Also, could you please try to build today's version of mavros (1.15.0)?

bilalkah, 2023-01-02:
Hi @vooon ,
I tried latest verison of mavros and it did seem work for both of the fcus. Thank you for your help.

closed 2023-01-02 by bilalkah

vooon, 2023-01-03:
It's good that update helps, but bad, that we didn't really understand what is the source.

bilalkah, 2023-01-04:
Me neither. It was working for 8 months approximately and everything was fine. The problem occurred after, I changed PX4 autopilot estimator to LPE and I lost connection with board. So I reset the board with ArduPilot because QGC could not establish any connection with board. So I work with it for a while and also I connected to board with same version. Then I changed back to PX4 autopilot and could not connect properly.

bilalkah, 2023-01-04:
Besides it works, it gives these warnings;
```
[ WARN] [1672756008.687417010]: Could not retrive negate_measured_roll parameter value, using default (0)
[ WARN] [1672756008.688300457]: Could not retrive negate_measured_pitch parameter value, using default (0)
[ WARN] [1672756008.689690356]: Could not retrive negate_measured_yaw parameter value, using default (0)
[ WARN] [1672756008.692487903]: Could not retrive debounce_s parameter value, using default (4.000000)
[ WARN] [1672756008.693221227]: Could not retrive err_threshold_deg parameter value, using default (10.000000)
```
```
[ WARN] [1672756012.785014845]: TM : RTT too high for timesync: 2298.89 ms.
[ WARN] [1672756012.785991249]: TM : RTT too high for timesync: 1699.90 ms.
[ WARN] [1672756012.787567395]: TM : RTT too high for timesync: 1101.47 ms.
[ WARN] [1672756012.789555661]: TM : RTT too high for timesync: 503.48 ms.
```

vooon, 2023-01-04:
First one - `mount_control` - https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/mount_control.cpp#L187

Second - really high RTT for time sync, but perhaps you can ignore.

bilalkah, 2023-01-04:
Also, may I ask that how modifiable odom plugin is? Because I want to feed my FCU with external odometry source except intel realsense t265 however I could not get the same stability. I am not sure should it be done by FCU or MAVROS. Should I focus on odometry topics or odometry tf of my odometry source. It seems complicated to me. Because I could not understand the instructions. They are too basic and does not briefly explain to me what to do in calculation part.

vooon, 2023-01-04:
Sorry, i can't recommend anything here. Suppose you to contact with the author of the plugin.

Nikhiljohnson21, 2023-08-31:
Hi I am also facing a similar issue
I'm trying to send commands via mavros to my pixhawk flight controller to control the drone. For now I'm testing with 1 motor.

Mavros is connected to pixhawk and I'm able to arm the motor
"rosrun mavros mavsafety arm"

Now I'm trying to take-off the drone using the command
"rosrun mavros mavcmd takeoff (with the yaw, pitch roll, latitude, longitude)"

But I'm getting "request failed. Check mavros logs. ACK:4"

I followed the steps instructed in mavros/utility commands documentation.